espressif/arduino-esp32 #7297: SdFat and Webserver

podaen, 2022-09-26:
### Related area

Webserver

### Hardware specification

ESP32

### Is your feature request related to a problem?

I am using SdFat lib and the webserver lib in to one project. If I use SdFat I can not use FS, but it is included in the webserver... i don't see a way to use others than the FS lib? I see it involves the request. 

### Describe the solution you'd like

?

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

podaen, 2022-09-28:
I don't get it why FS is needed to make a request. I stating to seem like a bug. Because my compiler is looking for the FS lib because of it is included in the webserver lib. So I imported it, but I can not compile because I use allread SDFat. Then I delete the FS lib and then it and it starts compiling??? The webserver works, but sometimes the internet connection at startup fails.

me-no-dev, 2022-09-28:
We have our own SD and SDMMC library to deal with SDCARDs. All file systems on ESP32 are extending FS, which allows to give any file system to the Web Server, not just SDCard.

podaen, 2022-09-28:
It could be the bug is software. The verbose log is turned off (after it was on) 

> [     3][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80,`��\.)
> [     3][V][WebServer.cpp:87] WebServer(): WebServer::Webserver(port=80)

I don't use a pointer `WebServer server(80);` 

Is there any solution for long strings with FS?

podaen, 2022-09-28:
> We have our own SD and SDMMC library to deal with SDCARDs. All file systems on ESP32 are extending FS, which allows to give any file system to the Web Server, not just SDCard.

Then I have to look for a more simple webserver I think, because I don't need all that I think. I only use it to login to the WiFi. The preferences lib does the rest.

closed 2022-10-01 by podaen