espressif/arduino-esp32 #7302: HID device with multiple 0x2a4d characteristics - can not register for notification

kostuch, 2022-09-26:
### Board

ESP32 ESP32C3

### Device Description

Devkit module

### Hardware Configuration

Nothing connected physically

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Win10

### Flash frequency

40 80

### PSRAM enabled

no

### Upload speed

115200

### Description

My BLE device is a steering wheel audio control like this: [link](https://www.joom.com/en/products/1506935141901327344-132-1-26193-339693970)
and it connect to phone or computer as a hid keyboard.
Unfortunately it does not notify client built on ESP32C3 or ESP32. Client code taken from examples.
According to log, there are no errors, but I receive no notifications from device.
I suspect, that the problem is caused by the fact, that this device have many 0x2a4d characteristics, and only writing to correct 0x2902 handle, makes notification works.
Below is a session from linux gatttool:
```

gatttool -i hci0 -b BE:0F:48:64:BB:DE -I
[BE:0F:48:64:BB:DE][LE]> connect
Attempting to connect to BE:0F:48:64:BB:DE
Connection successful
[BE:0F:48:64:BB:DE][LE]> primary 1812
Starting handle: 0x0023 Ending handle: 0x0049
[BE:0F:48:64:BB:DE][LE]> characteristics 23 49
handle: 0x0024, char properties: 0x06, char value handle: 0x0025, uuid: 00002a4e-0000-1000-8000-00805f9b34fb
handle: 0x0026, char properties: 0x1a, char value handle: 0x0027, uuid: 00002a4d-0000-1000-8000-00805f9b34fb <-- 1st 0x2a4d
handle: 0x002a, char properties: 0x1a, char value handle: 0x002b, uuid: 00002a4d-0000-1000-8000-00805f9b34fb <--
handle: 0x002e, char properties: 0x0e, char value handle: 0x002f, uuid: 00002a4d-0000-1000-8000-00805f9b34fb <--
handle: 0x0031, char properties: 0x1a, char value handle: 0x0032, uuid: 00002a4d-0000-1000-8000-00805f9b34fb <--
handle: 0x0035, char properties: 0x1a, char value handle: 0x0036, uuid: 00002a4d-0000-1000-8000-00805f9b34fb <--
handle: 0x0039, char properties: 0x1a, char value handle: 0x003a, uuid: 00002a4d-0000-1000-8000-00805f9b34fb <--
handle: 0x003d, char properties: 0x1a, char value handle: 0x003e, uuid: 00002a4d-0000-1000-8000-00805f9b34fb <-- 7th 0x2a4d
handle: 0x0041, char properties: 0x02, char value handle: 0x0042, uuid: 00002a4b-0000-1000-8000-00805f9b34fb
handle: 0x0043, char properties: 0x1a, char value handle: 0x0044, uuid: 00002a33-0000-1000-8000-00805f9b34fb
handle: 0x0046, char properties: 0x02, char value handle: 0x0047, uuid: 00002a4a-0000-1000-8000-00805f9b34fb
handle: 0x0048, char properties: 0x04, char value handle: 0x0049, uuid: 00002a4c-0000-1000-8000-00805f9b34fb

[BE:0F:48:64:BB:DE][LE]> char-read-uuid 2902
handle: 0x000b 	 value: 00 00 
handle: 0x0022 	 value: 00 00 
handle: 0x0028 	 value: 00 00 
handle: 0x002c 	 value: 00 00 
handle: 0x0033 	 value: 00 00 
[BE:0F:48:64:BB:DE][LE]> char-write-req 33 0100 <-- Writing other handles have no effect and no notification received
Characteristic value was written successfully
Notification handle = 0x0032 value: 01 <-- button press
Notification handle = 0x0032 value: 00 <-- button release
Notification handle = 0x0032 value: 10 <-- other button
Notification handle = 0x0032 value: 00 ...
Notification handle = 0x0032 value: 02 
Notification handle = 0x0032 value: 00 
Notification handle = 0x0032 value: 04 
Notification handle = 0x0032 value: 00 
Notification handle = 0x0032 value: 08 
Notification handle = 0x0032 value: 00

```
I have no idea which handle uses BLE library by default, in case multiple 0x2a4d are discovered. First or last one?
Apparently some HID devices need a possibility to pick a right handle to make notification work (in my case 4th 0x2a4d handle). Or maybe write all found 0x2902 descriptors? User have no possibility to know (without gatttool) which handle use, to make things work.

Currently registerForNotify() has no parameter to choose correct handle, and because of this notifications do not work in all cases.
Is it possible to modify library for such functionality?

### Sketch

```cpp
Straight BLE client code from examples.
```

### Debug Message

```plain
none
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Parsaabasi, 2025-01-16:
Hello, 

Due to the overwhelming volume of issues currently being addressed, we have decided to close the previously received tickets. If you still require assistance or if the issue persists, please don't hesitate to reopen the ticket.

Thanks.

closed 2025-01-16 by Parsaabasi

h0lmes, 2025-01-23:
I kindly ask you to reopen this issue.
I was waiting for it to be resolved for my project to proceed.

h0lmes, 2025-03-02:
@Parsaabasi
How do I reopen this issue?