espressif/arduino-esp32 #7306: LEDC pull request introduced regressions https://github.com/espressif/arduino-esp32/pull/6045

habazut, 2022-09-28:
### Board

ESP32 Dev Module

### Device Description

Dev board and LED

### Hardware Configuration

GPIO connected to LED

### Version

v2.0.4

### IDE Name

Arduino IDE

### Operating System

Linux

### Flash frequency

Whatever

### PSRAM enabled

no

### Upload speed

115200

### Description

The pull request https://github.com/espressif/arduino-esp32/pull/6045 introduced two regressions:

1. It is not possible any more to do the following any more:

ledcSetup(...); ledcAttachPin(...); ledcWrite(...); // set up channel with freqency an duty cycle to blink LED
ledcDetachPin(...) // remove pin (LED OFF)
ledcAttachPin(...) // attach pin again and it continues with SAME frequency and duty as before

as ledcAttachPin nowadays resets the duty cycle of a prefectly working channel setup.

2. ledcSetup(channel, 1, 8) can nor set up frequency of 1Hz any more

Harald.

### Sketch

```cpp
I think you understand anyway, see above.
```

### Debug Message

```plain
00:49:16.724 -> E (247241) ledc: requested frequency and duty resolution can not be achieved, try reducing freq_hz or duty_resolution. div_param=8460032
00:49:16.758 -> [247245][E][esp32-hal-ledc.c:75] ledcSetup(): ledc setup failed!
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

habazut, 2022-09-28:
Suggestion how to fix ledcAttach:
```
void ledcAttachPin(uint8_t pin, uint8_t chan)
{
    if(chan >= LEDC_CHANNELS){
        return;
    }
    uint8_t group=(chan/8), channel=(chan%8);    
    ledc_set_pin(pin, group, channel);
}
```
Regards,
Harald.

VojtechBartoska, 2022-09-29:
Hello @habazut, our Developer responsible for LedC is on vacation. He will take a look on this at mid of October. Thanks for understanding.

habazut, 2022-09-29:
I have a workaround for the ledcAttach() issue (call ledc_set_pin direct) but not for the frequency issue of ledcSetup() for 1Hz. So if anyone has an idea how to get a LED to blink with PWM 1Hz, I would be grateful if you let me know.

Regards,
Harald.

P-R-O-C-H-Y, 2022-10-12:
Hello @habazut ,
1. I am already working on a fix to keep the channels duty after pin attach. Will have PR ready today.
2. About the 1Hz frequency. There is no issue for setting that, but you need to be careful what frequency and duty resolution you select. See ESP-IDF LEDC docs for more info -> [link](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/ledc.html#supported-range-of-frequency-and-duty-resolutions)

Simple rule for using LEDC
lower frequencies -> higher resolution
higher frequencies -> lower resolution

I was able to set 1Hz frequency using 12 bits resolution `ledcSetup(0,1,12);`

PR #7346 (2022-11-11): LEDC - AnalogWrite new API + ledcAttachPin duty fix
## Description of Change
1. Function `ledcAttachPin()` now reads duty from channel before setup in order to keep its current duty.
2. Added new API functions for analogWrite, which sets frequency or resolution for all led channels used by analogWrite.

- `analogWriteFrequency(uint32_t freq)`
- `analogWriteResolution(uint8_t bits)`

## Tests scenarios
Tested on ESP32.

Sketch for testing the fix with analogWrite(connect led to pin 2 - should blink with 500ms delays):

```
uint8_t led = 2;

void setup() 
{            
  analogWrite(led, 128);
  [... cut]
closed 2022-11-11 by SuGlider