espressif/arduino-esp32 #7310: Empty payload on an MQTT

podaen, 2022-09-29:
### Board

ESP32

### Device Description

ESP32 dev Module

### Hardware Configuration

No

### Version

v2.0.4

### IDE Name

1.8.19

### Operating System

Windows 7

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I notist that I having some issues with an empty payload if I use a CA. With insecure everything just works fine! As well HTTP and HTTPS (MQTT) clients disconnecting (because of a timeout) after receiving the empty payload. I think you guys might be interested.

### Sketch

```cpp
...
```

### Debug Message

```plain
see my description.
```

### Other Steps to Reproduce

I do.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

podaen, 2022-09-29:
it's sync (non blocking).

podaen, 2022-09-29:
To be clear, the timeout is not the issue. The empty payload is with a CA.

VojtechBartoska, 2022-10-21:
@SuGlider Can you please help with triage of this issue? Thank you.

bertmelis, 2022-10-21:
This can be closed.

podaen, 2022-10-22:
I just did the test over with the modification I have made and it works. Two reasons that could be the issue for this.

1. Power suffering, but just enough to work without payload when insecure.
2. Only WiFi_Protocol_11b was set.

I solved those issues and now it's working with CA and without payload.

closed 2022-10-22 by podaen