espressif/arduino-esp32 #7316: Arduino IDE returns "exit status 1" error after compiling

Arda-Bildik, 2022-09-30:
### Board

ESP32 Dev Module

### Device Description

ESP32 DevKit v1

### Hardware Configuration

This does not has anything related to this problem.

### Version

v2.0.5

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

While trying to upload even compile the program arduino returns exit status 1 error. Already tried reinstalling board manager, arduino ide, libraries etc. but had no improvement so far.

### Sketch

```cpp
Sketch is 2600 ish lines of code that is divided between single ".ino" and ten ".h" files.
Google Drive link: https://drive.google.com/drive/folders/1waX313ppDM4agPSnRGZXLhImBEpdzlCk?usp=sharing
```

### Debug Message

```plain
In file included from c:\users\arda bildik\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\memory:65,
                 from C:\Users\Arda Bildik\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.5\libraries\FS\src/FS.h:24,
                 from C:\Users\Arda Bildik\Desktop\Projeler\MTE-2123\Main System And Front Panel Interface\Main_System_Code\Main_System_Code.ino:6:
c:\users\arda bildik\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\bits\stl_uninitialized.h:67:1: error: expected ',' or ';' before 'namespace'
 namespace std _GLIBCXX_VISIBILITY(default)
 ^~~~~~~~~
```

### Other Steps to Reproduce

I have tried on another Win10 computer and it works just fine.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-10-01:
Windows long path problem probably. Dont use google drive. Try to place your project on local drive (not to deep in folders!!).

Arda-Bildik, 2022-10-01:
My code files is already on my local drive I uploaded it to google drive to add it to the issue. An d it has nothing to do with the headers for the code it has something to do with the esp32 board manager. Same files compile without a problem on a different PC.

Jason2866, 2022-10-01:
White space can make weird issues. Dont do that `\Main System And Front Panel Interface\Main_System_Code\Main_System_C`. May work with Linux, may work sometimes...

Arda-Bildik, 2022-10-01:
I tried it without the spacing in the file names but it still gives the same error.

SuGlider, 2022-10-05:
@Arda-Bildik 

>While trying to upload even compile the program arduino returns exit status 1 error. Already tried reinstalling board manager, arduino ide, libraries etc. but had no improvement so far.

This occurs because there are errors in the code. It has nothing to do with the board, Arduino, Arduino IDE or the libraries.
In order to fix it, you must review the error messages from the compiler.

Turn on "Show verbose output during: [x] compilation [x] upload" in the Menu->File->Preferences 
You shall see the messages.

[image]

SuGlider, 2022-10-05:
This issue is solved...

closed 2022-10-05 by SuGlider