espressif/arduino-esp32 #7317: Part of the development version is missing in Arduino IDE 2.0, but it can be found in Arduino 1.8.19.

Tinyu-Zhao, 2022-10-01:
### Board

M5Stack-Station

### Device Description

M5Stack-Station

### Hardware Configuration

none

### Version

v2.0.5

### IDE Name

Arduino

### Operating System

macOS 11.7

### Flash frequency

none

### PSRAM enabled

no

### Upload speed

none

### Description

Part of the development version is missing in Arduino IDE 2.0, but it can be found in Arduino 1.8.19.

[image]

### Sketch

```cpp
None
```

### Debug Message

```plain
None
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-10-01:
This is not our core (this repository). Please post an issue in M5Stack's fork.

closed 2022-10-01 by me-no-dev

Tinyu-Zhao, 2022-10-01:
> This is not our core (this repository). Please post an issue in M5Stack's fork.

I am the maintainer of M5Stack, and I don't know how to deal with this issue.

me-no-dev, 2022-10-03:
Do you see the boards in the normal menu up on top of your screen? Maybe something else missing? Does it work fine with this core?

Tinyu-Zhao, 2022-10-03:
M5Station can be found in Arduino 1.8.19, and there are no errors when using M5Station.

Tinyu Chiu
2022年10月3日 +0800 15:38 espressif/arduino-esp32 ***@***.***>，写道：
>
> Reply

me-no-dev, 2022-10-03:
[image]

Works in our core. I am again asking if it shows in the top menu?

[image]

Tinyu-Zhao, 2022-10-03:
🤔 surprisingly it can appear in the official board, I don't have that installed, I will check our configuration in comparison, thank you

Tinyu Chiu