espressif/arduino-esp32 #7325: ESP32 Wroom32 doesn't show up for provisioning (Both SoftAP & BLE)

techiesms, 2022-10-04:
### Board

ESP32 Wroom 32 

### Device Description

I tried ESP Rainmaker code with several ESP32 boards and it works fine with all of them unless the one which I got from the local market. I have attached the images of that board. 

So when I upload the example sketch or any sketch of ESP Rainmaker, it doesn't show up any BLE Device or SoftAP device fro wifi provisioning. I tried with both Android and iOS devices but the device is not appearing at all. So can anyone tell me what am I doing wrong here or what's the issue with this board. 

I also tried changing ESP32 boards packages version, tried upload the sketch from a windows and Mac system, also tried uploaded different rainmaker sketches but the result is the same 
[image]
[image]

### Hardware Configuration

No

### Version

v2.0.5

### IDE Name

Arduino IDE

### Operating System

macOS 11.5.2

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

I tried ESP Rainmaker code with several ESP32 boards and it works fine with all of them unless the one which I got from the local market. I have attached the images of that board. 

So when I upload the example sketch or any sketch of ESP Rainmaker, it doesn't show up any BLE Device or SoftAP device fro wifi provisioning. I tried with both Android and iOS devices but the device is not appearing at all. So can anyone tell me what am I doing wrong here or what's the issue with this board. 

I also tried changing ESP32 boards packages version, tried upload the sketch from a windows and Mac system, also tried uploaded different rainmaker sketches but the result is the same 
[image]
[image]

### Sketch

```cpp
I tried uploading several sketches but the issue is the same
```

### Debug Message

```plain
NA
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-10-21:
Any updates @techiesms? 

Btw, thanks for your participation in Espressif DevCon22 :)

VojtechBartoska, 2023-02-23:
Closing es Expired due to no answer. If needed, you can reopen it.

closed 2023-02-23 by VojtechBartoska

pablovm43, 2024-04-19:
Hola como estan?, tengo el mismo problema y no lo pude solucionar?, alguno consiguio la solucion?