espressif/arduino-esp32 #7326: EEPROM error, ESP_ERR_NVS_NOT_FOUND if flash is encrypted

DvoIvoRak, 2022-10-05:
### Board

ESP32-S3-DevKitC-1

### Device Description

DevKitC

### Hardware Configuration

No

### Version

v2.0.5

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80 Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

Hi. 
I have problem with my app where I use Arduino EEPROM library and where I use flash encryption(burn my XTS_AES_256_KEY and bootlader from esp-idf project with enable only flash encryption over menuconfig). EEPROM library can't read and write anything. I found that problem is about NVS so I tested this example "nvs_rw_value" from ESP-IDF examples but I have problem if the flash encryption is enable and compile this code by Arduino IDE. File compiled over Arduino IDE still write "The value is not initialized yet!" but if I compile the same code from ESP-IDF I got expected result that "restart counter" increases.  

### Sketch

```cpp
/* Non-Volatile Storage (NVS) Read and Write a Value - Example

   For other examples please check:
   https://github.com/espressif/esp-idf/tree/master/examples

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "nvs_flash.h"
#include "nvs.h"

extern "C" void app_main(void)
{
    // Initialize NVS
    esp_err_t err = nvs_flash_init();
    if (err == ESP_ERR_NVS_NO_FREE_PAGES || err == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        // NVS partition was truncated and needs to be erased
        // Retry nvs_flash_init
        ESP_ERROR_CHECK(nvs_flash_erase());
[... 207 lines cut ...]
DIS_USB (BLOCK0)                                   Disables the USB OTG hardware                      = False R/W (0b0)
USB_EXCHG_PINS (BLOCK0)                            Exchanges USB D+ and D- pins                       = False R/W (0b0)
EXT_PHY_ENABLE (BLOCK0)                            Enables external USB PHY                           = False R/W (0b0)
BTLC_GPIO_ENABLE (BLOCK0)                          Enables BTLC GPIO                                  = 0 R/W (0b00)
DIS_USB_JTAG (BLOCK0)                              Disable usb_serial_jtag-to-jtag function           = True R/W (0b1)
DIS_USB_SERIAL_JTAG (BLOCK0)                       Disable usb_serial_jtag module                     = False R/W (0b0)
USB_PHY_SEL (BLOCK0)                               Select internal/external PHY for USB OTGand usb_se = False R/W (0b0)
                                                   rial_jtag

Vdd_Spi Config fuses:
VDD_SPI_XPD (BLOCK0)                               The VDD_SPI regulator is powered on                = True R/W (0b1)
VDD_SPI_TIEH (BLOCK0)                              The VDD_SPI power supply voltage at reset
   = Connect to VDD_RTC_IO R/W (0b1)
VDD_SPI_FORCE (BLOCK0)                             Force using VDD_SPI_XPD and VDD_SPI_TIEH to config = True R/W (0b1)
                                                   ure VDD_SPI LDO
PIN_POWER_SELECTION (BLOCK0)                       Sets default power supply for GPIO33..37           = VDD3P3_CPU R/W (0b0)

Wdt Config fuses:
WDT_DELAY_SEL (BLOCK0)                             Selects RTC WDT timeout threshold at startup       = 0 R/W (0b00)

Flash voltage (VDD_SPI) set to 3.3V by efuse.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-10-05:
You can not use flash encryption with Arduino IDE. You need to use Arduino as IDF component if you want encryption enabled. For the IDE, IDF is already compiled with encryption off.

closed 2022-10-05 by me-no-dev

apppie123, 2023-01-22:
Opening this again. I'm having the same/similar problem.

I'm using VS Code with framework = arduino, espidf and compile from scratch.
I'm using the arduino preferences library and code example.

I have 2 partitions: factory and OTA_0

When I run without any encryption, I can read and write the nvs in both factory as in OTA_0 over and over again.

However, when turning on flash encryption in development mode(using the makeconfig command in platformIO), I can write and read while staying/rebooting in the factory partition. However, as soon as I move to the OTA_0 partition, I get an error:
␛[0;32mI (707) nvs: NVS partition "nvs" is encrypted.␛[0m
␛[0;��[  1096][E][Preferences.cpp:50] begin(): nvs_open failed: NOT_FOUND

It looks as if the arduino library is not compatible with flash encryption.