espressif/arduino-esp32 #7331: Has anyone successfully migrated their VSCode Platform.IO ESP32 project to ESP32-S3?

Ugobyte, 2022-10-06:
### Board

ESP32-S3 DevKitC-1

### Device Description

ESP32-S3 Dev-Kit C attached to my Windows 10 PC

### Hardware Configuration

MPU6050 IMU Chip is connected to ESP32-S3 (SDA = GPIO 8, SCL = GPIO 9)

### Version

latest master (checkout manually)

### IDE Name

VSCode with Platform.IO

### Operating System

Windows 10

### Flash frequency

40MHz (No idea, i never changed it)

### PSRAM enabled

no

### Upload speed

460800

### Description

I have a project that reads an IMU using i2c. This project is written with VS Code and Platform.IO plug-in with Arduino framework.
I can compile, upload and run my project and everything works beautifully with ESP32.

Unfortunately, the ESP32 does not support USB HID, so I decided to move up to the ESP32-S3. 
I updated my platform.ini for the ESP32-S3. After compiling and uploading my project into the ESP32-S3,
the ESP32-S3 keeps printing out: 
```
[549186][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1
[549204][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1
[549216][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1
[549231][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1
[549246][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1
```

My connection is: SDA = GPIO 8, SCL = GPIO 9.
Has anyone been able to migrate their VS Code with Platform.IO from ESP32 to ESP32-S3? 
What am i missing?

### Sketch

```cpp
Calling  wire.requestFrom() causes error.
```

### Debug Message

```plain
[549186][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1
[549204][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1
[549216][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1
[549231][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1
[549246][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-10-07:
Without an example sketch, no one can help. I2C is working in general.
Probably the lib you use is outdated and needs changes to work.

Ugobyte, 2022-10-08:
I found the issue.
I was using ```Wire.requestFrom(address, length);``` method.
I switched to ```Wire.requestFrom(address, length, stop);``` now it works.

closed 2022-10-08 by Ugobyte

Fehmi-Karaal, 2023-03-16:
#include <LSM6DSV16XSensor.h>
#include <Wire.h>
#include <Arduino.h>
#define I2C_SDA 21
#define I2C_SCL 22

LSM6DSV16XSensor sensor (&Wire);
int32_t accel[3], angrate[3];

void setup() {
  Wire.setPins(I2C_SDA, I2C_SCL);
  Wire.begin();
  Serial.begin(115200);
  sensor.begin();
  sensor.Enable_X();
  sensor.Enable_G();
}

void loop() {
  sensor.Get_X_Axes(accel);
  sensor.Get_G_Axes(angrate);

  Serial.print("Accel-X[mg]:");
  Serial.print(accel[0], 2);
  Serial.print(",Accel-Y[mg]:");
  Serial.print(accel[0], 2);
  Serial.print(",Accel-Z[mg]:");
  Serial.print(accel[0], 2);

  Serial.print(",AngRate-X[mdps]:");
  Serial.println(angrate[0], 2);
  Serial.print(",AngRate-Y[mdps]:");
  Serial.println(angrate[1], 2);
  Serial.print(",AngRate-Z[mdps]:");
  Serial.println(angrate[2], 2);

  delay(500);
}

I used Stmdunio official library to communicate between ESP32 and   STEVAL-MKI227KA evaluation kit. However, I take one error below,

[cpp:499] requestFrom(): i2cWriteReadNonStop returned Error -1

AntarescorpIO, 2023-03-23:
> I found the issue. I was using `Wire.requestFrom(address, length);` method. I switched to `Wire.requestFrom(address, length, stop);` now it works.

sorry but where you change  Wire.requestFrom(address, length); becaus i have the same problem as you please give me a hand