espressif/arduino-esp32 #7333: esp_wifi_set_protocol can only be set to WIFI_PROTOCOL_11B

podaen, 2022-10-08:
### Board

ESP32

### Device Description

DevKit v3

### Hardware Configuration

No

### Version

v2.0.5

### IDE Name

1.8.19

### Operating System

Windows 7

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I discovered that the protocol can only be set  to` const uint8_t protocol = WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G || WIFI_PROTOCOL_11N;` OR `const uint8_t protocol = WIFI_PROTOCOL_11B` OR `const uint8_t protocol = WIFI_PROTOCOL_11B | 0 | 0;` anything else fails. 

It doesn't look the esp32 supports anything else than the very old protocol that only gives me a connection speed of 10Mbs. Any explanation on that?

### Sketch

```cpp
#include "esp_wifi.h"//for maximum power option
#include "esp_log.h"

esp_err_t err;

wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
	err = esp_wifi_init(&cfg);
	if (err != ESP_OK) {
		Serial.println(err);
		Serial.println("Could not set wifi power mode!");
		return false;
	}

	err = esp_wifi_set_mode(WIFI_MODE_STA);
	if (err != ESP_OK) {
		Serial.println(err);
		Serial.println("Could not set sta!");
		return false;
	}

	//const uint8_t protocol = WIFI_PROTOCOL_11G;
	//const uint8_t protocol = WIFI_PROTOCOL_11B | 0 | 0;
	const uint8_t protocol = WIFI_PROTOCOL_11B;

	err = esp_wifi_set_protocol(WIFI_IF_STA, protocol);
	if (err != ESP_OK) {
		//258 = ESP_ERR_INVALID_ARG
		Serial.println(err);
		Serial.println("Could not set protocol!");
		return false;
	}

	uint8_t getprotocol;
	//esp_err_t err;
	err = esp_wifi_get_protocol(WIFI_IF_STA, &getprotocol);
	if (err != ESP_OK) {
		Serial.println("Could not get protocol!");
		//log_e("Could not get protocol! %d", err);
		return false;
	}
	if (getprotocol & WIFI_PROTOCOL_11N) {
		Serial.println("WiFi_Protocol_11n");
	}
	if (getprotocol & WIFI_PROTOCOL_11G) {
		Serial.println("WiFi_Protocol_11g");
	}
	if (getprotocol & WIFI_PROTOCOL_11B) {//worst, but dificuld to set else
		Serial.println("WiFi_Protocol_11b");
	}
```

### Debug Message

```plain
No
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

podaen, 2022-10-08:
Bump, If there are other esp's that support that... I like to know please.

Rotzbua, 2022-10-09:
1. wrong repo. your mentioned function (all `esp_*`) is part of the idf: https://github.com/espressif/esp-idf
1. the restriction is as documented: https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html#_CPPv421esp_wifi_set_protocol16wifi_interface_t7uint8_t
1. `WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G` is also possible
1. `| 0 | 0` makes no change

podaen, 2022-10-09:
1. `esp_*` functions are config by the board/Arduino ESP32. So that's why here.
2. ok
3. Yes, But the WIFI_PROTOCOL_11B will be chosen!
4. I did that to so see if it make any difference to the output. But it doesn't...

me-no-dev, 2022-10-10:
Hi @podaen ! You can still report this in IDF's repo, stating that you are on ESP-IDF 4.4.2 and see if they know of this limitation and how to get around it.

VojtechBartoska, 2022-10-21:
asked to be moved under ESP-IDF.

podaen, 2022-10-21:
I cann't. I made the topic in IDF-ESP a week a go and I am able to set the protocol to B and G, but I am not sure why it works now. I have some ideas... But I solved an issue that could be related to it. I need to do more testing to be sure.