espressif/arduino-esp32 #7337: Improve speed of file system

ttlappalainen, 2022-10-10:
### Related area

vfs_api

### Hardware specification

ESP32

### Is your feature request related to a problem?

https://github.com/espressif/esp-idf/issues/3277

### Describe the solution you'd like

In vfs_api.cpp  FileImplPtr VFSImpl::open(...) calls stat, which is pretty slow with LittleFS or SPIFFS. After that succeeded open returns
`return std::make_shared<VFSFileImpl>(this, path, mode);`
VFSFileImpl constructor calls again stat. I made experimental new constructor, which gets stat structure and full path:
`VFSFileImpl(VFSImpl* fs, const struct stat &st, const char* fpath, const char* path, const char* mode);`
then instead of original constructor I call
```
        if (S_ISREG(st.st_mode) || S_ISDIR(st.st_mode)) {
            FileImplPtr ret=std::make_shared<VFSFileImpl>(this, st, temp, path, mode);
            free(temp);
            return ret;
        }
...
```
Since new constructor gets stat structure from caller, it does not need to call it again and so my open times are half of original. Currently I use that constructor only for existing files, but if new constructor would also get stat result as parameter, it could be used also for new files:
`VFSFileImpl(VFSImpl* fs, const struct stat &st, int statResult, const char* fpath, const char* path, const char* mode);`

Do you find any problem using open in this way?

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

BlueAndi, 2022-11-19:
Additional performance improvements:
* Replace all sprintf(buffer, "%s%s", ...) with strcpy/strcat
* Reduce heap usage by using String, which really slows down. See this code fragment, as example for possible improvement:
https://github.com/espressif/arduino-esp32/blob/bf53e2b8d60ed1942748d9edefaedf58d4bde2ec/libraries/FS/src/vfs_api.cpp#L469-L474

Zhu-jiatong, 2023-01-13:
This fix could be extremely helpful for projects where the iteration of entire directories is needed. Hope it gets implemented soon.

Parsaabasi, 2025-01-16:
Hello, 

Due to the overwhelming volume of issues currently being addressed, we have decided to close the previously received tickets. If you still require assistance or if the issue persists, please don't hesitate to reopen the ticket.

Thanks.

hitecSmartHome, 2025-02-19:
whats up with this?

ttlappalainen, 2025-02-19:
I use the solution I orignally explained in system I make code.

hitecSmartHome, 2025-02-19:
Could you reopen it pls? Fs is really slow

hitecSmartHome, 2025-02-19:
I have tried to tweak fs in the menuconfig to make it faster. I played with cache sizes and things like that. I was able to optimise it for a lot of small files but it was unstable.

ttlappalainen, 2025-02-27:
If there is any use for my changes, I can send them. I have them for Espressif 6.5 version.

hitecSmartHome, 2025-03-01:
It would be awesome

lbernstone, 2025-03-02:
@ttlappalainen Submit a PR.  The devs will make any suggestions they have in the PR, and then it will get merged into the code.

Parsaabasi, 2025-07-25:
Hi @ttlappalainen 

Do you still need help with this?