espressif/arduino-esp32 #7343: attachInterrupt() not supported from C++ static constructor

rimim, 2022-10-11:
### Board

ESP32 Dev Module or any ESP32

### Device Description

Plain ESP32 Dev Module

### Hardware Configuration

No

### Version

v2.0.5

### IDE Name

Arduino IDE

### Operating System

Ubuntu 22

### Flash frequency

240Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

Version 2.0.3 and later no longer support calling attachInterrupt() from a static C++ constructor. That is an acceptable limitation, but the undesirable side effect is that future calls to attachInterrupt() will crash. The reason is probably that previous to version 2.0.3 gpio_install_isr_service() was not used in esp32-hal-gpio.c

### Sketch

```cpp
static void IRAM_ATTR dummyFunction()
{
}

class MyClass
{
public:
	MyClass()
	{
		printf("C++ constructor\n");
		attachInterrupt(17, dummyFunction, CHANGE);
	}

	void success()
	{
		printf("Success\n");
	}
};

MyClass myClass;

void setup()
{
	attachInterrupt(34, dummyFunction, CHANGE);
    attachInterrupt(35, dummyFunction, CHANGE);
    myClass.success();
}

void loop()
{

}
```

### Debug Message

```plain
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
C++ constructor
E (106) gpio: esp_ipc_call_blocking failed (0x103)
[     1][E][esp32-hal-gpio.c:156] __attachInterruptFunctionalArg(): GІ�� ISR Service Failed To Start
[     5][D][esp32-hal-cpu.c:214] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
Setup started
E (38) gpio: gpio_install_isr_service(449): GPIO isr service already installed
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400e8a3f  PS      : 0x00060c33  A0      : 0x800d16ab  A1      : 0x3ffb2760  
A2      : 0x00000000  A3      : 0x3ffc1700  A4      : 0x3ffc1704  A5      : 0x00060c23  
A6      : 0x00060c20  A7      : 0x00000001  A8      : 0x800d1614  A9      : 0x3ffb2720  
A10     : 0x3ffbdba4  A11     : 0x00000004  A12     : 0x00000022  A13     : 0x3ffb2760  
A14     : 0x007bdb90  A15     : 0x003fffff  SAR     : 0x0000001e  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40085fd1  LEND    : 0x40085fe1  LCOUNT  : 0xfffffff7  

Backtrace:0x400e8a3c:0x3ffb27600x400d16a8:0x3ffb2780 0x400d0e8c:0x3ffb27b0 0x400d0ed9:0x3ffb27e0 0x400d0da7:0x3ffb2800 0x400d10d6:0x3ffb2820
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-10-12:
since you declare the class global, it will try to construct way before many necessary things has started. This is the reason why it does not work. If you do it in `setup()`, then it will work. Best option is to not have that in the constructor at all, but instead have it in a `begin()` method, which you can call in `begin` to attach the interrupt.

closed 2022-10-12 by me-no-dev

teldrive, 2022-10-13:
I think I have same problem
-----------------------------on file myclass.h
public:
void IRAM_ATTR timerISR();
void IRAM_ATTR pinchangeISR1();
void IRAM_ATTR pinchangeISR2();
--------------------------on file myclass.cpp
void begin(){
attachInterrupt(digitalPinToInterrupt(rxPin1), pinchangeISR1, CHANGE);
}
and I receive compile error
error: invalid use of non-static member function
 attachInterrupt(digitalPinToInterrupt(rxPin1), pinchangeISR1, CHANGE);

If another issue please help

rimim, 2022-10-14:
You can't take the address of pinchangeISR1() as it is a member function and not marked static. Change your ISR functions like this:

static void IRAM_ATTR timerISR();
static void IRAM_ATTR pinchangeISR1();
static void IRAM_ATTR pinchangeISR2();

teldrive, 2022-10-14:
hi rimim thanks to reply , i forgot to say that I obtain rxpin from function parameter
void begin(int rxPin1){
attachInterrupt(digitalPinToInterrupt(rxPin1), pinchangeISR1, CHANGE);
}
if I set static in myclass.h and myclass.cpp
i cant access to other volatil variables defined in miclass.h
static void IRAM_ATTR pinchangeISR1();

teldrive, 2022-10-14:
I have found a very good article explaining this issue
https://arduino.stackexchange.com/questions/89173/attach-the-arduino-isr-function-to-the-class-member
but after try with glue funcion that seems to be the solution I obtain this error
aliBus-esp32.cpp.o:(.literal._ZN12DaliBusClass18glue_pinchangeISR1Ev[myclass::glue_pinchangeISR1()]+0x0): undefined reference to `myclass::instances
Can anyone help?