espressif/arduino-esp32 #7347: Mac V:12.6 (21G115) +  VSCODE + Platformio   The problem

hzhh110, 2022-10-13:
### Related area

ESP32

### Hardware specification

ESP32

### Is your feature request related to a problem?

python3 -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/master/scripts/get-platformio.py)"

pio platform install espressif32

mkdir ~/Documents/PlatformIO/Projects/testproj1
cd ~/Documents/PlatformIO/Projects/testproj1

pio project init --board nodemcu-32s

After completion, it can be opened normally

However, as long as there are other versions of platform = espressif32 @ ~3.5.0 in the project, you have to repeat the above process, even if you change back to platform = espressif32, it will not work properly

### Describe the solution you'd like

python3 -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/master/scripts/get-platformio.py)"

pio platform install espressif32

mkdir ~/Documents/PlatformIO/Projects/testproj1
cd ~/Documents/PlatformIO/Projects/testproj1

pio project init --board nodemcu-32s

After completion, it can be opened normally

However, as long as there are other versions of platform = espressif32 @ ~3.5.0 in the project, you have to repeat the above process, even if you change back to platform = espressif32, it will not work properly

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

hzhh110, 2022-10-13:
I tried it many times in a row, but the result was the same

SuGlider, 2022-10-13:
@hzhh110 
This issue sounds to be related exclusively to PlatformIO.
I think that you will get a better response if it is open within 
https://github.com/platformio/platform-espressif32/issues
or
https://github.com/platformio/platformio-vscode-ide/issues

Jason2866, 2022-10-13:
@hzhh110  Thats not the recommended way to install Platformio when used with VSC
Install Platformio VSC Extension. The extension installs Platformio. No manuall install.

VojtechBartoska, 2022-10-21:
Closing. Please follow recommendations above. Thank you @SuGlider & @Jason2866.

closed 2022-10-21 by VojtechBartoska