espressif/arduino-esp32 #7349: StoreProhibited of esp_wifi_init(), remove_free_block() function in ESP-IDF core

maxpromer, 2022-10-14:
### Board

ESP32-S3 Dev Module (N8R8)

### Device Description

I use ESP32-S3 Dev Module (N8R8) on my custom board with TFT LCD 8 bit, Use RTC I2C, Keypad I2C and some i/o connect to other device.

### Hardware Configuration

I custom hardware and here is internal connection

[image]

### Version

v2.0.5

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I use WiFi on ESP32-S3 if i use WiFiScan example for test WiFi i don't have it issue but with i use WiFi with my code i have issue. I can't understand debug output `remove_free_block` maybe remove_free_block try to free() ram but i don't have to free() so it 

### Sketch

```cpp
#include <Arduino.h>
#include <SPIFFS.h>
#include <WiFi.h>
....

static const char* TAG = "Main";

void setup() {
    Serial.begin(115200);
    // esp_log_level_set("*", ESP_LOG_VERBOSE);

    if (!SPIFFS.begin(true)) {
      ESP_LOGE(TAG, "Init SPIFFS fail !");
    }
    I2CSetup();
    IO_init();
    ....
    WiFi.mode(WIFI_STA);
    
    Serial.println("Setup done");
}

void loop() {
   WiFiManager_process(); // my function
   SensorRecord_process(); // my function
   delay(100);
}
```

### Debug Message

```plain
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40382554  PS      : 0x00060933  A0      : 0x80382d10  A1      : 0x3fce2a80
A2      : 0x3d800014  A3      : 0x3d8a0ea0  A4      : 0x3fc9bb50  A5      : 0x00060723
A6      : 0x00060720  A7      : 0x00000001  A8      : 0x0000001a  A9      : 0x00000010
A10     : 0x3d80087c  A11     : 0x00000024  A12     : 0xb51db51d  A13     : 0xb51db51d
A14     : 0xb51db51d  A15     : 0xb51db51c  SAR     : 0x0000001f  EXCCAUSE: 0x0000001d
EXCVADDR: 0xb51db529  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff

Backtrace:0x40382551:0x3fce2a800x40382d0d:0x3fce2aa0 0x40382f58:0x3fce2ac0 0x403780bd:0x3fce2ae0 0x403780d1:0x3fce2b10 0x40378399:0x3fce2b30 0x403783e7:0x3fce2b50 0x40377e52:0x3fce2ba0 0x40377e65:0x3fce2bc0 0x420a5702:0x3fce2be0 0x42067d64:0x3fce2c00 0x42045af1:0x3fce2c20 0x42045cf7:0x3fce2da0 0x42014ffd:0x3fce2dc0 0x42053797:0x3fce2df0

  #0  0x40382551:0x3fce2a80 in remove_free_block at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_tlsf.c:212
      (inlined by) block_locate_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_tlsf.c:448
      (inlined by) tlsf_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_tlsf.c:757     
  #1  0x40382d0d:0x3fce2aa0 in multi_heap_malloc_impl at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:197
  #2  0x40382f58:0x3fce2ac0 in multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:230
      (inlined by) multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:219
  #3  0x403780bd:0x3fce2ae0 in heap_caps_malloc_base at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:147
  #4  0x403780d1:0x3fce2b10 in heap_caps_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:167
  #5  0x40378399:0x3fce2b30 in heap_caps_calloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:441
  #6  0x403783e7:0x3fce2b50 in heap_caps_calloc_prefer at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:299
  #7  0x40377e52:0x3fce2ba0 in wifi_calloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_wifi/esp32s3/esp_adapter.c:94
  #8  0x40377e65:0x3fce2bc0 in wifi_zalloc_wrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_wifi/esp32s3/esp_adapter.c:102
  #9  0x420a5702:0x3fce2be0 in esp_wifi_init_internal at ??:?
  #10 0x42067d64:0x3fce2c00 in esp_wifi_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_wifi/src/wifi_init.c:250
  #11 0x42045af1:0x3fce2c20 in wifiLowLevelInit(bool) at C:/Users/sonth/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:676
  #12 0x42045cf7:0x3fce2da0 in WiFiGenericClass::mode(wifi_mode_t) at C:/Users/sonth/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:1113
  #13 0x42014ffd:0x3fce2dc0 in setup() at src/main.cpp:32
  #14 0x42053797:0x3fce2df0 in loopTask(void*) at C:/Users/sonth/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42

ELF file SHA256: 0000000000000000

Rebooting...
ESP-ROM:esp32s3-20210327
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

maxpromer, 2022-10-14:
I have new crash log when i move `WiFi.mode(WIFI_STA);` to first in void loop()
```
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40382554  PS      : 0x00060233  A0      : 0x80382d10  A1      : 0x3fcf30a0
A2      : 0x3d800014  A3      : 0x3d8a0ea0  A4      : 0x3fc9bb4c  A5      : 0x00060023
A6      : 0x00060020  A7      : 0x00000001  A8      : 0x00000009  A9      : 0x00000000
A10     : 0x3d800038  A11     : 0x00000014  A12     : 0xb51db51d  A13     : 0xb51db51d
A14     : 0xb51db51d  A15     : 0xb51db51c  SAR     : 0x0000001b  EXCCAUSE: 0x0000001d
EXCVADDR: 0xb51db529  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff

Backtrace:0x40382551:0x3fcf30a00x40382d0d:0x3fcf30c0 0x40382f58:0x3fcf30e0 0x403780bd:0x3fcf3100 0x403780d1:0x3fcf3130 0x40378179:0x3fcf3150 0x40377e1d:0x3fcf31a0 0x420b918f:0x3fcf31c0 0x420c1137:0x3fcf31e0 0x420bea11:0x3fcf3200 0x4207dfee:0x3fcf3220

  #0  0x40382551:0x3fcf30a0 in remove_free_block at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_tlsf.c:212
      (inlined by) block_locate_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_tlsf.c:448
      (inlined by) tlsf_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_tlsf.c:757     
  #1  0x40382d0d:0x3fcf30c0 in multi_heap_malloc_impl at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:197
  #2  0x40382f58:0x3fcf30e0 in multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:230
      (inlined by) multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:219
  #3  0x403780bd:0x3fcf3100 in heap_caps_malloc_base at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:147
  #4  0x403780d1:0x3fcf3130 in heap_caps_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:167
  #5  0x40378179:0x3fcf3150 in heap_caps_malloc_prefer at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:261
  #6  0x40377e1d:0x3fcf31a0 in wifi_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_wifi/esp32s3/esp_adapter.c:68
  #7  0x420b918f:0x3fcf31c0 in ieee80211_timer_process at ??:?
  #8  0x420c1137:0x3fcf31e0 in pp_timer_process at ??:?
  #9  0x420bea11:0x3fcf3200 in lmac_stop_hw_txq at ??:?
  #10 0x4207dfee:0x3fcf3220 in timer_process_alarm at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_timer/src/esp_timer.c:360
      (inlined by) timer_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_timer/src/esp_timer.c:386

ELF file SHA256: 0000000000000000

Rebooting...
```

me-no-dev, 2022-10-14:
do you have debug enabled? PSRAM is OPI and suggest to have flash at 80MHz. Overall I do not think that that is your problem, but it's definitely something wrong on your side that manifests itself as this memory error. Try to start minimal with PSRAM disabled and debug on. Simple WiFi sketch should run without issues.

VojtechBartoska, 2022-11-08:
any updates @maxpromer?

mrengineer7777, 2022-11-09:
New IDF release includes a fix for PSRAM at 40Mhz and a couple edge cases for WiFi.  Might help your issue.  https://github.com/espressif/esp-idf/releases/tag/v4.4.3.  IDF 4.4.3 should be included in arduino-esp32 version 2.0.6.

VojtechBartoska, 2023-02-23:
Closing es Expired due to no answer. If needed, you can reopen it.

closed 2023-02-23 by VojtechBartoska