espressif/arduino-esp32 #7354: Crash on captive portal connection

zekageri, 2022-10-14:
### Board

esp-wrover-kit

### Device Description

Simple ESP32-wrover-E ( 8mb psram and 16 mb flash )

### Hardware Configuration

ETH_LAN_8720, I2C and SPI

### Version

latest master (checkout manually)

### IDE Name

Platform IO

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

ESP crashes immidiately on AP connection when using Captive portal.
8 times out of 10 try.

My app checks a config file for a flag named firstBoot. If this flag is true, it will abort the main task and creates a new one
which will handle the first boot process. This includes a captive portal for the user. So far so good. When the ESP opens it's AP mode and the user connects to it, it will immidiately crashes.

Debug level set to 5 for debugging.

### Sketch

```cpp
/*
*   A hook callback for external ram allocation fails.
*/
void heap_caps_alloc_failed_hook(size_t requested_size, uint32_t caps, const char *function_name){
    debug.print(
        DEBUG_ERROR,
        "%s was called but failed to allocate %d bytes with 0x%X capabilities. \n",
        function_name, requested_size, caps
    );
}

void setup() {
    Serial.begin(115200);
    vTaskDelay(1000);

    heap_caps_malloc_extmem_enable(500);
    esp_err_t error = heap_caps_register_failed_alloc_callback(heap_caps_alloc_failed_hook);
    // Setting up the file system.
    fileSys.init();
    // Loading the configuration file.
    config.load();
[... 214 lines cut ...]
  #3  0x40116b1e:0x3ffb6060 in tcp_update_rcv_ann_wnd at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/tcp.c:951 
      (inlined by) tcp_update_rcv_ann_wnd at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/tcp.c:931
  #4  0x40116bcc:0x3ffb6080 in tcp_recved at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/tcp.c:991
  #5  0x401827b6:0x3ffb60a0 in _tcp_recved_api(tcpip_api_call_data*) at .pio/libdeps/esp-wrover-kit/AsyncTCP/src/AsyncTCP.cpp:419
  #6  0x40113458:0x3ffb60c0 in tcpip_thread_handle_msg at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/tcpip.c:172
      (inlined by) tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/tcpip.c:154

ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46
```

### Other Steps to Reproduce

Just use the captive portal example.

There are a variety of crashes including mine.

There is `assert failed: tcp_update_rcv_ann_wnd IDF/components/lwip/lwip/src/core/tcp.c:951 (new_rcv_ann_wnd <= 0xffff)`
But mostly `malloc` and `heap_caps_free errors`.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-10-14:
can you try this with PSRAM disabled please?

zekageri, 2022-10-14:
I will try if ASAP. I must tweak my JsonDocuments a little bit in order for it to work.

zekageri, 2022-10-17:
PSRAM disabled error:

```cpp
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
[     4][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[FS] - File system mounted.
[  1096][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
[  1186][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring SoftAP static IP: 8.8.8.8, MASK: 255.255.255.0, GW: 8.8.8.8
[  1186][V][WiFiGeneric.cpp:390] _arduino_event_cb(): AP Started
[  1192][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 10 - AP_START
[  1194][V][WiFiGeneric.cpp:143] set_esp_interface_ip(): SoftAP: 8.8.8.8 | Gateway: 8.8.8.8 | DHCP Start: 0.0.0.0 | Netmask: 255.255.255.0
[  1211][V][WiFiGeneric.cpp:190] set_esp_interface_ip(): DHCP Server Range: 8.8.8.9 to 8.8.8.19
[  1222][V][WiFiGeneric.cpp:393] _arduino_event_cb(): AP Stopped
[  1229][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 11 - AP_STOP
[  1229][V][WiFiGeneric.cpp:390] _arduino_event_cb(): AP Started
[FIRST_BOOT] - Ap started. IP: 8.8.8.8
[  1238][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 10 - AP_START
[  1249][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (2593) gpio: gpio_set_level(226): GPIO output gpio_num error
[  1260][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
[  1266][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (2611) gpio: gpio_set_level(226): GPIO output gpio_num error
[  1278][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
[ 13964][V][WiFiGeneric.cpp:405] _arduino_event_cb(): AP Station Connected: MAC: f2:06:27:71:d0:02, AID: 1
[ 13964][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 12 - AP_STACONNECTED
[ 14168][V][WiFiGeneric.cpp:419] _arduino_event_cb(): AP Station IP Assigned:8.8.8.9
[ 14169][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 14 - AP_STAIPASSIGNED
[ 14335][E][vfs_api.cpp:104] open(): /littlefs/public/generate_204 does not exist, no permits for creation
[ 14340][E][vfs_api.cpp:104] open(): /littlefs/public/generate_204.gz does not exist, no permits for creation
[ 14348][E][vfs_api.cpp:104] open(): /littlefs/public/generate_204/index.htm does not exist, no permits for creation
[ 14360][E][vfs_api.cpp:104] open(): /littlefs/public/generate_204/index.htm.gz does not exist, no permits for creation
[ 14459][E][vfs_api.cpp:104] open(): /littlefs/public/mobile/status.php does not exist, no permits for creation
[ 14464][E][vfs_api.cpp:104] open(): /littlefs/public/mobile/status.php.gz does not exist, no permits for creation
[ 14473][E][vfs_api.cpp:104] open(): /littlefs/public/mobile/status.php/index.htm does not exist, no permits for creation
[ 14483][E][vfs_api.cpp:104] open(): /littlefs/public/mobile/status.php/index.htm.gz does not exist, no permits for creation
[ 14583][E][vfs_api.cpp:104] open(): /littlefs/public/mobile/status.php does not exist, no permits for creation
[ 14589][E][vfs_api.cpp:104] open(): /littlefs/public/mobile/status.php.gz does not exist, no permits for creation
[ 14598][E][vfs_api.cpp:104] open(): /littlefs/public/mobile/status.php/index.htm does not exist, no permits for creation
[ 14608][E][vfs_api.cpp:104] open(): /littlefs/public/mobile/status.php/index.htm.gz does not exist, no permits for creation

assert failed: tcp_update_rcv_ann_wnd IDF/components/lwip/lwip/src/core/tcp.c:951 (new_rcv_ann_wnd <= 0xffff)

Backtrace:0x40083b41:0x3ffb5b900x4008d9e1:0x3ffb5bb0 0x400931dd:0x3ffb5bd0 0x40114a56:0x3ffb5d00 0x40114b04:0xC:\Users\zekag\.platformio\packages\toolchain-xtensa-esp32@8.4.0+2021r2-patch3\bin\xtensa-esp32-elf-addr2line.exe: 'C:\Users\zekag\OneDrive\Dokumentumok\PlatformIO\Projects\HsH_OOP\.pio\build\esp-wrover-kit\firmware.elf': No such file
Esp32ExceptionDecoder: failed to call C:\Users\zekag\.platformio\packages\toolchain-xtensa-esp32@8.4.0+2021r2-patch3\bin\xtensa-esp32-elf-addr2line.exe: Command '['C:\\Users\\zekag\\.platformio\\packages\\toolchain-xtensa-esp32@8.4.0+2021r2-patch3\\bin\\xtensa-esp32-elf-addr2line.exe', '-fipC', '-e', 'C:\\Users\\zekag\\OneDrive\\Dokumentumok\\PlatformIO\\Projects\\HsH_OOP\\.pio\\build\\esp-wrover-kit\\firmware.elf', '0x40083b41:0x3ffb5b90']' returned non-zero exit status 1.
3ffb5d20 0x4018018a:0x3ffb5d40 0x4010f438:0x3ffb5d60

ELF file SHA256: 0000000000000000
```

zekageri, 2022-10-17:
ONE crash out of 30 connection without PSRAM so far.

zekageri, 2022-10-18:
I'm ready to test this further if you have any idea.

me-no-dev, 2022-10-21:
@david-cermak do you have any ideas what could cause this error?

zekageri, 2022-10-26:
This usse is still there. I get different crash results on almost every connection.

```cpp
E (24505) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (24505) task_wdt:  - async_tcp (CPU 0/1)
E (24505) task_wdt: Tasks currently running:
E (24505) task_wdt: CPU 0: IDLE
E (24505) task_wdt: CPU 1: IDLE
E (24505) task_wdt: Aborting.

abort() was called at PC 0x400fdcec on core 0

Backtrace:0x40083cbd:0x3ffbeafc |<-CORRUPTED

  #0  0x40083cbd:0x3ffbeafc in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402

ELF file SHA256: 0000000000000000
```

**Other one:**

```cpp
[FIRST_BOOT] - Ap started. IP: 8.8.8.8
E (4147) gpio: gpio_set_level(226): GPIO output gpio_num error
E (4148) gpio: gpio_set_level(226): GPIO output gpio_num error
CORRUPT HEAP: Bad tail at 0x3ffd45fc. Expected 0xbaad5678 got 0x00000000

assert failed: multi_heap_free multi_heap_poisoning.c:253 (head != NULL)

Backtrace:0x40083cbd:0x3ffd29500x4008ece5:0x3ffd2970 0x40094509:0x3ffd2990 0x40094133:0x3ffd2ac0 0x40084151:0x3ffd2ae0 0x40094539:0x3ffd2b00 0x4017c8f5:0x3ffd2b20 0x400e05d1:0x3ffd2b40 0x400e28a3:0x3ffd2b60 0x400e31c5:0x3ffd2b80 0x400e5679:0x3ffd2ba0 0x400e0823:0x3ffd2bc0 0x400e082f:0x3ffd2be0 0x401805f9:0x3ffd2c00 0x401808e2:0x3ffd2c30 0x401808f8:0x3ffd2c50 0x40180a1a:0x3ffd2c70

  #0  0x40083cbd:0x3ffd2950 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402
  #1  0x4008ece5:0x3ffd2970 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:128
  #2  0x40094509:0x3ffd2990 in __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
  #3  0x40094133:0x3ffd2ac0 in multi_heap_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/multi_heap_poisoning.c:253
      (inlined by) multi_heap_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/multi_heap_poisoning.c:245
  #4  0x40084151:0x3ffd2ae0 in heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/heap_caps.c:340
  #5  0x40094539:0x3ffd2b00 in free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/heap.c:39
  #6  0x4017c8f5:0x3ffd2b20 in operator delete(void*) at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/del_op.cc:49
  #7  0x400e05d1:0x3ffd2b40 in std::_Function_handler<void (AsyncWebHeader* const&), AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*)::{lambda(AsyncWebHeader*)#1}>::_M_invoke(std::_Any_data const&, AsyncWebHeader* const&) at lib/ESPAsyncWebServer/src/WebRequest.cpp:56
      (inlined by) _M_invoke at c:\users\pc\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:297
  #8  0x400e28a3:0x3ffd2b60 in std::function<void (AsyncWebHeader* const&)>::operator()(AsyncWebHeader* const&) const at c:\users\pc\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:687
  #9  0x400e31c5:0x3ffd2b80 in LinkedList<AsyncWebHeader*, LinkedListNode>::free() at 
lib/ESPAsyncWebServer/src/StringArray.h:175
      (inlined by) AsyncWebServerRequest::~AsyncWebServerRequest() at lib/ESPAsyncWebServer/src/WebRequest.cpp:81
  #10 0x400e5679:0x3ffd2ba0 in AsyncWebServer::_handleDisconnect(AsyncWebServerRequest*) at lib/ESPAsyncWebServer/src/WebServer.cpp:102 (discriminator 1)
  #11 0x400e0823:0x3ffd2bc0 in AsyncWebServerRequest::_onDisconnect() at lib/ESPAsyncWebServer/src/WebRequest.cpp:229
  #12 0x400e082f:0x3ffd2be0 in std::_Function_handler<void (void*, AsyncClient*), AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*)::{lambda(void*, AsyncClient*)#6}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&) at lib/ESPAsyncWebServer/src/WebRequest.cpp:74
      (inlined by) _M_invoke at c:\users\pc\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:297
  #13 0x401805f9:0x3ffd2c00 in std::function<void (void*, AsyncClient*)>::operator()(void*, AsyncClient*) const at c:\users\pc\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:687
  #14 0x401808e2:0x3ffd2c30 in AsyncClient::_fin(tcp_pcb*, signed char) at .pio/libdeps/esp-wrover-kit/AsyncTCP/src/AsyncTCP.cpp:888
      (inlined by) AsyncClient::_fin(tcp_pcb*, signed char) at .pio/libdeps/esp-wrover-kit/AsyncTCP/src/AsyncTCP.cpp:885
  #15 0x401808f8:0x3ffd2c50 in AsyncClient::_s_fin(void*, tcp_pcb*, signed char) at .pio/libdeps/esp-wrover-kit/AsyncTCP/src/AsyncTCP.cpp:1195
  #16 0x40180a1a:0x3ffd2c70 in _async_service_task(void*) at .pio/libdeps/esp-wrover-kit/AsyncTCP/src/AsyncTCP.cpp:162
      (inlined by) _async_service_task at .pio/libdeps/esp-wrover-kit/AsyncTCP/src/AsyncTCP.cpp:194

ELF file SHA256: 0000000000000000

Rebooting...
```

Firstboot loop still has a `vTaskDelay(1)` in it to let run other tasks.

dr3adx, 2023-02-23:
@zekageri did you find any solution bro? this bug is crucial

zekageri, 2023-02-23:
I think the Async dns solved this problem.

zekageri, 2023-02-23:
#4222

dr3adx, 2023-02-23:
@zekageri bro, unrelated to this issue, you got any idea why this error happens? assert failed: tcp_update_rcv_ann_wn
I'm using AsyncHTTPRequest_Generic lib and some requests fail with that error

closed 2024-02-13 by zekageri