espressif/arduino-esp32 #7355: WiFiClientSecure / WiFiClient inconsistency in get/set timeout

cziter15, 2022-10-14:
### Board

N/A

### Device Description

N/A

### Hardware Configuration

N/A

### Version

latest master (checkout manually)

### IDE Name

N/A

### Operating System

N/A

### Flash frequency

N/A

### PSRAM enabled

no

### Upload speed

N/A

### Description

This is inconsistent behaviour:
- setTimeout uses seconds as parameter while
- getTimeout returns number of miliiseconds

### Sketch

```cpp
WiFiClientSecure s;
s.setTimeout(10); // <= this overrides virtual method and sets _timeout to passed value multiplied by 1000
int gv = s.getTimeout();  // <= comes from Stream due to derived chain of classes and returns _timeout

Serial.println(gv); //will print 10000
```

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-10-17:
Hello @cziter15,

this is already discussed in this Pull Request: https://github.com/espressif/arduino-esp32/pull/6676

It's planned for next major release as it's a breaking change.

cziter15, 2022-10-17:
Closing issue then.

VojtechBartoska, 2024-02-20:
PR got merged, closing this as solved.

closed 2024-02-20 by VojtechBartoska