espressif/arduino-esp32 #7356: WiFiClientSecure don't respect timeout in ssl_send

cziter15, 2022-10-14:
### Board

N/A

### Device Description

N/A

### Hardware Configuration

N/A

### Version

latest master (checkout manually)

### IDE Name

N/A

### Operating System

N/A

### Flash frequency

N/A

### PSRAM enabled

no

### Upload speed

N/A

### Description

Function send_ssl_data, called by WiFiClientSecure doesn't respect timeout set on socket.

After disconnecting the network, not WiFi directly and having some small piece of code sending periodically few messages to MQTT broker via SSL, I've encountered freeze inside send_ssl_data function.

After some investigation, it appears to hang inside while loop.

Temporary working solution, but 5000 is a magic constant:

```
int send_ssl_data(sslclient_context *ssl_client, const uint8_t *data, size_t len)
{
    log_v("Writing HTTP request with %d [to %d] bytes...", len, ssl_client->handshake_timeout); //for low level debug
    int ret = -1;

    unsigned long send_start_time=millis();
    while ((ret = mbedtls_ssl_write(&ssl_client->ssl_ctx, data, len)) <= 0) {
        if((millis()-send_start_time)>5000)
            return -1;

        if (ret != MBEDTLS_ERR_SSL_WANT_READ && ret != MBEDTLS_ERR_SSL_WANT_WRITE && ret < 0) {
            log_v("Handling error %d", ret); //for low level debug
            return handle_error(ret);
        }
        //wait for space to become available
        vTaskDelay(2);
    }

    return ret;
}
```

### Sketch

```cpp
N/A
```

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

cziter15, 2022-10-18:
I've submitted PR which adds socket_timeout to ssl_context and copies timeout value there. Then this socket_timeout variable is used in send_ssl_data as timeout.

VojtechBartoska, 2022-10-21:
@cziter15 Thank you for contribution, we will review your PR soon.

TD-er, 2022-11-07:
Could the timeout (also) be related to this one?
https://github.com/espressif/arduino-esp32/pull/6676

cziter15, 2022-11-09:
This timeout here is something specific, not related to ms vs second interpretation.

dsilletti, 2022-11-20:
@cziter15 could you please provide your fix (socket_timeout)?  This is probably the reason why my ESP32 remain stuck if I disconnect the WAN interface of my Wi-Fi router, while connected via Wi-Fi to the MQTT broker (using PubSubClient  with WiFiClientSecure and publishing every second).

The PubSubClient setSocketTimeout does not help.

Thanks

dsilletti, 2022-11-20:
temporary fixed limiting the loop to 100 retries, finally the MQTT get disconnected and reconnects properly when the WAN is back.
The timer based fix you used (I tried with 3000ms) was not working properly for be, the MQTT was disconnecting but not reconnecting.

```
int send_ssl_data(sslclient_context *ssl_client, const uint8_t *data, size_t len)
{
    log_v("Writing HTTP request with %d bytes...", len); //for low level debug
    int ret = -1;

    int retry=0; 

    while ((ret = mbedtls_ssl_write(&ssl_client->ssl_ctx, data, len)) <= 0) {
        
        if (retry>100) {
            return -1;
        }

        if (ret != MBEDTLS_ERR_SSL_WANT_READ && ret != MBEDTLS_ERR_SSL_WANT_WRITE && ret < 0) {
            log_v("Handling error %d", ret); //for low level debug
            return handle_error(ret);
        }
        //wait for space to become available
        vTaskDelay(2);
       
        retry = retry +1;
    }

    return ret;
}
```

cziter15, 2022-11-20:
This is strange, but to be clear about my project:
- I'm setting timeout directly on the WiFiClientSecure, not touching
PubSubClient to 5 sec
- I'm setting handshake timeout on WiFiClientSecure to 5 sec
- The fix on send ssl data is applied to my arduino-esp32 fork

Using retry conunt is somehow strange. Maybe 3000ms was too low for your
case? I think you should investigate, for example by adding logs. ;)

dsilletti, 2022-11-20:
Thanks @cziter15, I had to tune my timers, it works now properly also configuring the timeout in seconds, according to your code.

It should definitely be fixed, probably using the value configured with:

```
int WiFiClientSecure::setTimeout(uint32_t seconds)
```

because it's not a timeout in the connection but while sending the data.

@VojtechBartoska could you please prioritize this? All ESP32 devices using WiFiClientSecure end up in an infinite while loop if they lose internet connectivity while sending data (if Wi-Fi stays connected). Thanks

cziter15, 2022-11-20:
Yup, I can confirm that behavior. Device is geting stuck inside task and it's not triggering. Network stack is working as you can still ping the device. Afaik, my PR has been reviewed by @me-no-dev, but @SuGlider has still to review it. Anyway, handshake timeout might be probably tweaked as now it has really huge timeout.

dsilletti, 2022-11-20:
my workaround to have a timeout configurable in the sketch using the handshake timeout:

set in the sketch the handshaketimeout to a reasonable value:
```
espClientSecure.setTimeout(3);
espClientSecure.setHandshakeTimeout(3);
```

 then add the timeout in ssl_client.cpp in the send_ssl_data() function like this:

```
int send_ssl_data(sslclient_context *ssl_client, const uint8_t *data, size_t len)
{
    log_v("Writing HTTP request with %d bytes...", len); //for low level debug
    int ret = -1;

    unsigned long send_start_time=millis(); //start time sending data
    while ((ret = mbedtls_ssl_write(&ssl_client->ssl_ctx, data, len)) <= 0) {   
        if (ret != MBEDTLS_ERR_SSL_WANT_READ && ret != MBEDTLS_ERR_SSL_WANT_WRITE && ret < 0) {
            log_v("Handling error %d", ret); //for low level debug
            return handle_error(ret);
        }
        if ((millis()-send_start_time)>ssl_client->handshake_timeout) //timeout configured to handshake timeout
            return -1;
        //wait for space to become available
        vTaskDelay(2);
    }

    return ret;
}
```

dsilletti, 2022-11-20:
Sure, it's just a temporary workaround.

cziter15, 2022-11-20:
Check out my fork, I'll keep it up to date until this issue is fixed on master of arduino-esp32.

There's also another fix for getHostByName function which checks if passed string is an IP address and instead of triggering dns flow in that case it just converts passed string ip to IPAddress structure.

https://github.com/cziter15/arduino-esp32

dsilletti, 2022-11-20:
cool, thanks!

cziter15, 2022-12-30:
Closing issue. Fixed in 2.0.6 release.

closed 2022-12-30 by cziter15