espressif/arduino-esp32 #7359: Serial.end() does not disconnect the Serial monitor

fanfanlatulipe26, 2022-10-15:
### Board

ESP-C3-32S-Kit  and ESP32 Dev Kit 

### Device Description

ESP32 C3 board (or std ESP32)

### Hardware Configuration

Nothing

### Version

v2.0.5

### IDE Name

Arduino IDE 1.8.19

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

Need to use UART0 after a while, on pins others than the default pins used by uart0, the USB adaptor/Serial monitor.
Even after a Serial.end()  and a new begin assigning the new pins to use by the newSerial link, the output are still going to the Serial Monitor.

A logic analyzer show that the correct bit stream is also generated on the new output pin New_TX
There is also a lot of pulses on New_RX, even with a pull up resistor

I have the same behavior on a ESP32 Dev Kit v1  but here the New_RX signal is clean.

ESP32C3 only have 2 UARTs and re-assigning UART0 is a must.

### Sketch

```cpp
#define New_RX 19   //
#define New_TX 18   //
//#define New_TX 21   //   U0TX ESP32C3
//#define New_RX 20   //   U0RX ESP32C3
HardwareSerial NewSerial(0);
void setup() {
  Serial.begin(115200);
  delay(2000);
  Serial.println("\n\nStarting on Serial, default pins, and ending Serial");
  Serial.flush();
  Serial.end();
  //delay(2000);
  NewSerial.begin(115200, SERIAL_8N1, New_RX, New_TX);
  //delay(2000);
  NewSerial.println("\n\n...now printing on NewSerial, others pins .. " );
  NewSerial.println(" but the message appears also on Serial Monitor ... " );
  NewSerial.flush();
  Serial.println("\n\n printing on Serial");  // No output: OK serial is closed
  Serial.flush();
  NewSerial.println("\n\n and once more printing on NewSerial");
  NewSerial.flush();
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
Starting on Serial, default pins, and ending Serial

...now printing on NewSerial, others pins .. 
 but the message appears also on Serial Monitor ... 

 and once more printing on NewSerial
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-10-26:
@fanfanlatulipe26 - After checking the issue, it seems to be related to the way that IDF set the pins using IOMUX and then it doesn't disconnect them from the IO pads. 

I'm currently working on an UART detach function to fix it and then submit a PR.

SuGlider, 2022-10-26:
Although it is possible to see UART0 output in the default TX pin, when this pin is assigned to any other peripheral, it works fine.
Therefore, it doesn't prevent the default RX/TX UART0 pins from being used for any other functionality.

@fanfanlatulipe26  - please check the PR and, if possible, test it to ensure that this issue is solved.

fanfanlatulipe26, 2022-10-27:
@SuGlider  I don't know how to install your branch [uart_detach_pins ](https://github.com/SuGlider/arduino-esp32/tree/uart_detach_pins )instead of my standard 2.0.5. I ended up copying your new HardwareSerial and esp32-hal_uart in my local 2.0.5 tree
I ran tests for ESP32 and ESP32C3 but only tested that the RX/X UART0 can then be used as output.
Good results.
Also no more "noise" on the NEW_RX line.

fanfanlatulipe26, 2022-10-28:
When idle or not used the pins are HIGH. The default TX/UART0 is HIGH when there is no serial transmission but after Seria.lend(), the line is set LOW.
 I don't know if it matters

SuGlider, 2022-10-31:
> When idle or not used the pins are HIGH. The default TX/UART0 is HIGH when there is no serial transmission but after Seria.lend(), the line is set LOW. I don't know if it matters

It really doesn't matter because after `Serial.end()`, previous RX/TX pins are not used any more by the UART. 
But it can then be assigned to any other function, such as Digital IO, and then be set to high or low as the application may need.

Ugobyte, 2022-11-01:
I have a question similar to this topic.

The ESP32-S3 has two Micro-USB ports. One is labeled 'UART", the other is labeled "USB". I'm currently using the "USB' port to send HID Mouse/Keyboard data. Is it possible to also use this same "USB" port to send out serial data?

me-no-dev, 2022-11-02:
@Ugobyte yes it is. You need to either enable "CDC On Boot" in the board menu, or start the USBSerial yourself: https://github.com/espressif/arduino-esp32/blob/master/libraries/USB/examples/USBSerial/USBSerial.ino

closed 2022-11-02 by me-no-dev