espressif/arduino-esp32 #7360: Please replace BLE_iBeacon with a more fleshed-out example.

Human, 2022-10-15:
### Related area

iBeacon example

### Hardware specification

ESP32

### Is your feature request related to a problem?

Forum and github searches reveal that many people have problems getting ESP32 boards running https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/examples/BLE_iBeacon/BLE_iBeacon.ino to be detected by strict iBeacon scanners. Even when a scanner does detect an ESP32 iBeacon, many fields are missing. I came across a discussion [here](https://github.com/nkolban/esp32-snippets/issues/385#issuecomment-362535434) that links to reworked example code [here](https://pastebin.com/0Zh07Zzn). This worked out of the box for me, whereas BLE_iBeacon.ino only worked partially on some scanners and not at all on others.

### Describe the solution you'd like

Either _replace_ the legacy BLE_iBeacon.ino with [31337Ghost](https://github.com/31337Ghost)'s reference code, or incorporate its changes into the legacy BLE_iBeacon.ino.

### Describe alternatives you've considered

I searched github and forums, and it seems as if most people end up having to modify BLE_iBeacon.ino to achieve basic functions. I don't think there's really an alternative, other than everyone performing the same forum searches and performing the same workarounds.

### Additional context

I recommend that devs try the legacy BLE_iBeacon code, then switch to the suggested new reference code, and compare the functionality in various scanners.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2022-10-15:
Make a pull request with the example code that you would like to see.  The BLE library was originally a community submission, and it continues to be supported largely by the community rather than Espressif.
Also, the code at https://github.com/h2zero/NimBLE-Arduino/tree/release/1.4/examples might have some bits that would be useful in your example.

Human, 2022-10-16:
@31337Ghost I don't want to steal credit for your work, but I also don't want to give you things to do :) Would you prefer to submit a PR for this, or should I do it?

VojtechBartoska, 2022-10-21:
@PilnyTomas Can you please check this example and propose some possible solutions? Thank you.

PilnyTomas, 2022-11-09:
@Human Open Source software is not only about transparency and free access, it is also about community contribution. If you have a clear idea about the example and you are able to do that we will happily review your Pull Request and add it to this repository.

Human, 2022-11-13:
@PilnyTomas Ok. I gave the original author time and opportunity to respond, so I will prepare a PR with his work.

PR #7470 (2022-12-21): replaced original BLE_iBeacon.ino example with a known-good one
## Description of Change
I had no luck getting the original `BLE_iBeacon.ino` code to work as expected with iBeacon scanners, but the included reference code worked completely. Thanks to @31337Ghost for writing it! I am simply submitting the PR.
## Tests scenarios
Tested on DOIT ESP32 DEVKIT V1, using various scanner client apps on Android devices.

## Related links
Closes #7360
closed 2022-12-21 by P-R-O-C-H-Y