espressif/arduino-esp32 #7363: esp_crt_bundle.h & esp_http_client.h --> "Failed to attach bundle"

Duster98, 2022-10-16:
### Board

Esp32

### Device Description

I have this esp32 dev kit board

[image]

### Hardware Configuration

No GPIO connected.

### Version

v2.0.5

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

240 Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

Hi.
I was trying to connect the esp32 module to the googleapis server via the libraries _esp_http_client.h_ and _esp_crt_bundle.h_ but unfortunately the logs tell me that it is not possible to attach the bundle or that it fails during the operation.

Any ideas to fix this?

### Sketch

```cpp
#include <Arduino.h>
#include <WiFi.h>
#include <esp_crt_bundle.h>
#include <esp_http_client.h>

#define MAX_HTTP_OUTPUT_BUFFER 4096
#define MAX_HTTP_TX_BUFFER 2048
#define GoogleAuthUri "https://identitytoolkit.googleapis.com"
#define FIREBASE_API_KEY "xxxxxxxxxxxxxxxxxxxxxxxxxxx"
const char *FIREBASE_USERNAME = "xxxxxxxxxxxxxxxxxxxxxxxxxxx";
const char *FIREBASE_PASSWORD = "xxxxxxxxxxxxxxxxxxxxxxxxxxx";

void http_client_request(String Url, esp_http_client_method_t Method, String Payload = "")
{
  Serial.print("Staring sample Https client request... ");
  char output_buffer[MAX_HTTP_OUTPUT_BUFFER] = {0};
  int _content_length = 0;
  esp_http_client_config_t config{};
  config.url = Url.c_str();
  config.crt_bundle_attach = esp_crt_bundle_attach;
  config.buffer_size = MAX_HTTP_OUTPUT_BUFFER;
  config.buffer_size_tx = MAX_HTTP_TX_BUFFER;
  esp_http_client_handle_t HttpsClient = esp_http_client_init(&config);
  Serial.println("Done!");
  esp_http_client_set_method(HttpsClient, Method);
  esp_http_client_set_header(HttpsClient, "Content-Type", "application/json");
  Serial.print("Opening... ");
  esp_err_t err = esp_http_client_open(HttpsClient, Payload.length());
  if (err != ESP_OK)
    Serial.println("Failed to open HTTP connection: " + String(esp_err_to_name(err)));
  else
  {
    Serial.println("Writting...");
    int wlen = esp_http_client_write(HttpsClient, Payload.c_str(), Payload.length());
    if (wlen < 0)
      Serial.println("Write failed");
    _content_length = esp_http_client_fetch_headers(HttpsClient);
    Serial.printf("Content lenght: %d", _content_length);
    if (_content_length < 0)
      Serial.println("HTTP HttpsClient fetch headers failed");
    else
    {
      int data_read = esp_http_client_read_response(HttpsClient, output_buffer, MAX_HTTP_OUTPUT_BUFFER);
      if (data_read >= 0)
      {
        Serial.printf("HTTP POST Status = %d, content_length = %d", esp_http_client_get_status_code(HttpsClient), esp_http_client_get_content_length(HttpsClient));
        Serial.println(output_buffer);
      }
      else
        Serial.println("Failed to read response");
    }
  }

  esp_http_client_close(HttpsClient);
  esp_http_client_cleanup(HttpsClient);
}

void setup()
{
  Serial.begin(115200);
  Serial.println("\nEsp32 Https Test");
  Serial.println("Available Ram: " + String(ESP.getFreeHeap()) + " byte");
  WiFi.hostname("EspHttps");
  WiFi.mode(WIFI_STA);
  Serial.print("Connecting to WiFi... ");
  WiFi.begin("xxxxxxxxxxxxxxxxxxxxxxxxxxx", "xxxxxxxxxxxxxxxxxxxxxxxxxxx");
  uint8_t StatusWifi = WiFi.waitForConnectResult();
  if (StatusWifi == WL_CONNECTED)
  {
    Serial.println("Connected!");
    String FireBaseAuthReq = "{\"email\":\"" + String(FIREBASE_USERNAME) + "\",\"password\":\"" + String(FIREBASE_PASSWORD) + "\",\"returnSecureToken\": true}";
    http_client_request(String(GoogleAuthUri) + "/v1/accounts:signInWithPassword?key=" + String(FIREBASE_API_KEY), HTTP_METHOD_POST, FireBaseAuthReq);
  }
  else
    Serial.println("Fail!");

  Serial.println("Available Ram: " + String(ESP.getFreeHeap()) + " byte");
}

void loop()
{
}
```

### Debug Message

```plain
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4

Esp32 Https Test
Available Ram: 304584 byte
Connecting to WiFi... Connected!
Staring sample Https client request... Done!
Opening... [   559][E][esp_crt_bundle.c:184] esp_crt_bundle_attach(): Failed to attach bundle
E (716) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7680
E (717) esp-tls: Failed to open new connection
E (717) TRANSPORT_BASE: Failed to open a new connection
E (724) HTTP_CLIENT: Connection failed, sock < 0
Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
Available Ram: 256220 byte
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

atanisoft, 2022-10-16:
You haven't told the esp_http_client what CA bundle to use and it is defaulting to an empty bundle. That is what the error code indicates. You should call:
```
esp_err_t esp_crt_bundle_set(const uint8_t *x509_bundle, size_t bundle_size);
```
to configure the CA chain or you can pass the certificate pem file as a constant in esp_http_client_config_t.cert_pem and not set crt_bundle_attach.

Duster98, 2022-10-17:
> You haven't told the esp_http_client what CA bundle to use and it is defaulting to an empty bundle. That is what the error code indicates. You should call:
> 
> ```
> esp_err_t esp_crt_bundle_set(const uint8_t *x509_bundle, size_t bundle_size);
> ```
> 
> to configure the CA chain or you can pass the certificate pem file as a constant in esp_http_client_config_t.cert_pem and not set crt_bundle_attach.
> 
> Or you could use this example: https://github.com/espressif/arduino-esp32/blob/master/libraries/HTTPClient/examples/BasicHttpsClient/BasicHttpsClient.ino

Hi, thanks for your reply,

I believe that this function works only with esp-idf framework not with arduino framework.
I found the corresponding function in the file **_esp_crt_bundle.c_** in the folder of **_WiFiClientSecure.h_** library
```c++
void esp_crt_bundle_set(const uint8_t *x509_bundle)
```
however I added the "cert_pem"
```c++
extern const char server_googleapis_root_cert_pem_start[] asm("_binary_src_server_googleapis_root_cert_pem_start");
extern const char server_googleapis_root_cert_pem_end[] asm("_binary_src_server_googleapis_root_cert_pem_end");
```
and pass it as a constant in esp_http_client_config_t.cert_pem, i have also pass the len
```c++
  config.cert_pem = server_googleapis_root_cert_pem_start;
  config.cert_len = server_googleapis_root_cert_pem_end - server_googleapis_root_cert_pem_start;
```
so finally the code is:
```c++
[...]
const char *FIREBASE_USERNAME = "xxxxxxxxxxxxxxxxxxxxxxxxxxx";
const char *FIREBASE_PASSWORD = "xxxxxxxxxxxxxxxxxxxxxxxxxxx";

extern const char server_googleapis_root_cert_pem_start[] asm("_binary_src_server_googleapis_root_cert_pem_start"); // <-- ADDED
extern const char server_googleapis_root_cert_pem_end[] asm("_binary_src_server_googleapis_root_cert_pem_end"); // <-- ADDED

void http_client_request(String Url, esp_http_client_method_t Method, String Payload = "")
{
  Serial.print("Staring sample Https client request... ");
  char output_buffer[MAX_HTTP_OUTPUT_BUFFER] = {0};
  int _content_length = 0;
  esp_http_client_config_t config{};
  config.url = Url.c_str();
  config.cert_pem = server_googleapis_root_cert_pem_start; // <-- ADDED
  config.cert_len = server_googleapis_root_cert_pem_end - server_googleapis_root_cert_pem_start; // <-- ADDED
  config.buffer_size = MAX_HTTP_OUTPUT_BUFFER;
  config.buffer_size_tx = MAX_HTTP_TX_BUFFER;
  esp_http_client_handle_t HttpsClient = esp_http_client_init(&config);
  Serial.println("Done!");
[...]
```

but the esp crash and print 2 type of different report on serial monitor:

### 1 REPORT
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4

Esp32 Https Test
Available Ram: 304792 byte
Connecting to WiFi... Connected!
Staring sample Https client request... Done!
Opening... Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception).
[... 62 lines cut ...]
  #13 0x40113c40:0x3ffb12a0 in ecdsa_verify_restartable at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/mbedtls/mbedtls/library/ecdsa.c:669
  #14 0x40113f96:0x3ffb1340 in mbedtls_ecdsa_read_signature_restartable at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/mbedtls/mbedtls/library/ecdsa.c:891
  #15 0x40113fbe:0x3ffb1380 in mbedtls_ecdsa_read_signature at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/mbedtls/mbedtls/library/ecdsa.c:840
  #16 0x40118dd1:0x3ffb13a0 in ecdsa_verify_wrap at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/mbedtls/mbedtls/library/pk_wrap.c:629
  #17 0x40118e03:0x3ffb13c0 in eckey_verify_wrap at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/mbedtls/mbedtls/library/pk_wrap.c:252
  #18 0x401189fa:0x3ffb1490 in mbedtls_pk_verify_restartable at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/mbedtls/mbedtls/library/pk.c:319
  #19 0x4010d629:0x3ffb14c0 in ssl_parse_server_key_exchange at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/mbedtls/mbedtls/library/ssl_cli.c:3457
      (inlined by) mbedtls_ssl_handshake_client_step at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/mbedtls/mbedtls/library/ssl_cli.c:4536
  #20 0x4010b35d:0x3ffb1560 in mbedtls_ssl_handshake_step at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/mbedtls/mbedtls/library/ssl_tls.c:5872
  #21 0x4010b397:0x3ffb1580 in mbedtls_ssl_handshake at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/mbedtls/mbedtls/library/ssl_tls.c:5909
  #22 0x40145e39:0x3ffb15a0 in esp_mbedtls_handshake at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp-tls/esp_tls_mbedtls.c:176
  #23 0x401459e9:0x3ffb15c0 in esp_tls_handshake at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp-tls/esp_tls.c:80 (discriminator 15)
      (inlined by) esp_tls_low_level_conn at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp-tls/esp_tls.c:427 (discriminator 15)
  #24 0x40145ad3:0x3ffb1600 in esp_tls_conn_new_sync at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp-tls/esp_tls.c:489
  #25 0x401078e0:0x3ffb1630 in ssl_connect at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/tcp_transport/transport_ssl.c:111
[... cut ...]
  #30 0x400d2dfd:0x3ffb2780 in setup() at src/main.cpp:83 (discriminator 13)
  #31 0x400d619e:0x3ffb2820 in loopTask(void*) at C:/Users/giuse/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42
```

I have already test this code [BasicHttpsClient.ino](https://github.com/espressif/arduino-esp32/blob/master/libraries/HTTPClient/examples/BasicHttpsClient/BasicHttpsClient.ino) and it works fine... but the problem is that use a lot of ram, so I was looking for alternative methods.

atanisoft, 2022-10-17:
Both of your crashes appear to be due to exceeding the stack size for the `loopTask` which defaults to 8kb (rather large). It is not clear what part of your provided code is causing this though.

Using ESP-IDF functions may be best served by using Arduino as ESP-IDF component (which you can configure in your platformio.ini file). Using that you might be able to fine tune various configuration settings to reduce the memory consumption.

Setoboss, 2023-01-04:
Hi Duster98,
i tried your solution but i always get the Error below.
c:/.../.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: C:/.../.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:342: undefined reference to `_binary_src_server_googleapis_root_cert_pem_start'

am i missing a reference/package?

Duster98, 2023-01-05:
Hi Setoboss,

Sorry for the delay.

Now I try to explain how to do for the cert.

1. First of all to get the certificate you need to download and install OpenSSL
2. Run this command on the terminal to get the certificate of any site
    `openssl s_client -showcerts -connect host:443`
    in our case the command will be
    `openssl s_client -showcerts -connect identitytoolkit.googleapis.com:443`
3. I personally chose the 4096 bit certificate which expires in 2028
    here the cert of _googleapis_

    ```
    -----BEGIN CERTIFICATE-----
    MIIFYjCCBEqgAwIBAgIQd70NbNs2+RrqIQ/E8FjTDTANBgkqhkiG9w0BAQsFADBX
    MQswCQYDVQQGEwJCRTEZMBcGA1UEChMQR2xvYmFsU2lnbiBudi1zYTEQMA4GA1UE
    CxMHUm9vdCBDQTEbMBkGA1UEAxMSR2xvYmFsU2lnbiBSb290IENBMB4XDTIwMDYx
    OTAwMDA0MloXDTI4MDEyODAwMDA0MlowRzELMAkGA1UEBhMCVVMxIjAgBgNVBAoT
    GUdvb2dsZSBUcnVzdCBTZXJ2aWNlcyBMTEMxFDASBgNVBAMTC0dUUyBSb290IFIx
    MIICIjANBgkqhkiG9w0BAQEFAAOCAg8AMIICCgKCAgEAthECix7joXebO9y/lD63
    ladAPKH9gvl9MgaCcfb2jH/76Nu8ai6Xl6OMS/kr9rH5zoQdsfnFl97vufKj6bwS
    iV6nqlKr+CMny6SxnGPb15l+8Ape62im9MZaRw1NEDPjTrETo8gYbEvs/AmQ351k
    KSUjB6G00j0uYODP0gmHu81I8E3CwnqIiru6z1kZ1q+PsAewnjHxgsHA3y6mbWwZ
    DrXYfiYaRQM9sHmklCitD38m5agI/pboPGiUU+6DOogrFZYJsuB6jC511pzrp1Zk
    j5ZPaK49l8KEj8C8QMALXL32h7M1bKwYUH+E4EzNktMg6TO8UpmvMrUpsyUqtEj5
    cuHKZPfmghCN6J3Cioj6OGaK/GP5Afl4/Xtcd/p2h/rs37EOeZVXtL0m79YB0esW
    CruOC7XFxYpVq9Os6pFLKcwZpDIlTirxZUTQAs6qzkm06p98g7BAe+dDq6dso499
    iYH6TKX/1Y7DzkvgtdizjkXPdsDtQCv9Uw+wp9U7DbGKogPeMa3Md+pvez7W35Ei
    Eua++tgy/BBjFFFy3l3WFpO9KWgz7zpm7AeKJt8T11dleCfeXkkUAKIAf5qoIbap
    sZWwpbkNFhHax2xIPEDgfg1azVY80ZcFuctL7TlLnMQ/0lUTbiSw1nH69MG6zO0b
    9f6BQdgAmD06yK56mDcYBZUCAwEAAaOCATgwggE0MA4GA1UdDwEB/wQEAwIBhjAP
    BgNVHRMBAf8EBTADAQH/MB0GA1UdDgQWBBTkrysmcRorSCeFL1JmLO/wiRNxPjAf
    BgNVHSMEGDAWgBRge2YaRQ2XyolQL30EzTSo//z9SzBgBggrBgEFBQcBAQRUMFIw
    JQYIKwYBBQUHMAGGGWh0dHA6Ly9vY3NwLnBraS5nb29nL2dzcjEwKQYIKwYBBQUH
    MAKGHWh0dHA6Ly9wa2kuZ29vZy9nc3IxL2dzcjEuY3J0MDIGA1UdHwQrMCkwJ6Al
    oCOGIWh0dHA6Ly9jcmwucGtpLmdvb2cvZ3NyMS9nc3IxLmNybDA7BgNVHSAENDAy
    MAgGBmeBDAECATAIBgZngQwBAgIwDQYLKwYBBAHWeQIFAwIwDQYLKwYBBAHWeQIF
    AwMwDQYJKoZIhvcNAQELBQADggEBADSkHrEoo9C0dhemMXoh6dFSPsjbdBZBiLg9
    NR3t5P+T4Vxfq7vqfM/b5A3Ri1fyJm9bvhdGaJQ3b2t6yMAYN/olUazsaL+yyEn9
    WprKASOshIArAoyZl+tJaox118fessmXn1hIVw41oeQa1v1vg4Fv74zPl6/AhSrw
    9U5pCZEt4Wi4wStz6dTZ/CLANx8LZh1J7QJVj2fhMtfTJr9w4z30Z209fOU0iOMy
    +qduBmpvvYuR7hZL6Dupszfnw0Skfths18dG9ZKb59UhvmaSGZRVbNQpsg3BZlvi
    d0lIKO2d1xozclOzgjXPYovJJIultzkMu34qQb9Sz/yilrbCgj8=
    -----END CERTIFICATE-----
    ```
4. Copy this cert into a pem file, i renamed the file with this name: _server_googleapis_root_cert.pem_
5. Put this file in the _src_ folder

    [image]

6. Add this code in file _platformio.ini_
    `board_build.embed_txtfiles = src/server_googleapis_root_cert.pem`

7. Remember to add this part in your code
    ```c++
    extern const char server_googleapis_root_cert_pem_start[] asm("_binary_src_server_googleapis_root_cert_pem_start");
    extern const char server_googleapis_root_cert_pem_end[] asm("_binary_src_server_googleapis_root_cert_pem_end");
    ```

That's it, finished!
I hope everything works.
Let me know if something goes wrong!

Regards,
Duster

Setoboss, 2023-01-10:
What a great tutorial! Thank you very much!!!