espressif/arduino-esp32 #7364: ESP32 C3/S3 BLE doesn't work properly compared to ESP32 S1

Seref, 2022-10-17:
### Board

ESP32-C3-DevKitM-1

### Device Description

Nothing, just the board itself

### Hardware Configuration

It's just creating a BLE Server like in the arduino examples

### Version

v2.0.5

### IDE Name

PlatformIO

### Operating System

Windows 10 

### Flash frequency

40 Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

It seems like BLE is kind of broken on the ESP32 C3&S3 compared to the ESP32 OG.

When trying to connect to the device, for once only my phone is able to see the ESP32 C3 not my PC even though I have a Bluetooth 5 card.

In addition to that the ESP32 C3 spits out multiple errors and doesn't connect properly.

The errors are all related to the pairing security:```
04:35:18.343 ->E (26465) BT_SMP: smp_calculate_link_key_from_long_term_key failed to update link_key. Sec Mode = 2, sm4 = 0x00
04:35:18.343 -> E (26466) BT_SMP: smp_derive_link_key_from_long_term_key failed
04:35:18.343 -> 
04:35:18.388 -> E (26474) BT_BTM: btm_proc_smp_cback received for unknown device
04:35:18.621 -> E (26724) BT_BTM: BTM_GetSecurityFlags false
04:35:18.621 -> 
04:35:18.668 -> E (26799) BT_BTM: BTM_GetSecurityFlags false
04:35:18.668 -> 
04:35:19.043 -> E (27136) BT_BTM: BTM_GetSecurityFlags false
04:35:19.043 -> 
04:35:23.695 -> E (31824) BT_BTM: BTM_GetSecurityFlags false
04:35:23.695 -> 
04:35:23.883 -> E (31974) BT_BTM: BTM_GetSecurityFlags false
04:35:23.883 -> 
04:35:23.883 -> E (31989) BT_BTM: BTM_GetSecurityFlags false
04:35:23.883 -> 
04:35:23.930 -> E (32049) BT_BTM: BTM_GetSecurityFlags false
04:35:23.930 -> 
04:35:23.977 -> E (32071) BT_BTM: BTM_GetSecurityFlags false
04:35:23.977 -> 
04:35:23.977 -> E (32086) BT_BTM: BTM_GetSecurityFlags false
04:35:23.977 -> 
04:35:24.022 -> E (32131) BT_BTM: BTM_GetSecurityFlags false
04:35:24.022 -> 
04:35:24.067 -> E (32154) BT_BTM: BTM_GetSecurityFlags false
04:35:24.067 -> 
04:35:24.067 -> E (32169) BT_BTM: BTM_GetSecurityFlags false
04:35:24.067 -> 
04:35:24.067 -> E (32184) BT_BTM: BTM_GetSecurityFlags false
04:35:24.067 -> 
04:35:24.114 -> E (32199) BT_BTM: BTM_GetSecurityFlags false
04:35:24.114 -> 
04:35:24.114 -> E (32221) BT_BTM: BTM_GetSecurityFlags false
04:35:24.114 -> 
04:35:24.114 -> E (32236) BT_BTM: BTM_GetSecurityFlags false
04:35:24.114 -> 
04:35:24.162 -> E (32251) BT_BTM: BTM_GetSecurityFlags false
04:35:24.162 -> 
04:35:24.162 -> E (32266) BT_BTM: BTM_GetSecurityFlags false
04:35:24.162 -> 
04:35:24.255 -> E (32341) BT_BTM: BTM_GetSecurityFlags false
04:35:24.255 -> 
04:35:24.255 -> E (32364) BT_BTM: BTM_GetSecurityFlags false
04:35:24.255 -> 
04:35:24.255 -> E (32379) BT_BTM: BTM_GetSecurityFlags false

```

Library like ESP32 BLEKeyboard or ESP32 BLE Gamepad don't work at all on the ESP32 C3 and S3.

### Sketch

```cpp
/*
    Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleServer.cpp
    Ported to Arduino ESP32 by Evandro Copercini
    updates by chegewara
*/

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");

  BLEDevice::init("ESP32 C3");
  BLEServer *pServer = BLEDevice::createServer();
  BLEService *pService = pServer->createService(SERVICE_UUID);
  BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_WRITE
                                       );

  pCharacteristic->setValue("Hello World says Neil");
  pService->start();
  // BLEAdvertising *pAdvertising = pServer->getAdvertising();  // this still is working for backward compatibility
  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->setScanResponse(true);
  pAdvertising->setMinPreferred(0x06);  // functions that help with iPhone connections issue
  pAdvertising->setMinPreferred(0x12);
  BLEDevice::startAdvertising();
  Serial.println("Characteristic defined! Now you can read it in your phone!");
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(2000);
}
```

### Debug Message

```plain
04:35:18.343 ->E (26465) BT_SMP: smp_calculate_link_key_from_long_term_key failed to update link_key. Sec Mode = 2, sm4 = 0x00
04:35:18.343 -> E (26466) BT_SMP: smp_derive_link_key_from_long_term_key failed
04:35:18.343 ->
04:35:18.388 -> E (26474) BT_BTM: btm_proc_smp_cback received for unknown device
04:35:18.621 -> E (26724) BT_BTM: BTM_GetSecurityFlags false
04:35:18.621 ->
04:35:18.668 -> E (26799) BT_BTM: BTM_GetSecurityFlags false
04:35:18.668 ->
04:35:19.043 -> E (27136) BT_BTM: BTM_GetSecurityFlags false
04:35:19.043 ->
04:35:23.695 -> E (31824) BT_BTM: BTM_GetSecurityFlags false
04:35:23.695 ->
04:35:23.883 -> E (31974) BT_BTM: BTM_GetSecurityFlags false
04:35:23.883 ->
04:35:23.883 -> E (31989) BT_BTM: BTM_GetSecurityFlags false
04:35:23.883 ->
04:35:23.930 -> E (32049) BT_BTM: BTM_GetSecurityFlags false
04:35:23.930 ->
04:35:23.977 -> E (32071) BT_BTM: BTM_GetSecurityFlags false
04:35:23.977 ->
04:35:23.977 -> E (32086) BT_BTM: BTM_GetSecurityFlags false
04:35:23.977 ->
04:35:24.022 -> E (32131) BT_BTM: BTM_GetSecurityFlags false
04:35:24.022 ->
04:35:24.067 -> E (32154) BT_BTM: BTM_GetSecurityFlags false
04:35:24.067 ->
04:35:24.067 -> E (32169) BT_BTM: BTM_GetSecurityFlags false
04:35:24.067 ->
04:35:24.067 -> E (32184) BT_BTM: BTM_GetSecurityFlags false
04:35:24.067 ->
04:35:24.114 -> E (32199) BT_BTM: BTM_GetSecurityFlags false
04:35:24.114 ->
04:35:24.114 -> E (32221) BT_BTM: BTM_GetSecurityFlags false
04:35:24.114 ->
04:35:24.114 -> E (32236) BT_BTM: BTM_GetSecurityFlags false
04:35:24.114 ->
04:35:24.162 -> E (32251) BT_BTM: BTM_GetSecurityFlags false
04:35:24.162 ->
04:35:24.162 -> E (32266) BT_BTM: BTM_GetSecurityFlags false
04:35:24.162 ->
04:35:24.255 -> E (32341) BT_BTM: BTM_GetSecurityFlags false
04:35:24.255 ->
04:35:24.255 -> E (32364) BT_BTM: BTM_GetSecurityFlags false
04:35:24.255 ->
04:35:24.255 -> E (32379) BT_BTM: BTM_GetSecurityFlags false
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

newfoxnose, 2023-01-18:
I have the similar issue. the T-vK/ESP32-BLE-Keyboard sketch works fine on one board, but won't work on other boards. I can find the device name on PC, but can't pair.

Jason2866, 2023-04-19:
Does the problem still exists with actual core 2.0.7?

Seref, 2023-04-25:
> Does the problem still exists with actual core 2.0.7?

Sorry for the late reply. I tried 2.0.8 out and I get the same message with the example sketch above when trying to connect to it.

```
19:31:19.441 -> Starting BLE work!
19:31:19.441 -> Characteristic defined! Now you can read it in your phone!
19:31:19.441 -> E (18112) BT_SMP: smp_calculate_link_key_from_long_term_key failed to update link_key. Sec Mode = 2, sm4 = 0x00
19:31:19.441 -> E (18113) BT_SMP: smp_derive_link_key_from_long_term_key failed
19:31:19.441 -> 
19:31:19.441 -> E (18119) BT_BTM: btm_proc_smp_cback received for unknown device
19:31:19.674 -> E (18335) BT_BTM: BTM_GetSecurityFlags false
19:31:19.674 -> 
19:31:19.767 -> E (18425) BT_BTM: BTM_GetSecurityFlags false
19:31:19.767 -> 
19:31:20.047 -> E (18695) BT_BTM: BTM_GetSecurityFlags false
19:31:20.047 -> 
19:31:24.798 -> E (23465) BT_BTM: BTM_GetSecurityFlags false
19:31:24.798 -> 
19:31:24.891 -> E (23555) BT_BTM: BTM_GetSecurityFlags false
19:31:24.891 -> 
19:31:24.985 -> E (23645) BT_BTM: BTM_GetSecurityFlags false
19:31:24.985 -> 
19:31:25.124 -> E (23775) BT_BTM: BTM_GetSecurityFlags false
19:31:25.124 -> 
19:31:25.124 -> E (23798) BT_BTM: BTM_GetSecurityFlags false
19:31:25.124 -> 
19:31:25.169 -> E (23813) BT_BTM: BTM_GetSecurityFlags false
19:31:25.169 -> 
19:31:25.216 -> E (23857) BT_BTM: BTM_GetSecurityFlags false
19:31:25.216 -> 
19:31:25.216 -> E (23880) BT_BTM: BTM_GetSecurityFlags false
19:31:25.216 -> 
19:31:25.216 -> E (23895) BT_BTM: BTM_GetSecurityFlags false
19:31:25.216 -> 
```

Also neither does the BLE-Keyboard/Mouse/Gamepad libraries work with the ESP32-C3 or S3. Only with the original ESP32. I know that classic is missing with the newer mcus but the BLE stuff should still work right?

Cengls, 2023-06-16:
I got the same error, in the environment 

vscode plo 

specifically manifested as a bluetooth mouse can be used, but the keyboard and the gamepad keep flashing links

VojtechBartoska, 2023-08-17:
Maybe those issues will be covered by https://github.com/espressif/arduino-esp32/pull/7867

Seref, 2023-08-17:
I can test out those fixes when they are merged and testable :)

Seref, 2023-11-30:
I still get the same error messages, but now the keyboard library, which wasn't updated since 2021, works.

The only thing that needs to be changed is the AuthenticationMode to ESP_LE_AUTH_BOND from ESP_LE_AUTH_REQ_SC_MITM_BOND .

I tried this change before without any luck but now a year later and with arduino-esp32 v2.0.14 it seems to work :)

pcace, 2025-02-09:
Any news on this? seems to be not fixed.

SuGlider, 2025-02-09:
I'll reopen this issue the link it to the NimBLE refactoring effort that is right now under development #9878 
@lucasssvaz is in charge of this project and he can test all the examples necessary.

lucasssvaz, 2025-08-30:
We recently moved from Bluedroid to NimBLE. Could you please test it with the newest version ?

closed 2025-08-30 by lucasssvaz