espressif/arduino-esp32 #7371: The BT device of the esp32 cannot be found on an iPhone running IOS16

fary99, 2022-10-19:
### Board

esp32

### Device Description

Esp32 development board

### Hardware Configuration

none

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

windows10

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

921600

### Description

iphone cannot search for Bluetooth devices for esp32

### Sketch

```cpp
//This example code is in the Public Domain (or CC0 licensed, at your option.)
//By Evandro Copercini - 2018
//
//This example creates a bridge between Serial and Classical Bluetooth (SPP)
//and also demonstrate that SerialBT have the same functionalities of a normal Serial

#include "BluetoothSerial.h"
#include "esp_bt.h"
#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif

BluetoothSerial SerialBT;

void setup() {
  Serial.begin(115200);
  SerialBT.begin("ESP32test"); //Bluetooth device name
  Serial.println("The device started, now you can pair it with bluetooth!");
}

void loop() {
  if (Serial.available()) {
    SerialBT.write(Serial.read());
  }
  if (SerialBT.available()) {
    Serial.write(SerialBT.read());
  }
  delay(20);
}
```

### Debug Message

```plain
none
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-10-19:
https://github.com/espressif/arduino-esp32/tree/master/libraries/BluetoothSerial

fary99, 2022-10-20:
> https://github.com/espressif/arduino-esp32/tree/master/libraries/BluetoothSerial

I cannot access the IOS app link provided by Readme

me-no-dev, 2022-10-21:
Could it be that the iPhone does not ave classic BT radio? And supports only BLE?

sirapol, 2024-01-05:
Is it solve ?
I still waiting.

marioraffin, 2024-02-07:
I have the same problem.

hperrot, 2024-03-19:
I have the same problem.
@VojtechBartoska could you please leave this issue open or hint us in any direction we could explore?

VojtechBartoska, 2024-03-19:
Hello, reopening.

me-no-dev, 2024-04-15:
iOS Devices do not support classic BT SPP (BTSerial). Only BLE works with iOS

closed 2024-04-15 by me-no-dev

P-R-O-C-H-Y, 2024-04-22:
@sirapol I am just posting a comment from IDF issue about BT Classic on iPhone. Take a look to see the options you have:  https://github.com/espressif/esp-idf/issues/6513#issuecomment-776244467

Also link to [Apple's supported Bluetooth profiles on iOS and iPadOS](https://support.apple.com/en-gb/102842) may help.

nectio, 2024-09-29:
iPhones don't support Bluetooth Classic

Sagigamil, 2025-03-05:
I tried BLE and not Bluetooth classic (using esp32-c3) without a success, the device won't found on iPhone but works well with android, Any suggestions?