espressif/arduino-esp32 #7373: SD card issues with ESP32S3 in SDMMC mode

mtk11, 2022-10-19:
### Board

Custom ESP32S3 board

### Device Description

Custom board

### Hardware Configuration

SDMMC_CMD         35
SDMMC_CLK         36
SDMMC_D0          37
SDMMC_D1          38
SDMMC_D2          39
SDMMC_D3          40

### Version

v2.0.5

### IDE Name

Arduino

### Operating System

Windows 7

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Errors appear.ed

### Sketch

```cpp
https://github.com/espressif/arduino-esp32/blob/master/libraries/SD_MMC/examples/SDMMC_Test/SDMMC_Test.ino

Pin config: SD_MMC.setPins(36, 35, 37, 38, 39, 40);
```

### Debug Message

```plain
..... 
E (1237) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (1237) diskio_sdmmc: Check status failed (0x107)
E (2238) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (2238) diskio_sdmmc: Check status failed (0x107)
Writing file: /hello.txt
File written
Appending to file: /hello.txt
Message appended
Reading file: /hello.txt
Read from file: Hello World!
Deleting file: /foo.txt
File deleted
Renaming file /hello.txt to /foo.txt
File renamed
Reading file: /foo.txt
Read from file: Hello World!
1048576 bytes read for 198 ms
E (3491) sdmmc_req: handle_idle_state_events unhandled: 00000042 00000000
1048576 bytes written for 1413 ms
Total space: 1881MB
Used space: 21MB

With ESP3D 3.0 (https://github.com/luc-github/ESP3D/tree/3.0), getting the following:

E (61131) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (61131) diskio_sdmmc: Check status failed (0x107)
E (62132) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (62132) diskio_sdmmc: Check status failed (0x107)
E (78447) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (78447) diskio_sdmmc: Check status failed (0x107)
E (80560) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (80560) diskio_sdmmc: Check status failed (0x107)
E (81561) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (81561) diskio_sdmmc: Check status failed (0x107)
E (82562) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (82562) diskio_sdmmc: Check status failed (0x107)
E (82564) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x109
E (82568) diskio_sdmmc: sdmmc_read_blocks failed (265)
E (82574) sdmmc_req: handle_idle_state_events unhandled: 00000008 00000002
E (83697) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (83697) diskio_sdmmc: Check status failed (0x107)
E (84698) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (84698) diskio_sdmmc: Check status failed (0x107)
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-10-21:
depending on the chip/module model, pins 35 to 37 might be in use by OPI flash/psram. Any chip/module that ends on R8 will be using OPI PSRAM and have those pins reserved.

mtk11, 2022-10-21:
@me-no-dev , The module does not have ext SPI RAM. Esp-idf’s [SD Card example (SDMMC)](https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card/sdmmc) works fine. Recently esp-idf had some updates - [sdmmc/sdspi: allow custom setup of SD card frequency](https://github.com/espressif/esp-idf/commit/56f20013174a352edebf5c9d983002237734c1a2) ? Do you intend to include them soon ?

me-no-dev, 2022-10-21:
That fix will be part of ESP32 Arduino 3.0.0: https://github.com/espressif/arduino-esp32/issues/6225#issuecomment-1264289718

timknapen, 2023-01-09:
Could this be related to [[SDMMC] ESP32-S3-N8R8 fails to start a microSD card in SDMMC_FREQ_HIGHSPEED mode (IDFGH-6901)](https://github.com/espressif/arduino-esp32/issues/7373#top)?

It looks like @igrr has a patch there, but I don't know if it's already included in Arduino-esp32

P-R-O-C-H-Y, 2023-01-10:
@timknapen Can you please fix the link to the related issue? Thanks

timknapen, 2023-01-10:
I'm sorry about that I don't know how that happened. Link should work now.

timknapen, 2023-01-12:
Hi @P-R-O-C-H-Y ,
I only have a custom PCB with an ESP32-S3-WROOM-1 (N4R8) and I'm using 4 data lines, all short traces with 10K pull-ups.
I've never had any issues using 20MHz. Everything works perfectly fine (but slowly) at 20Mhz. 
Only at 40MHz I start getting problems. Just like the ESP-IDF issue I referred to above.

mtk11, 2023-01-12:
@P-R-O-C-H-Y , ESP32-S3-EYE board uses ESP32-S3-WROOM-1-N8R8 (8MB PSRAM). I had experienced issue with N8R2 (2MB PSRAM) version and I replaced the module with N8 (no PSRAM) version. After the replacement I haven’t experienced the presented issues with the default config.  [ESP3D 3.0](https://github.com/luc-github/ESP3D/tree/3.0) also works fine, however the file upload speed via WiFi is still slow around 110-120 kB/s.

P-R-O-C-H-Y, 2023-01-13:
@timknapen @mtk11 There is an investigation by ESP-IDF team. We can just wait, once the fix is released. I will inform you about any updates about this issue.

timknapen, 2023-01-13:
@P-R-O-C-H-Y thank you! looking forward to have this solved!
Side note: Is it possible for me to clone the Arduino-esp project and recompile the ESP-IDF in it myself? (to for example test the patch myself) 
I have looked but can't find documentation on how to do this.

P-R-O-C-H-Y, 2023-01-19:
@timknapen You are welcome!
Answer for the side note: Yes it is possible. You can use the esp32-arduino-lib-builder. Links to [docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/lib_builder.html) and [repo](https://github.com/espressif/esp32-arduino-lib-builder).

HelloDB, 2023-03-11:
> 该修复将成为 ESP32 Arduino 3.0.0 的一部分：[#6225（评论）](https://github.com/espressif/arduino-esp32/issues/6225#issuecomment-1264289718)

Hello, when is the 3.0.0 release expected? If it takes too long, is it possible to pre-release a small fix for SDMMC speed issues.

VojtechBartoska, 2023-03-13:
Hello @HelloDB, we don't have exact date for 3.0.0 as we recently started on it. Will consider adding this to following bug fix release.

0x0fe, 2023-05-31:
Hello,is there a way to see the patch? there is some problems with sd_mmc driver, and switching to 20MHz (in 1-bit mode) does nto solve them (write block error). Also on PIO it is particularly not working well. And finally i notice somethign concerning, allocating a large working buffer in PSRAM (64K) affects the sd_mmc driver, to the point it cannot intilize in many cases (dma errors).

P-R-O-C-H-Y, 2023-08-21:
@timknapen @mtk11 Can you try to retest the issue with 40 MHz with the latest version 2.0.11 please? Thanks

timknapen, 2023-08-21:
I just tested on two different PCBs with esp32-s3. 40.000 is still buggy. But I don't think I'm running latest 2.0.11... 
When I compile it says `PACKAGES:   - framework-arduinoespressif32 @ 3.20009.0 (2.0.9) `
And when I edit platformio.ini to include
`
platform = https://github.com/platformio/platform-espressif32.git
framework = arduino
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32#master
`
as documented here: [https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#how-to-update-to-the-latest-code](url)

I get `PACKAGES:  - framework-arduinoespressif32 @ 2.0.6+sha.099b432 `

I'm confused.
Both 2.0.9 and 2.0.6+sha... give me read errors like these at 40.000
`
E (499) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x109
E (500) diskio_sdmmc: sdmmc_read_blocks failed (265)
E (500) sdmmc_req: handle_idle_state_events unhandled: 00000000 00000002
`

Jason2866, 2023-08-22:
Try this
```
platform = https://github.com/platformio/platform-espressif32.git
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
```

timknapen, 2023-08-22:
@P-R-O-C-H-Y Thanks for the hint! I assumed platformio would automatically be able to grab the latest version of the arduino-esp32 package from Github, but that does not seem to be the case.. 

@Jason2866 Thank you for the suggestion but I still get the same results with platformio: (using your suggestion or my previous settings)

```
PLATFORM: Espressif 32 (6.3.2+sha.f1fdbc5)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
- framework-arduinoespressif32 @ 2.0.6+sha.099b432 
- tool-esptoolpy @ 1.40501.0 (4.5.1) 
- toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
- toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
```

This looks like it's using version 2.0.6 ( I don't know what the sha.099b432 means)
It doesn't look like esp IDF version is mentioned during compile. 

When I just compile with my 'normal' settings, it uses 2.0.9!
```
// normal settings:
platform = espressif32 
framework = arduino
```
gives this:
```
PLATFORM: Espressif 32 (6.3.2)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20009.0 (2.0.9) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
```

mikaeltulldahl, 2023-08-22:
@timknapen 
Here's how I specify specific version of framework-arduinoespressif32:
```
platform = espressif32
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.11
board = esp32-s3-devkitc-1
framework = arduino
```
gives this:
```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (6.3.2) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 2.0.11+sha.ae9dae4 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
```
It picks the correct tag (2.0.11) and commit (ae9dae4)

P-R-O-C-H-Y, 2023-08-22:
Thanks @mikaeltulldahl. @timknapen  can you give it another try please?

timknapen, 2023-08-22:
Thank you @mikaeltulldahl !!
That does indeed get me to compile with 2.0.11
output:
```
PLATFORM: Espressif 32 (6.3.2+sha.f1fdbc5)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif32 @ 2.0.11+sha.ae9dae4 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
```

Looks good! Testing on my hardware right now!

timknapen, 2023-08-22:
Thank you @P-R-O-C-H-Y , @mikaeltulldahl and @Jason2866 
Just tested with one of my ESP32-S3-WROOM-1 (N4R8) based prototypes and I can confirm 2.0.11 works much better at 40Mhz. I have another board here that I can test later too.
I have done a bunch of file reads and listing of files from microSD and haven't seen one error so far!
This looks very promising!

One weird thing: the read speed is barely faster than at 40Mhz compared to 20Mhz. Does anyone see this too?
I'll have to investigate further.

closed 2023-08-23 by P-R-O-C-H-Y

nhphuc699, 2024-01-25:
Hi @timknapen , I tried in 1bit mode it worked with 40mhz config. but when use 4bit mode, it only worked with 20mhz config and 40mhz not worked.

timknapen, 2024-02-15:
> Hi @timknapen , I tried in 1bit mode it worked with 40mhz config. but when use 4bit mode, it only worked with 20mhz config and 40mhz not worked. do you have same issuse? Thank You

No I don't have any issues at the moment with 4bit mode at 40mhz. By now I've tested this with at least 3 completely different hardware designs. All work without problems.