espressif/arduino-esp32 #7390: ESP32S2 USB CDC Self Powered Device

HX2003, 2022-10-22:
### Related area

USB

### Hardware specification

ESP32S2 Dev Module

### Is your feature request related to a problem?

In the ESP-IDF programming guide, there is a section that states self-powered devices must monitor the voltage level on USB’s VBUS signal. https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/usb_device.html#self-powered-device I was wondering if there is a need to do that (equivalent api) when using the ESP32S2 as a USB Serial Device, it seems that even without the monitoring connecting and disconnecting works.

### Describe the solution you'd like

-

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

VojtechBartoska, 2022-10-24:
@SuGlider Please help with triage.

me-no-dev, 2022-10-24:
The IDF documentation above is for ESP-IDF v5.1. Support for VBUS monitoring is not available before that. In Arduino we abuse the "RESET" event a bit to make disconnection work. We will have proper support with pin monitoring in ESP32 Arduino 3.0.0

closed 2022-10-24 by me-no-dev