espressif/arduino-esp32 #7392: Print::vprintf (and syntactic glue for flash) (IDFGH-8582)

dirkx, 2022-10-24:


dirkx, 2022-10-24:
@atanisoft @Alvin1Zhang my apologies for misfiling. Thanks for the correction.

VojtechBartoska, 2023-04-26:
We'll triage this issue for 3.0.0 milestone and let you know if this request will be implemented.

lucasssvaz, 2023-09-26:
`vprintf` can easily be added and used to implement printf. My main concern is that by adding `size_t Print::printf(const __FlashStringHelper *ifsh, ...)` we would lose the compile-time checks when using `__FlashStringHelper` as we cannot use `__attribute__ ((format (printf, 2, 3)))` like we do with the regular printf. Idk if that's really safe. Wouldn't it be possible to use the regular `Serial.print(const __FlashStringHelper *ifsh)` ?

PR #8677 (2023-10-06): Add printf method with FlashStringHelper argument
## Description of Change

Added `vprintf`, modified `printf` to use `vprintf` and added new `printf` overload that uses a `FlashStringHelper` argument (This doesn't support compile time checks. Idk if this is a good idea)

## Tests scenarios

Tested using ESP32-DevKitC

## Related links

Closes #7392
closed 2023-10-06 by me-no-dev