espressif/arduino-esp32 #7394: [documentation] where are the thread libs listed (e.g., like WiFi)?

dsyleixa, 2022-10-24:
### Board

ESP32 (all)

### Device Description

ESP32 (all)

### Hardware Configuration

default

### Version

v1.0.6

### IDE Name

Arduino IDE

### Operating System

Windows 7/10/11

### Flash frequency

any

### PSRAM enabled

no

### Upload speed

all

### Description

[documentation] where are the thread libs listed (e.g., like WiFi)?
I can't find it in the libraries folders next to FS, i2s, WiFi, SD, Wiire, whatever...

### Sketch

```cpp
any
```

### Debug Message

```plain
any
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-10-25:
so far, we have documented this: https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/libraries.html#apis

dsyleixa, 2022-10-25:
hmmh, I don't see anything there which is named "thread" or s.th. (for `#include <thread>`)

```
<html><body>
<!--StartFragment--><div role="main" class="document" itemscope="itemscope" itemtype="http://schema.org/Article">
           <div itemprop="articleBody">
             
  <div class="section" id="libraries">
<h1>Libraries<a class="headerlink" href="https://docs.espressif.com/projects/arduino-esp32/en/latest/libraries.html#libraries" title="Permalink to this headline">¶</a></h1>
<p>Here is where the Libraries API’s descriptions are located:</p>
<div class="section" id="supported-peripherals">
<h2>Supported Peripherals<a class="headerlink" href="https://docs.espressif.com/projects/arduino-esp32/en/latest/libraries.html#supported-peripherals" title="Permalink to this headline">¶</a></h2>
<p>Currently, the Arduino ESP32 supports the following peripherals with Arduino APIs.</p>
<div class="wy-table-responsive">

Peripheral | ESP32 | ESP32-S2 | ESP32-C3 | Comments
-- | -- | -- | -- | --
ADC | Yes | Yes | Yes |  
Bluetooth | Yes | Not Supported | Not Supported | Bluetooth Classic
BLE | Yes | Not Supported | Yes |  
DAC | Yes | Yes | Not Supported |  
Ethernet | Yes | Not Supported | Not Supported | (*)
GPIO | Yes | Yes | Yes |  
Hall Sensor | Yes | Not Supported | Not Supported |  
I2C | Yes | Yes | Yes |  
I2S | Yes | No | No | WIP
LEDC | Yes | Yes | Yes |  
Motor PWM | No | Not Supported | Not Supported |  
Pulse Counter | No | No | No |  
RMT | Yes | Yes | Yes |  
SDIO | No | No | No |  
SPI | Yes | Yes | Yes |  
Timer | Yes | Yes | Yes |  
Temp. Sensor | Not Supported | Yes | Yes |  
Touch | Yes | Yes | Not Supported |  
TWAI | No | No | No |  
UART | Yes | Yes | Yes |  
USB | Not Supported | Yes | Yes | ESP32-C3 only CDC/JTAG
Wi-Fi | Yes | Yes | Yes |  

</div>
<div class="section" id="notes">
<h3>Notes<a class="headerlink" href="https://docs.espressif.com/projects/arduino-esp32/en/latest/libraries.html#notes" title="Permalink to this headline">¶</a></h3>
<p>(*) SPI Ethernet is supported by all ESP32 families and RMII only for ESP32.</p>
<div class="admonition note">
<p class="admonition-title">Note</p>
<p>Some peripherals are not available for all ESP32 families. To see more details about it, see the corresponding SoC at <a class="reference external" href="https://products.espressif.com">Product Selector</a> page.</p>
</div>
</div>
</div>
<div class="section" id="datasheet">
<h2>Datasheet<a class="headerlink" href="https://docs.espressif.com/projects/arduino-esp32/en/latest/libraries.html#datasheet" title="Permalink to this headline">¶</a></h2>
<ul class="simple">
<li><p><a class="reference external" href="https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf">ESP32</a> (Datasheet)</p></li>
<li><p><a class="reference external" href="https://www.espressif.com/sites/default/files/documentation/esp32-s2_datasheet_en.pdf">ESP32-S2</a> (Datasheet)</p></li>
<li><p><a class="reference external" href="https://www.espressif.com/sites/default/files/documentation/esp32-c3_datasheet_en.pdf">ESP32-C3</a> (Datasheet)</p></li>
<li><p><a class="reference external" href="https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf">ESP32-S3</a> (Datasheet)</p></li>
</ul>
</div>
<div class="section" id="apis">
<h2>APIs<a class="headerlink" href="https://docs.espressif.com/projects/arduino-esp32/en/latest/libraries.html#apis" title="Permalink to this headline">¶</a></h2>
<p>The Arduino ESP32 offers some unique APIs, described in this section:</p>
<div class="toctree-wrapper compound">
<ul>
<li class="toctree-l1"><a class="reference internal" href="https://docs.espressif.com/projects/arduino-esp32/en/latest/api/adc.html">ADC</a></li>
<li class="toctree-l1"><a class="reference internal" href="https://docs.espressif.com/projects/arduino-esp32/en/latest/api/bluetooth.html">Bluetooth API</a></li>
<li class="toctree-l1"><a class="reference internal" href="https://docs.espressif.com/projects/arduino-esp32/en/latest/api/dac.html">DAC</a></li>
<li class="toctree-l1"><a class="reference internal" href="https://docs.espressif.com/projects/arduino-esp32/en/latest/api/deepsleep.html">Deep Sleep</a></li>
<li class="toctree-l1"><a class="reference internal" href="https://docs.espressif.com/projects/arduino-esp32/en/latest/api/espnow.html">ESP-NOW</a></li>
<li class="toctree-l1"><a class="reference internal" href="https://docs.espressif.com/projects/arduino-esp32/en/latest/api/gpio.html">GPIO</a></li>
[... 331 lines cut ...]
    [ESP32-S3](https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf) (Datasheet)

APIs[¶](https://docs.espressif.com/projects/arduino-esp32/en/latest/libraries.html#apis)

The Arduino ESP32 offers some unique APIs, described in this section:

    [ADC](https://docs.espressif.com/projects/arduino-esp32/en/latest/api/adc.html)
    [Bluetooth API](https://docs.espressif.com/projects/arduino-esp32/en/latest/api/bluetooth.html)
    [DAC](https://docs.espressif.com/projects/arduino-esp32/en/latest/api/dac.html)
    [Deep Sleep](https://docs.espressif.com/projects/arduino-esp32/en/latest/api/deepsleep.html)
    [ESP-NOW](https://docs.espressif.com/projects/arduino-esp32/en/latest/api/espnow.html)
    [GPIO](https://docs.espressif.com/projects/arduino-esp32/en/latest/api/gpio.html)
    [I2C](https://docs.espressif.com/projects/arduino-esp32/en/latest/api/i2c.html)
[... cut ...]
    [SigmaDelta](https://docs.espressif.com/projects/arduino-esp32/en/latest/api/sigmadelta.html)
    [Timer](https://docs.espressif.com/projects/arduino-esp32/en/latest/api/timer.html)
    [USB API](https://docs.espressif.com/projects/arduino-esp32/en/latest/api/usb.html)
    [USB CDC](https://docs.espressif.com/projects/arduino-esp32/en/latest/api/usb_cdc.html)
    [USB MSC](https://docs.espressif.com/projects/arduino-esp32/en/latest/api/usb_msc.html)
    [Wi-Fi API](https://docs.espressif.com/projects/arduino-esp32/en/latest/api/wifi.html)
```

lbernstone, 2022-10-25:
<thread> is C-std library. https://www.tutorialspoint.com/cpp_standard_library/thread.htm
pthread is the esp-idf implementation of posix threads. https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/system/pthread.html
Since these are not part of the Arduino standard, they are not documented in this repo.

Rotzbua, 2022-10-25:
@lbernstone Thread is also an iot network protocol. 
@dsyleixa You have to specify which `thread` you mean. Process or network related?

dsyleixa, 2022-10-25:
I mean ESP32 std::thread and pthread-like impementations for multitasking, but which are  not 100% identical to POSIX- or C++14 std::thread 
as to
https://www.tutorialspoint.com/cpp_standard_library/thread.htm
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/system/pthread.html
this is basically what I was searching, albeit still missing program examples like provided for all the other libraries (in the libraries examples folders), and some things like thread prios have a very special design...

me-no-dev, 2022-10-26:
this seems quite a bit out of the scope of Arduino to explain or show examples.

dsyleixa, 2022-10-26:
My point is: 
Which is deviating from the standard  C++ and POSIX standards, and what is fully standard compliant?  
(e.g., mutex, thread prios, detached, joinable, get_id, native_handle, atomic,...)
And finally no single example is shown and/or explained about anything of threads, and IIUC, there quite a bunch of peculiarities in the implementation for RTOS.

dsyleixa, 2022-10-26:
My point is:
Which is deviating from the C++ and POSIX standards, and what is fully standard compliant?
(e.g., mutex, thread prios, detached, joinable, get_id, native_handle, atomic, delay vs. std::this_thread::sleep_for vs. usleep,...)
And finally no single example is shown and/or explained about anything of threads, and IIUC, there is quite a bunch of peculiarities in the implementation for RTOS.

me-no-dev, 2022-10-26:
I get your point, but what you ask about is outside of the scope of Arduino. You can ask about this in ESP-IDF and/or Espressif's crosstool-NG repositories. We use FreeRTOS tasks in ESP32 Arduino, because we are running on FreeRTOS and have no way around it. Generally Arduino is single-threaded framework.

dsyleixa, 2022-10-26:
My question is not about standard Arduino API but about the special std::thread a/o pthread implementation by the arduino-esp32 APi, and so this interest in examples, explanations, and examples  is 100% on-topic and fully justified IMO.

me-no-dev, 2022-10-26:
I'm not sure you understand what is Arduino and what is not. Your questions are about ESP-IDF and the toolchain, not about what is implemented in Arduino. Arduino implementations are in the core and libraries folder of this repo. Everything outside of those 2 folders is not Arduino.

dsyleixa, 2022-10-26:
well, I am using the thread functions (e.g. std::thread(), thread.join(), standard::mutex,...) by the Arduino ID, the Arduino API and the entire Arduino ecosystem, so my question is finally exactly concerning this arduino-esp32 repo.

me-no-dev, 2022-10-26:
I'm always happy to be schooled on what is Arduino API and what is not :) Maybe someone more knowledgable on this topic than me will be able to answer your questions.

PilnyTomas, 2022-10-26:
If you are using ESP chips with Arduino IDE then 100% you ARE using ESP-IDF, it is just hidden in arduino-esp32. It would not make sense to repeat documentation for the entire ESP-IDF here.

dsyleixa, 2022-10-26:
well, I am asking because I am searching example code which can be compiled by the Arduino IDE, i.e. Arduino examples of *.ino type, and also docs about which single thread functions are available by which Arduino syntax.

me-no-dev, 2022-10-26:
So you are looking for something like this: https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/FreeRTOS/FreeRTOS.ino

me-no-dev, 2022-10-26:
what's wrong with RTOS functions? This is what we run on and what we support. For other things, you are on your own (since you refuse to ask your questions in the relevant repositories).

dsyleixa, 2022-10-26:
I don't use this RTOS low-level stuff at all, I am only using C++ std:: class syntax which is intuitive, common, popular, and handy (also for RaspberryPi/Linux)

me-no-dev, 2022-10-26:
Use whatever you want. That is up to you. Just don't expect to find help about that here :) We use FreeRTOS and it's functions. Regardless how many times you say that "it's included", it will not make `std::thread` be part of Arduino or it's APIs.

Docs for pthread: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/pthread.html
Fourth paragraph reads:
```
C++ Standard Library implementations for std::thread, std::mutex, std::condition_variable, etc. are implemented using pthreads (via GCC libstdc++). Therefore, restrictions mentioned here also apply to the equivalent C++ standard library functionality.
```

dsyleixa, 2022-10-26:
the esp32 syntax is different from standard C/C++.
e.g., for prios and parameters the standard syntax is (optionally/arbitrarily)
```
std::thread thread0(); 
// or perhaps
// pthread_create(&thread0, NULL, timerloop, NULL);
param.sched_priority = 50;   
pthread_setschedparam(thread0, SCHED_RR, &param);
```

this syntax is not possible for arduino-esp32 (IIUC, CMMW).
So I was searching the related docs and examples.

me-no-dev, 2022-10-26:
Syntax is not possible not in Arduino but in ESP-IDF. Since Arduino is on top of ESP-IDF, would it not make sense to actually ask there? I really struggle to understand your reluctance to go and ask at the correct place...

dsyleixa, 2022-10-26:
as you said,
`Syntax is not possible not in Arduino but in ESP-IDF. `
so as I am just using arduino-esp32 it would make no sense to ask in ESP-IDF, does it?
Ad are ther *.ino examples in ESP-IDF? 8I guess: no).
So what is so hard to understand when I ask  about  docs and ino examples for arduino-esp32 here in the arduino-esp32 repo?

me-no-dev, 2022-10-26:
My English is obviously too limited to explain that to you, so I will just stop answering and leave it up to chance that someone here might be able to help you :)

dsyleixa, 2022-10-26:
thanks a lot for your examples, 
An easier example for the start might be:
```
// std::thread for ESP32, Arduino IDE

// thread loops can be exited by internal conditions (e.g. via break or return) 
// or all inner loops by setting THREADRUN=false at any location.

#include <Arduino.h>
#include <thread>
#include <freertos/task.h>
#include <esp_task.h>

volatile static bool THREADRUN = true;

//--------------------------------------------------------------------
void function2() {
   vTaskPrioritySet( NULL, ESP_TASK_MAIN_PRIO ); // set Priority = main prio   
   Serial.println((String)"   function2 started"); 
   while(THREADRUN) {      
      // do stuff
   }
   Serial.println((String)"   function2 terminating"); 
}

//--------------------------------------------------------------------

void function1() {
   vTaskPrioritySet( NULL, ESP_TASK_MAIN_PRIO ); // set Priority = main prio   
   Serial.println((String)"   function1 started"); 
   while(THREADRUN) {      
      // do stuff
   }
   Serial.println((String)"   function1 terminating"); 
}

//--------------------------------------------------------------------

void function0() {
   vTaskPrioritySet( NULL, ESP_TASK_MAIN_PRIO ); // set Priority = main prio 
   Serial.println((String)"   function0 started"); 
   while(THREADRUN) {      
      // do stuff
   }
   Serial.println((String)"   function0 terminating"); 
}

//--------------------------------------------------------------------
//--------------------------------------------------------------------
void setup() {
   Serial.begin(115200);

   std::thread thread_0 (function0);
   std::thread thread_1 (function1);
   std::thread thread_2 (function2);
   
   Serial.println((String)"\n all threads being started, now waiting for all being terminated and to join main thread \n");  // <<<<<<  
   
   thread_0.join();
   thread_1.join();
   thread_2.join();
   
   Serial.println((String)"\n all threads joined, program terminated\n");  // <<<<<<  
}

//--------------------------------------------------------------------
void loop() { // empty placeholder for Arduino compatibility
   // when using repetitive code also in the standard loop() then this loop won't terminate after main/setup has finished
}
```

And now my question is about a list of available functions (simple and advanced features) and step-by-step examples to show either esp32 feature in ino files, ready to compile.
No problem if there is one or the other feature not available or under construction, but one should have the chance to see what's there and what not and how to use it, like shown e.g. in the cplusplus.com or QNX examples:
https://cplusplus.com/reference/thread/thread/
http://www.qnx.com/developers/docs/6.5.0/index.jsp?topic=%2Fcom.qnx.doc.neutrino_lib_ref%2Fp%2Fpthread_create.html

Finally it's not just for me but for all Arduino users, starting into multithreading.

dsyleixa, 2022-10-26:
thanks a lot for your examples, 
An easier Arduino example for the start (showing even some extra features) might be:
```
// std::thread for ESP32, Arduino IDE

// thread loops can be exited by internal conditions (e.g. via break or return) 
// or all inner loops by setting THREADRUN=false at any location.

#include <Arduino.h>
#include <thread>
#include <freertos/task.h>
#include <esp_task.h>

using namespace std;

volatile static bool THREADRUN = true;

//--------------------------------------------------------------------
void function2() {
   vTaskPrioritySet( NULL, ESP_TASK_MAIN_PRIO ); // set Priority = main prio   
   Serial.println((String)"   function2 started"); 
   while(THREADRUN) {      
      // do stuff
   }
   Serial.println((String)"   function2 terminating"); 
}

//--------------------------------------------------------------------

void function1() {
   vTaskPrioritySet( NULL, ESP_TASK_MAIN_PRIO ); // set Priority = main prio   
   Serial.println((String)"   function1 started"); 
   while(THREADRUN) {      
      // do stuff
   }
   Serial.println((String)"   function1 terminating"); 
}

//--------------------------------------------------------------------

void function0() {
   vTaskPrioritySet( NULL, ESP_TASK_MAIN_PRIO ); // set Priority = main prio 
   Serial.println((String)"   function0 started"); 
   while(THREADRUN) {      
      // do stuff
   }
   Serial.println((String)"   function0 terminating"); 
}

//--------------------------------------------------------------------
//--------------------------------------------------------------------
void setup() {
   Serial.begin(115200);

   thread thread_0 (function0);
   thread thread_1 (function1);
   thread thread_2 (function2);
   
   Serial.println((String)"\n all threads being started, now waiting for all being terminated and to join main thread \n");  // <<<<<<  
   
   thread_0.join();
   thread_1.join();
   thread_2.join();
   
   Serial.println((String)"\n all threads joined, program terminated\n");  // <<<<<<  
}

//--------------------------------------------------------------------
void loop() { // empty placeholder for Arduino compatibility
   // when using repetitive code also in the standard loop() then this loop won't terminate after main/setup has finished
}
```

And now my question is about a list of available functions (simple and advanced features) and step-by-step examples to show either esp32 feature in ino files, ready to compile.
No problem if there is one or the other feature not available or under construction, but one should have the chance to see what's there and what not and how to use it, like shown e.g. in the cplusplus.com or QNX.com examples:
https://cplusplus.com/reference/thread/thread/
http://www.qnx.com/developers/docs/6.5.0/index.jsp?topic=%2Fcom.qnx.doc.neutrino_lib_ref%2Fp%2Fpthread_create.html

Finally it's not just for me but for all Arduino users, starting into multithreading.

(just a proposal, I would appreciate if the esp-pthread functions looked exactly like the related original POSIX pthread ones)

dsyleixa, 2022-10-26:
ok, not about sthe simple td::thread, but for the all rest, when it comes e.g. to prios,  pthread_join(), pthread_detach(), pthread_exit(), sched_yield(), ...., and CPU_SET (some day), 
Most of the Arduino-esp-users do not even know that it's available at all.

dsyleixa, 2022-10-26:
ok, perhaps not about the simple std::thread needed, but for all the rest, when it comes e.g. to prios, pthread_join(), pthread_detach(), pthread_exit(), sched_yield(), ...., and CPU_SET (some day),
Most of the Arduino-esp-users do not even know that it's available at all.
So some simple examples would be precious though.

SuGlider, 2022-10-27:
I that @dsyleixa is participating on the Arduino Thread discussion https://github.com/arduino/language/discussions/2

@dsyleixa - Arduino proposed this library as an Arduino-Style API for MultiTasking: 
https://github.com/arduino-libraries/Arduino_Threads
https://github.com/arduino-libraries/Arduino_Threads/blob/main/docs/threading-basics.md

Do you see that as an "Arduino Way" for implementing Threads?
Arduino_Threads Library seems more "Arduinoish" than std:thread and FreeRTOS API.

dsyleixa, 2022-10-28:
what I can see the current "Arduinish" is not suitable,and a "new" Arduinish style is not needed at all.
As I already have stated, I understand that Arduino often does not use standardized C/C++ lib syntax for reasons of simplicity (e.g., digitalRead/Write, analogRead/Write, delay). **But std::thread / std::mutex are already very easy, handsome, logical, suitable, and very common,** so IMO there is no need to invent another API syntax from the scratch with only doubtfully (and falsely claimed) better usability or benefits.
What I can see what the Arduino devs currently try to achieve for their proprietary boards (ARM M1, M4, H7) is still in its early infancy and is quite a bit of gibberish and hotchpotch, tbh - and also supports only MT on one single core.
So IMO there is no reason for an Arduinish lib, std::thread is absolutely fine and absolutely suitable for beginners to multithreading, for esp32-arduino it just lacks of ducumentation and examples.

dsyleixa, 2022-10-28:
what I can see the current "Arduinish MT" is not suitable,and a "new" Arduinish API style is not needed at all.
As I already have stated, I understand that Arduino often does not use standardized C/C++ lib syntax for reasons of simplicity (e.g., digitalRead/Write, analogRead/Write, delay). But std::thread / std::mutex are already very easy, handsome, logical, suitable, and very common, so IMO there is no need to invent another API syntax from the scratch with only doubtfully (and falsely claimed) better usability or benefits.
How easy std:thread actually is I have already shown above https://github.com/espressif/arduino-esp32/issues/7394#issuecomment-1292166967
What I can see what the Arduino devs currently try to achieve for their proprietary boards (ARM M1, M4, H7) is still in its early infancy and is quite a bit of gibberish and hotchpotch, tbh - and also supports only MT on one single core.
So IMO there is no reason for an Arduinish lib, std::thread (plus some standard POSIX pthread commands for advanced optional configuration) is absolutely fine and absolutely suitable for beginners to multithreading, and the confusing low-level RTOS stuff behind that can be perfectly hidden for the end-users - for esp32-arduino it just lacks of ducumentation and examples.

dsyleixa, 2023-02-16:
My reply was related to SuGlider's comment
```
Do you see that as an "Arduino Way" for implementing Threads?
Arduino_Threads Library seems more "Arduinoish" than std::thread and FreeRTOS API.
```
I think what Arduino.cc is trying to do is just nonsense and they better should use std::thread syntax as well.

OTOH, what I am missing here in this https://github.com/espressif/arduino-esp32 repo is:
I am missing a couple of Arduino-API-examples in your repo directory  https://github.com/espressif/arduino-esp32/tree/master/libraries about std::thread usage including mutexes, semaphores, thread prios, cpu-core assignment, suitable for tutorial purposes  how to use multithreading accordingly.
For WiFi, SD, or SPI there are lots of examples, for multithreading nothing (found).
so for std::thread I would have expected  a sub-folder perhaps "multithreading" providing step-by-step examples, usable as a tutorial.

My own examples are most probably not suitable for general usage, but you may find some personal approaches in here:
https://github.com/dsyleixa/Arduino/tree/master/ESP32_Test

closed 2025-01-16 by Parsaabasi