espressif/arduino-esp32 #7395: Importe Zip Arduino IDE fail

Magnytt, 2022-10-25:
### Board

NodeMCU

### Device Description

I juste importe the zip and fail..

### Hardware Configuration

I use arducam OV2640

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

windows11

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

115200

### Description

[image]

### Sketch

```cpp
I juste importe the zip, so no code ...
```

### Debug Message

```plain
Erroe : Resquest installZip failed with message: 13 INTERNAL: Librairy install failed: library not valid
```

### Other Steps to Reproduce

no

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-10-25:
Hello @Magnytt? What exactly are you trying to do?

It's just initial setup, if yes, please take a look here: https://docs.espressif.com/projects/arduino-esp32/en/latest/getting_started.html

Magnytt, 2022-10-25:
I juste want to import the library master to use "esp_camera.h"

Magnytt, 2022-10-25:
> Hello @Magnytt? What exactly are you trying to do?
> 
> It's just initial setup, if yes, please take a look here: https://docs.espressif.com/projects/arduino-esp32/en/latest/getting_started.html

I juste want to import the library master to use "esp_camera.h"

me-no-dev, 2022-10-25:
esp_camera.h is already present in ESP32 Arduino. You do not need to do anything, but install ESP32 support in Arduino.

closed 2022-10-25 by me-no-dev