espressif/arduino-esp32 #7396: Meditation Error: Core  0 panic'ed ESP Crash

hicham-vv, 2022-10-25:
### Board

ESP32 Dev module

### Device Description

DecKitc

### Hardware Configuration

Nothing Special

### Version

v1.0.6

### IDE Name

Platform IO

### Operating System

Windows 10

### Flash frequency

40 Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

Hello, i am facing a problem with my ESP32, 
The problem is when i powered up the SIM800C, my esp crash 90% of time, 
sometimes it's reboting sometimes its stop working,
i noticed  that When i unplug the GSM antenna, the esp32 work perfectly but in other hand i face  a problem where the signal of GSM is  so poor,

any idea about the cause of this problem, and how to fix that 

### Sketch

```cpp
#include <Arduino.h>

#define PWS 18 // Pin power stat
#define KEY 26 // pin Key 
#define NS 23 //  Network stat

void setup()
{
  delay(500);
  Serial.begin(115200);
  while(!Serial);

  delay(2500);
  Serial.println("Powering");

  pinMode(PWS,INPUT);
  pinMode(NS,INPUT);

  byte r=digitalRead(PWS);

  if(r==HIGH){
    Serial.println("Sim800 Already Powered Up");
  }
  else{
    delay(500);
    Serial.println("Powering Up SIM800");
    pinMode(KEY, OUTPUT); //PWR KEY
    delay(10);
    digitalWrite(KEY, LOW);
    delay(1200);
    pinMode(KEY,INPUT);
    delay(3500);
    unsigned long premillis=millis();
    while(r==LOW && (millis()-premillis)<5000){
      r=digitalRead(PWS);
      delay(500);
    }
    if(r==HIGH){
      Serial.println("SIM800 Powered successfully ");
      delay(2000);
    }
    else{
      Serial.println("Can't Power Up SIM800 ");
    }

  }
}
```

### Debug Message

```plain
Powering
Powering Up SIM800
SIM800 Powered successfully 
�-V␁g-V��ѥ���Error: Core  1 panic'ed (LoadStoreErro^�!geption was unhandled.
Core 1 register dump:
P���BP'���0088ef4  Guru Meditation Error: Core  1 pa��ԪIP�塅������debug exception)
Debug exception reason: BREAK instr
Core 1 register dump:
PC      : 0x400803c0  PS      : 0x00060036  A0      : 0x40083b0c  A1      : 0x3ffb1bf0  
A2      : 0xa5a5a5a5  A3      : 0x00000000  A4      : 0x3ffc1478  A5      : 0x00000001  
A6      : 0x00000002  A7      : 0x3f40e363  A8      : 0x3ff40000  A9      : 0x0000007d  
A10     : 0x00ff0000  A11     : 0xff000000  A12     : 0x800d3864  A13     : 0x3ffb1ec0  
A14     : 0x3ffbdc54  A15     : 0x3f40e365  SAR     : 0x0000000a  EXCCAUSE: 0x00000001
EXCVADDR: 0x8008525f  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x400803c0:0x3ffb1bf0 0x40083b09:0x3ffb1cd0 0x4008551f:0x3ffb1cf0 0x40085669:0x3ffb1dc0 0x40085951:0x40083b0c

Rebooting...
```

### Other Steps to Reproduce

When i unplug the GSM antenna from the SIM800C, the problem is no more reproduced 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-10-25:
You need to decode the backtrace. https://github.com/me-no-dev/EspExceptionDecoder

SuGlider, 2022-10-26:
@hicham-vv - I tested it using Arduino Core 2.0.5. It works fine.
Are you using Core 1.0.6?

Any special reason for it? 
I'm sorry, but currently we don't support Core 1.0.x anymore and no bug fixes will be done.

dsyleixa, 2022-10-27:
cores 2.04, 2.05 etc are not 100% backwards compatible to programs running fine by 1.0.6, that is a severe drawback which prevents a switch to 2.0.x in my experience, e.g. as reported here: https://github.com/espressif/arduino-esp32/issues/7340#issuecomment-1287318689

SuGlider, 2022-11-09:
I really don't know how to help you because I can't reproduce such issue here.

lbernstone, 2022-11-09:
Are you supplying enough power to the device?  The SIM800C needs to have a separate decoupling capacitor and can draw as much as 2A on initialization.

SuGlider, 2022-11-09:
> [SuGlider:] cores 2.04, 2.05 etc are not 100% backwards compatible to programs running fine by 1.0.6, that is a severe drawback which prevents a switch to 2.0.x in my experience, e.g. as reported here: [#7340 (comment)](https://github.com/espressif/arduino-esp32/issues/7340#issuecomment-1287318689)

When the major version number changes, it means that there is a breaking change. 
The breaking change in this case is due to the launching of new SoC (ESP32-S2/S3/C3 and so on).
We decided to refactor all the Arduino code to use IDF calls instead of manipulating registers directly, as done in 1.0.6.
This is why we changed it from 1.x.x to 2.x.x

Maybe the community may want to continue to support an open version starting from 1.0.6?
For instance, ESP8266 is supported exclusively by the community.

Please participate in our Community Meeting. It is open to everybody and happens through Google Meet.
Bring this topic to the table.
Last Meeting was reported here: https://github.com/espressif/arduino-esp32/discussions/7314

SuGlider, 2022-11-09:
OK, related to Memory consumption. and https://github.com/espressif/arduino-esp32/issues/7340#issuecomment-1287318689

Version 1.0.6 is very light. It has its own structure for dealing directly with ESP32 registers. The HEAP footprint seems lower.
Version 2.0.x is based on IDF and its own structure. Its HEAP footprint is higher when compared to 1.0.6 version.

At this time, there is no way for us to solve this specific issue. I'm sorry.

I see 2 possible alternatives:
Stick with Arduino 1.0.6 for your project. You may want to modify and improve the 1.0.6 core code by yourself.
Move to Arduino 2.x.x, but it will be necessary  for you to refactor your project.

Best Regards!