espressif/arduino-esp32 #7397: ENC28j60 or any ethernet support for ota from https

KJPOUNDY132, 2022-10-25:
### Board

ESP32 dev mıodule

### Device Description

DevKit C

### Hardware Configuration

just esp32 with 1 led and button

### Version

latest master (checkout manually)

### IDE Name

Arduino  IDE

### Operating System

macOS

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

115200

### Description

Hi wanto make ota updates from https server via ethernet how does it support?

### Sketch

```cpp
f
```

### Debug Message

```plain
f
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-11-05:
This link may help:
https://forum.arduino.cc/t/esp32-with-enc28j60-module-update-via-ethernet/1038318

You may try this (at your risk):
https://github.com/jandrassy/ArduinoOTA
https://stackoverflow.com/questions/60432074/updating-esp32-firmware-via-ethernetinternet

VojtechBartoska, 2022-12-15:
is this still valid?

VojtechBartoska, 2023-01-18:
Hello, I'm closing this issue as expired. If needed, you can reopen it.

closed 2023-01-18 by VojtechBartoska