espressif/arduino-esp32 #7398: Serial Bluetooth Library Issue

blodhgarm96, 2022-10-25:
### Board

ESP32 Dev Kit

### Device Description

No peripherals attached. Just the ESP32 board. 

### Hardware Configuration

N/A

### Version

v2.0.0

### IDE Name

Arduino IDE

### Operating System

Windows 10 

### Flash frequency

90MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Error when using BluetoothSerial.h libarary the example sketch SerialToSerialBT fails to compile. 

I'm not sure what the problem is. I didn't see anything in the readme mentioning the need to define or modify any variables in the SerialBluetooth.cpp file. 

Any help or advice would be greatly appreciated. I'm trying to get this up and running for a Halloween costume. 

### Sketch

```cpp
//This example code is in the Public Domain (or CC0 licensed, at your option.)
//By Evandro Copercini - 2018
//
//This example creates a bridge between Serial and Classical Bluetooth (SPP)
//and also demonstrate that SerialBT have the same functionalities of a normal Serial

#include "BluetoothSerial.h"

#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif

#if !defined(CONFIG_BT_SPP_ENABLED)
#error Serial Bluetooth not available or not enabled. It is only available for the ESP32 chip.
#endif

BluetoothSerial SerialBT;

void setup() {
  Serial.begin(115200);
  SerialBT.begin("ESP32test"); //Bluetooth device name
  Serial.println("The device started, now you can pair it with bluetooth!");
}

void loop() {
  if (Serial.available()) {
    SerialBT.write(Serial.read());
  }
  if (SerialBT.available()) {
    Serial.write(SerialBT.read());
  }
  delay(20);
}
```

### Debug Message

```plain
C:\Users\eshel\OneDrive\Documents\Arduino\libraries\Bluetooth Serial\src\BluetoothSerial.cpp: In function 'void esp_spp_cb(esp_spp_cb_event_t, esp_spp_cb_param_t*)':
C:\Users\eshel\OneDrive\Documents\Arduino\libraries\Bluetooth Serial\src\BluetoothSerial.cpp:374:76: error: 'struct esp_spp_cb_param_t::spp_discovery_comp_evt_param' has no member named 'service_name'
                     sdpRecords[param->disc_comp.scn[i]] = param->disc_comp.service_name[0];
                                                                            ^
C:\Users\eshel\OneDrive\Documents\Arduino\libraries\Bluetooth Serial\src\BluetoothSerial.cpp: In function 'void esp_bt_gap_cb(esp_bt_gap_cb_event_t, esp_bt_gap_cb_param_t*)':
C:\Users\eshel\OneDrive\Documents\Arduino\libraries\Bluetooth Serial\src\BluetoothSerial.cpp:570:14: error: 'ESP_BT_GAP_CONFIG_EIR_DATA_EVT' was not declared in this scope
         case ESP_BT_GAP_CONFIG_EIR_DATA_EVT:
              ^
C:\Users\eshel\OneDrive\Documents\Arduino\libraries\Bluetooth Serial\src\BluetoothSerial.cpp:574:14: error: 'ESP_BT_GAP_READ_REMOTE_NAME_EVT' was not declared in this scope
         case ESP_BT_GAP_READ_REMOTE_NAME_EVT:
              ^
C:\Users\eshel\OneDrive\Documents\Arduino\libraries\Bluetooth Serial\src\BluetoothSerial.cpp:575:24: error: 'union esp_bt_gap_cb_param_t' has no member named 'read_rmt_name'
             if (param->read_rmt_name.stat == ESP_BT_STATUS_SUCCESS ) {
                        ^
C:\Users\eshel\OneDrive\Documents\Arduino\libraries\Bluetooth Serial\src\BluetoothSerial.cpp:582:14: error: 'ESP_BT_GAP_MODE_CHG_EVT' was not declared in this scope
         case ESP_BT_GAP_MODE_CHG_EVT:
              ^
C:\Users\eshel\OneDrive\Documents\Arduino\libraries\Bluetooth Serial\src\BluetoothSerial.cpp: In member function 'BTScanResults* BluetoothSerial::discover(int)':
C:\Users\eshel\OneDrive\Documents\Arduino\libraries\Bluetooth Serial\src\BluetoothSerial.cpp:1103:30: error: 'ESP_BT_CONNECTABLE' was not declared in this scope
     esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_GENERAL_DISCOVERABLE);
                              ^
C:\Users\eshel\OneDrive\Documents\Arduino\libraries\Bluetooth Serial\src\BluetoothSerial.cpp:1103:50: error: 'ESP_BT_GENERAL_DISCOVERABLE' was not declared in this scope
     esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_GENERAL_DISCOVERABLE);
                                                  ^
C:\Users\eshel\OneDrive\Documents\Arduino\libraries\Bluetooth Serial\src\BluetoothSerial.cpp: In member function 'bool BluetoothSerial::discoverAsync(BTAdvertisedDeviceCb, int)':
C:\Users\eshel\OneDrive\Documents\Arduino\libraries\Bluetooth Serial\src\BluetoothSerial.cpp:1129:30: error: 'ESP_BT_CONNECTABLE' was not declared in this scope
     esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_GENERAL_DISCOVERABLE);
                              ^
C:\Users\eshel\OneDrive\Documents\Arduino\libraries\Bluetooth Serial\src\BluetoothSerial.cpp:1129:50: error: 'ESP_BT_GENERAL_DISCOVERABLE' was not declared in this scope
     esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_GENERAL_DISCOVERABLE);
                                                  ^

exit status 1

Compilation error: exit status 1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-10-26:
Looking at the paths that give error, you are not using this core's Bluetooth Serial library, but some other one located in your sketch folder: `C:\Users\eshel\OneDrive\Documents\Arduino\libraries\Bluetooth Serial`

closed 2022-10-26 by me-no-dev

costycnc, 2022-12-30:
I compile for first time a esp32 bluetooth example and give this error "Bluetooth is not enabled! Please run `make menuconfig` to and enable it"
So... the problem is that in BluetoothSerial.h call sdkconfig.h but in \AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.5\tools\sdk\esp32  is declared sdkconfig without h extension

So i put .h a sdkconfig and working fine  sdkconfig.h