espressif/arduino-esp32 #7400: [I][WiFiClient.cpp:548] connected(): Unexpected: RES: 0, ERR: 119

DisasterofPuppets, 2022-10-25:
### Board

ESP32

### Device Description

Wemos Lolin32 Lite ,  DOIT ESP32 Dev Kit V1

### Hardware Configuration

[](url)
[image]

### Version

other

### IDE Name

Arduino 1.8.12

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

Wifi Connects and then disconnects with the below error:

09:36:20.841 -> WiFi connected
09:36:20.841 -> IP address: 
09:36:20.841 -> 192.168.1.42
09:36:20.841 -> Attempting to connect to #mytwitchchannel
09:36:21.077 -> connected and ready to rock
09:36:26.047 -> [  7885][I][WiFiClient.cpp:548] connected(): Unexpected: RES: 0, ERR: 119
09:36:26.047 -> [  7886][I][WiFiClient.cpp:548] connected(): Unexpected: RES: 0, ERR: 119

### Sketch

```cpp
/*******************************************************************
    Connect to Twtich Chat with a Bot

   Created with code from TheOtherLoneStar (https://www.twitch.tv/theotherlonestar)
   Hackaday IO: https://hackaday.io/otherlonestar

   By Brian Lough (https://www.twitch.tv/brianlough)
   YouTube: https://www.youtube.com/channel/UCezJOfu7OtqGzd5xrP3q6WA
 *******************************************************************/

#include <WiFi.h>          //https://github.com/esp8266/Arduino
#include <IRCClient.h>

//define your default values here, if there are different values in config.json, they are overwritten.

#define IRC_SERVER   "irc.chat.twitch.tv"
#define IRC_PORT     6667

//------- Replace the following! ------
char ssid[] = "ssid";       // your network SSID (name)
char password[] = "pass";  // your network key

//The name of the channel that you want the bot to join
const String twitchChannelName = "mytwitchchannel";

//The name that you want the bot to have
#define TWITCH_BOT_NAME "mytwitchbot"

//OAuth Key for your twitch bot
// https://twitchapps.com/tmi/
#define TWITCH_OAUTH_TOKEN "oauth:xxxxxxxxxxxxxxxxxxxxxx

//------------------------------

int led = LED_BUILTIN;
String ircChannel = "";

WiFiClient wiFiClient;
IRCClient client(IRC_SERVER, IRC_PORT, wiFiClient);

// put your setup code here, to run once:
void setup() {
  Serial.begin(115200);

  pinMode(led, OUTPUT);
  Serial.println();

  // Attempt to connect to Wifi network:
  Serial.print("Connecting Wifi: ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  IPAddress ip = WiFi.localIP();
  Serial.println(ip);

  ircChannel = "#" + twitchChannelName;

  client.setCallback(callback);
}

void loop() {

  // Try to connect to chat. If it loses connection try again
  if (!client.connected()) {
    Serial.println("Attempting to connect to " + ircChannel );
    // Attempt to connect
    // Second param is not needed by Twtich
    if (client.connect(TWITCH_BOT_NAME, "", TWITCH_OAUTH_TOKEN)) {
      client.sendRaw("JOIN " + ircChannel);
      Serial.println("connected and ready to rock");
      sendTwitchMessage("Oh no, that boi connected");
      delay(5000);
    } else {
      Serial.println("failed... try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(5000);
    }
    return;
  }
  client.loop();
}

void sendTwitchMessage(String message) {
  client.sendMessage(ircChannel, message);
}

void callback(IRCMessage ircMessage) {
  //Serial.println("In CallBack");

  if (ircMessage.command == "PRIVMSG" && ircMessage.text[0] != '\001') {
    //Serial.println("Passed private message.");
    ircMessage.nick.toUpperCase();

    String message("<" + ircMessage.nick + "> " + ircMessage.text);

    //prints chat to serial
    Serial.println(message);

    for (int i = 0; i < 6; i++) {
      digitalWrite(led, HIGH);
      delay(50);
      digitalWrite(led, LOW);
      delay(25);
    }

    return;
  }
}
```

### Debug Message

```plain
09:36:18.215 -> Connecting Wifi: SLOWKEVIN
09:36:18.215 -> [    58][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
09:36:18.308 -> [   147][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
09:36:18.355 -> [   145][V][WiFiGeneric.cpp:338] _arduino_event_cb(): STA Started
09:36:18.355 -> [   153][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
09:36:18.355 -> .[   419][V][WiFiGeneric.cpp:353] _arduino_event_cb(): STA Connected: SSID: MYWIFI, BSSID: 10:0c:6b:e2:89:b2, Channel: 1, Auth: WPA2_PSK
09:36:18.591 -> [   420][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
09:36:18.827 -> ....[  2551][V][WiFiGeneric.cpp:367] _arduino_event_cb(): STA Got New IP:192.168.1.42
09:36:20.746 -> [  2551][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
09:36:20.746 -> [  2554][D][WiFiGeneric.cpp:991] _eventCallback(): STA IP: 192.168.1.42, MASK: 255.255.255.0, GW: 192.168.1.1
09:36:20.841 -> 
09:36:20.841 -> WiFi connected
09:36:20.841 -> IP address: 
09:36:20.841 -> 192.168.1.42
09:36:20.841 -> Attempting to connect to #Mytwitchchannel
09:36:21.077 -> connected and ready to rock
09:36:26.047 -> [  7885][I][WiFiClient.cpp:548] connected(): Unexpected: RES: 0, ERR: 119
09:36:26.047 -> [  7886][I][WiFiClient.cpp:548] connected(): Unexpected: RES: 0, ERR: 119
```

### Other Steps to Reproduce

Removing the irc code and using the standard wifi connection seems to ok. Only starts having issues with the IRC code included. Works without issue on ESP8266.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2022-10-28:
I suspect the issue is with IRCClient, not WiFiClient.  The error returned from WiFiClient is 119, which is defined as:
#define ENOBUFS         119

Looking at WiFiClient.cpp:548-550, WiFiClient fails on recv(), so it may not have a buffer to receive data.  The WiFiClient itself should stay connected.  Here's the relevant function:
`uint8_t WiFiClient::connected()
{
    if (_connected) {
        uint8_t dummy;
        int res = recv(fd(), &dummy, 0, MSG_DONTWAIT);
        // avoid unused var warning by gcc
        (void)res;
        // recv only sets errno if res is <= 0
        if (res <= 0){
          switch (errno) {
              case EWOULDBLOCK:
              case ENOENT: //caused by vfs
                  _connected = true;
                  break;
              case ENOTCONN:
              case EPIPE:
              case ECONNRESET:
              case ECONNREFUSED:
              case ECONNABORTED:
                  _connected = false;
                  log_d("Disconnected: RES: %d, ERR: %d", res, errno);
                  break;
              default:
                  log_i("Unexpected: RES: %d, ERR: %d", res, errno);
                  _connected = true;
                  break;
          }
        } else {
          _connected = true;
        }
    }
    return _connected;
}`

VojtechBartoska, 2022-12-15:
Closing es expired, thanks for help @lbernstone :)

closed 2022-12-15 by VojtechBartoska

theotherjenkutler, 2022-12-21:
I am having the same issue. What is weird about it is that I don't have this problem with certain ESP32-WROVER modules. I had a custom board made and it was working fine. I did a second run which is identical and the code no longer works. 

Here is what I'm working with:

`
#include <SPI.h>
#include <Adafruit_VS1053.h>
#include <WiFi.h>
#include <Arduino.h>
#include <WiFiManager.h>           //https://github.com/tzapu/WiFiManager
#include "FS.h"
#include "SPIFFS.h"
#include "ADS1X15.h"               //https://github.com/RobTillaart/ADS1X15

ADS1115 ADS(0x48); 

#define FORMAT_SPIFFS_IF_FAILED true
const char filename[] = "/i.txt";
File file;

unsigned long ohShit; //in case node-red doesn't respond fast enough with stream

char macIdentifierNodeRed[17];

char uniquePath[20];
const char *host = "146.190.221.89";
String path = "/000000000000.mp3";
int httpPort = 8000;

WiFiClient transferClient;
const int transferPort = 49153;

WiFiClient initClient;
const int initPort = 49152;

//int edasamplecounter = 0;
size_t edaBuffSize = 128;
int16_t edaUnfilt[128] = {0}; 

//#define EDAUNFILTPIN 33
#define PLAYING_INDICATOR 25
#define END_INDICATOR 26
#define VS1053_RESET   -1     // VS1053 reset pin (not used!)
#define VS1053_CS      27     // VS1053 chip select pin (output)
#define VS1053_DCS     14     // VS1053 Data/command select pin (output)
#define VS1053_DREQ    13     // VS1053 Data request, ideally an Interrupt pin

#define VOLUME_KNOB    34

int lastvol = 30;

Adafruit_VS1053 musicPlayer =  Adafruit_VS1053(VS1053_RESET, VS1053_CS, VS1053_DCS, VS1053_DREQ);

WiFiClient audioClient;

void setup() {
  WiFi.mode(WIFI_STA); 
  
  Serial.begin(115200);

  Serial.println("beginning of setup");printMem();

  // Wait for VS1053 to initialize
  delay(3000);
   if (! musicPlayer.begin()) { // initialise the music player
     Serial.println(F("VS1053 Failure"));
     while (1);
  }
  
  musicPlayer.setVolume(lastvol, lastvol);
  pinMode(VOLUME_KNOB, INPUT_PULLUP);
  pinMode(PLAYING_INDICATOR, OUTPUT);
  pinMode(END_INDICATOR, OUTPUT);
  digitalWrite(PLAYING_INDICATOR, LOW);
  digitalWrite(END_INDICATOR, LOW);

  WiFiManager wm;
  bool res;
  res = wm.autoConnect("Measures","measures");
  if(!res){
    Serial.println(F("WiFi Failure"));
    //ESP.restart();
  }else{
    Serial.println(F("WiFi Connected"));
  }
  while(WiFi.status() != WL_CONNECTED){
    delay(100);
  }

  if(!SPIFFS.begin(FORMAT_SPIFFS_IF_FAILED)){
    Serial.println(F("SPIFFS Mount Failed"));
    return;
  }

  if (!SPIFFS.remove(filename)){
    Serial.println(F("File not deleted"));
  };

  delay(10);

  file = SPIFFS.open(filename, FILE_APPEND);
  if(!file){
    Serial.println(F("SPIFFS can't open file"));
  }
  if(!ADS.begin()){
    Serial.println(F("ADS Failed"));
    while(1);
  }

  getMacIdentifier(); //gets mac address for node-red

  //send mac address to node-red
  if(!initClient.connect(host, initPort)){
    Serial.println(F("Init TCP failed"));
    while(true){
      ;
    }
  }else{
    Serial.println(F("Init TCP success"));
    initClient.println(macIdentifierNodeRed);
    delay(10);
    initClient.stop();
  }
  
  delay(3000);
  
  if (!audioClient.connect(host, httpPort)) {
    Serial.println(F("Icecast failed"));
    return;
  }
  delay(1000);
  
  // This will send the stream request to the server
  audioClient.print(String("GET ") + path + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" + 
               "Connection: close\r\n\r\n");

  delay(1000);
  
  ADS.requestADC(0);

  bool audioClientStatus;
  Serial.print("audioClientStatus: "); Serial.println(audioClientStatus);

  Serial.println("endofsetup");printMem();
}`