espressif/arduino-esp32 #7401: Simple time example doesn't work

Plumal73, 2022-10-26:
### Board

ESP32 C

### Device Description

I use :
- Espressif ESP32-C3-DevKitM-1
- Arduino IDE 1.8.15
- ESP32 boards on Arduino IDE V2.0.5

### Hardware Configuration

Dont use any GPIO output

### Version

v2.0.5

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Hello,

When I try to upload this basic example (I didn't change anything in the code) I've an issue on my serial monitor.

Here is what I have :

`Rebooting...
ESP-ROM:esp32c3-20200918
Build:Sep 18 2020
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40381976
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6ac
load:0x403d0000,len:0x24e4
entry 0x403ce000
Connecting to YOUR_SSID Guru Meditation Error: Core  0 panic'ed (Illegal instruction). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x40001be4  RA      : 0x420689ae  SP      : 0x3fca3130  GP      : 0x3fc8e400  
TP      : 0x3fc90e38  T0      : 0x40057fa6  T1      : 0x4038362a  T2      : 0xffffffff  
S0/FP   : 0x3fc95198  S1      : 0x40383b02  A0      : 0x3fc8f33c  A1      : 0x40383a04  
A2      : 0x4038528a  A3      : 0x40383746  A4      : 0x4038377e  A5      : 0x3fcdf4f8  
A6      : 0x40383814  A7      : 0x403837e8  S2      : 0x3c09dfa4  S3      : 0x00000002  
S4      : 0x3fc95000  S5      : 0x3fce0000  S6      : 0x3fce0000  S7      : 0x3fce0000  
S8      : 0x3ff1b000  S9      : 0x3fce0000  S10     : 0x3fcdf8d4  S11     : 0x00000000  
T3      : 0x40383f60  T4      : 0x40383cf6  T5      : 0x3fc8f000  T6      : 0x42011a9a  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000002  MTVAL   : 0x00000000  
[... 174 lines cut ...]
3fca3330: 0x00000017 0x00000001 0x00000000 0x00000000 0x00000000 0x3fc967f4 0x3fc9685c 0x3fc968c4
3fca3350: 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x4206f460 0x00000000
3fca3370: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca3390: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca33b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca33d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca33f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca3410: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca3430: 0x00000000 0xba000000 0xbaad5678 0x000008b4 0xabba1234 0x000008a8 0xdeadbeef 0x00000000
3fca3450: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca3470: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca3490: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca34b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca34d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca34f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca3510: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: 0000000000000000

And repeat repeat ... repeat... repeat again..
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Plumal73, 2022-11-02:
Hello, 

I tried different USB cable and USB port, doesn't change. I bought one more card and I'll test with it tomorrow. 

To answer your question, yes the crash happens after turning on the Wifi (I tried to write"Hello world" before turning on the wifi, and I saw "Hello world" in my Serial Terminal).

Do u have another idea ?

Thanks

VojtechBartoska, 2023-01-18:
@PilnyTomas can you please retest the example if it works smoothly? Thanks

PilnyTomas, 2023-01-18:
Hi, I was not able to reproduce this error:
ESP32-C3
Linux

VojtechBartoska, 2023-01-18:
I'm closing this as won't fix, if needed, you can reopen it. Thanks.

closed 2023-01-18 by VojtechBartoska