espressif/arduino-esp32 #7405: eMMC driver

marchingband, 2022-10-26:
### Board

ESP32-s3

### Device Description

DevKitC-1

### Hardware Configuration

I have a custom PCB with eMMC attached as follows
```
#define EMMC_CLK     GPIO_NUM_16
#define EMMC_CMD    GPIO_NUM_15
#define EMMC_DATA0 GPIO_NUM_4
#define EMMC_DATA1 GPIO_NUM_5
#define EMMC_DATA2 GPIO_NUM_6
#define EMMC_DATA3 GPIO_NUM_7
```

### Version

v2.0.5

### IDE Name

Arduino IDE

### Operating System

macOS 10.15

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

In this issue: https://github.com/espressif/esp-idf/issues/8257
It seems there was a patch to esp-idf 4.4 which fixes some issues with the eMMC driver timing.
Has this patch been applied to ESP32-Arduino ?
Thanks!

### Sketch

```cpp
#include "Arduino.h"
#include "FS.h"
#include "driver/sdmmc_types.h"
#include "sd_defines.h"
#include "vfs_api.h"

#include <dirent.h>
#include "esp_vfs_fat.h"
#include "driver/sdmmc_host.h"
#include "driver/sdmmc_defs.h"
#include "sdmmc_cmd.h"
#include "soc/sdmmc_pins.h"
#include "ff.h"
#include "USB.h"

#define EMMC_CLK GPIO_NUM_16
#define EMMC_CMD GPIO_NUM_15
#define EMMC_DATA0 GPIO_NUM_4
#define EMMC_DATA1 GPIO_NUM_5
#define EMMC_DATA2 GPIO_NUM_6
#define EMMC_DATA3 GPIO_NUM_7

sdmmc_card_t *card;
sdmmc_host_t host = SDMMC_HOST_DEFAULT();
  
void init_emmc(){
//    Serial.begin(115200);
   // host.max_freq_khz = SDMMC_FREQ_HIGHSPEED;
   host.max_freq_khz = SDMMC_FREQ_DEFAULT;
//   host.flags = SDMMC_HOST_FLAG_4BIT;
   host.flags = SDMMC_HOST_FLAG_1BIT;
   host.flags &= ~SDMMC_HOST_FLAG_DDR;

   sdmmc_slot_config_t slot_config = SDMMC_SLOT_CONFIG_DEFAULT();
   slot_config.width = 1;
   slot_config.clk = EMMC_CLK;
   slot_config.cmd = EMMC_CMD;
   slot_config.d0 = EMMC_DATA0;
//   slot_config.d1 = EMMC_DATA1;
//   slot_config.d2 = EMMC_DATA2;
//   slot_config.d3 = EMMC_DATA3;

   sdmmc_host_init_slot(SDMMC_HOST_SLOT_1, &slot_config);
   gpio_reset_pin(EMMC_CLK);
   gpio_reset_pin(EMMC_CMD);
   gpio_reset_pin(EMMC_DATA0);
   gpio_reset_pin(EMMC_DATA1);
   gpio_reset_pin(EMMC_DATA2);
   gpio_reset_pin(EMMC_DATA3);
   gpio_pullup_en(EMMC_CLK);
   gpio_pulldown_dis(EMMC_CLK);
   gpio_pullup_en(EMMC_CMD);
   gpio_pulldown_dis(EMMC_CMD);
   gpio_pullup_en(EMMC_DATA0);
   gpio_pulldown_dis(EMMC_DATA0);
   gpio_pullup_en(EMMC_DATA1);
   gpio_pulldown_dis(EMMC_DATA1);
   gpio_pullup_en(EMMC_DATA2);
   gpio_pulldown_dis(EMMC_DATA2);
   gpio_pullup_en(EMMC_DATA3);
   gpio_pulldown_dis(EMMC_DATA3);

//    puts("Initialize HOST and SLOT");

   esp_vfs_fat_sdmmc_mount_config_t mount_config = {
       .format_if_mount_failed = true,
       .max_files = 5,
   };
   
   esp_err_t ret = esp_vfs_fat_sdmmc_mount("/root", &host, &slot_config, &mount_config, &card);
   ESP_ERROR_CHECK(ret);
}

void setup(){
  Serial.begin();
  Serial.setDebugOutput(true);
  USB.begin();
}

int inited = 0;

void loop(){
    Serial.println("run");
    log_i("debug");
    vTaskDelay(1000);
    if(Serial.available()){
        if(inited == 0){
            log_i("init emmc");
            inited = 1;
            init_emmc();
        }
    }
}
```

### Debug Message

```plain
ESP_ERROR_CHECK failed: esp_err_t 0x109 (ESP_ERR_INVALID_CRC) at 0x4037a43c
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-10-27:
That patch was merged before v2.0.5.  However, your issue has a different error, which seems more like it is either noise on the line or truly an invalid crc.
Here's where it is thrown in the source code: https://github.com/espressif/esp-idf/blob/master/components/driver/sdmmc_transaction.c#L341

marchingband, 2022-10-27:
Ok thank you @lbernstone
What would an invalid CRC indicate other then noise? Perhaps a faulty eMMC chip?

me-no-dev, 2022-10-27:
@marchingband why not use the Arduino API for SDMMC?

BTW you are calling `sdmmc_host_init_slot(SDMMC_HOST_SLOT_1, &slot_config);` and after that you are changing the GPIOs, which would most probably put them in a state that is not what the driver expects.

marchingband, 2022-10-27:
@me-no-dev thanks, I have switched the order, but the error persists.
@igrr @lbernstone 
let me summarize because this is a little confusing, can you confirm this is correct?

ESP32-Arduino 2.0.5 using SDMMC class API -> eMMC works
ESP32-Arduino 2.0.5 using idf API -> eMMC works
ESP-IDF 4.x with patch -> eMMC works
ESP-IDF 4.x without patch -> eMMC does not work

Can it be confirmed that it works with Arduino on hardware? I have not seen an example, and there are no example sketches, blog posts, or any mention online of anything, other then the GitHub issue i linked. Has anyone actually succeeded with Arduino using an eMMC chip? I have the eMMC chip KLM4G1FETE-B041

marchingband, 2022-10-27:
Ok thank you so so much @igrr that saves me a lot of tinkering. I will work with a patched version of 4.x for now.

@me-no-dev @lbernstone so am I correct that we will have to wait for ESP-Arduino v2.0.6 for eMMC support?

marchingband, 2022-10-28:
I can confirm that with esp-idf 4.4 and the patch applied, the above sketch works as intended on my PCB.
Both 1 and 4 line work, both `SDMMC_FREQ_DEFAULT` and `SDMMC_FREQ_HIGHSPEED` work.
Thanks again @igrr :)

me-no-dev, 2022-10-28:
> am I correct that we will have to wait for ESP-Arduino v2.0.6 for eMMC support?

That depends on when the patch will be released in the v4.4 branch of ESP-IDF. 2.0.6 will be on IDF 4.4.3

marchingband, 2022-10-28:
@me-no-dev ok cool thanks. Can we leave this issue open for now then? Just thinking it might help others in my position, and it would be great to get a ping when it is merged into Arduino.

marchingband, 2023-01-10:
Any movement on this? Thanks!

P-R-O-C-H-Y, 2023-01-31:
Hi @marchingband, I am moving the Milestone to 3.0.0 version, as the fix is still not merged yet and won't be part of IDF-4.4.4.

P-R-O-C-H-Y, 2023-02-01:
@marchingband I can't tell that for sure (did not test), but it should still work with the patch :)

marchingband, 2023-02-01:
@P-R-O-C-H-Y thats good news thanks :) it also makes me wonder why the patch wasn't merged into 4.4 if it works, but I'm sure there are reasons.

marchingband, 2023-05-06:
@P-R-O-C-H-Y @me-no-dev any updates here? Happy to do some more testing when it is time. Thanks!

igrr, 2023-05-06:
IDF fix has been merged into release/v4.4 and will be available in IDF v4.4.5. We'll need to see whether there will be another Arduino 2.0.x release with IDF v4.4.5.

marchingband, 2023-05-06:
@igrr amazing! thanks. Does this mean it will merge into 5.x as well?

marchingband, 2023-06-24:
@igrr @me-no-dev hoping this patch gets into Arduino ... any updates on the progress/prosprects?
Thank you :) I have version 2 of [this](https://www.sparkfun.com/products/21307) dev board waiting on this patch ... feeling patient :)

VojtechBartoska, 2023-06-26:
This is included in ESP-IDF v4.4.5, will be included in upcoming Arduino release 2.10.0.

igrr, 2023-06-26:
@marchingband I'm sorry, have missed your question:

> Does this mean it will merge into 5.x as well?

We have merged this fixes to:
* release/v4.4: https://github.com/espressif/esp-idf/commit/fb6d9cd7417c90932529c2d83cdc42d5fbd6ff50
* release/v5.0: https://github.com/espressif/esp-idf/commit/1ebeea7763c180bf3046d01bcabee91bbc74ce83
* release/v5.1: https://github.com/espressif/esp-idf/commit/5b18007d92e317f8f0cbaedd140567357961e840
* master: https://github.com/espressif/esp-idf/commit/e9badf51c73a15ccea063368649d44a48d0ee97c

marchingband, 2023-06-26:
@igrr @VojtechBartoska TYSM!

VojtechBartoska, 2023-07-18:
https://github.com/espressif/arduino-esp32/releases/tag/2.0.10 was released with sdmmc updates. I'm closing, if you face any problem you can reopen this issue.

marchingband, 2023-09-27:
@VojtechBartoska @igrr 
I'm trying this with Arduino IDE, where I have 2.0.13 installed.

```
#include "Arduino.h"
#include <string.h>
#include <sys/unistd.h>
#include <sys/stat.h>
#include "esp_vfs_fat.h"
#include "sdmmc_cmd.h"
#include "driver/sdmmc_host.h"

#define EMMC_CLK GPIO_NUM_16
#define EMMC_CMD GPIO_NUM_15
#define EMMC_DATA0 GPIO_NUM_4
#define EMMC_DATA1 GPIO_NUM_5
#define EMMC_DATA2 GPIO_NUM_6
#define EMMC_DATA3 GPIO_NUM_7

const char *TAG = "wvr3";

sdmmc_card_t *card;

void setup(){
    esp_err_t ret;
    
    esp_vfs_fat_sdmmc_mount_config_t mount_config = {
        .format_if_mount_failed = true,
        .max_files = 5,
        .allocation_unit_size = 16 * 1024
    };

    ESP_LOGI(TAG, "Initializing SD card");
    ESP_LOGI(TAG, "Using SDMMC peripheral");
    sdmmc_host_t host = SDMMC_HOST_DEFAULT();
    sdmmc_slot_config_t slot_config = SDMMC_SLOT_CONFIG_DEFAULT();

    slot_config.width = 1;
    slot_config.clk = EMMC_CLK;
    slot_config.cmd = EMMC_CMD;
    slot_config.d0 = EMMC_DATA0;
    // slot_config.d1 = CONFIG_EXAMPLE_PIN_D1;
    // slot_config.d2 = CONFIG_EXAMPLE_PIN_D2;
    // slot_config.d3 = CONFIG_EXAMPLE_PIN_D3;

    slot_config.flags |= SDMMC_SLOT_FLAG_INTERNAL_PULLUP;

    gpio_pullup_en(EMMC_CLK);
    gpio_pulldown_dis(EMMC_CLK);
    gpio_pullup_en(EMMC_CMD);
    gpio_pulldown_dis(EMMC_CMD);
    gpio_pullup_en(EMMC_DATA0);
    gpio_pulldown_dis(EMMC_DATA0);
    gpio_pullup_en(EMMC_DATA1);
    gpio_pulldown_dis(EMMC_DATA1);
    gpio_pullup_en(EMMC_DATA2);
    gpio_pulldown_dis(EMMC_DATA2);
    gpio_pullup_en(EMMC_DATA3);
    gpio_pulldown_dis(EMMC_DATA3);

    ESP_LOGI(TAG, "Mounting filesystem");
    ret = esp_vfs_fat_sdmmc_mount("/root", &host, &slot_config, &mount_config, &card);

    if (ret != ESP_OK) {
        if (ret == ESP_FAIL) {
            ESP_LOGE(TAG, "Failed to mount filesystem. "
                     "If you want the card to be formatted, set the EXAMPLE_FORMAT_IF_MOUNT_FAILED menuconfig option.");
        } else {
            ESP_LOGE(TAG, "Failed to initialize the card (%s). "
                     "Make sure SD card lines have pull-up resistors in place.", esp_err_to_name(ret));
        }
        return;
    }
    ESP_LOGI(TAG, "Filesystem mounted");

    // Card has been initialized, print its properties
    sdmmc_card_print_info(stdout, card);
}

void loop(){
    ESP_LOGI(TAG, "loop");
    vTaskDelay(1000);
}
```

but unfortunately i still get
```
E (38) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
E (38) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
```

Have I missed something? Any ideas?
thank you :)

marchingband, 2023-09-27:
And I can confirm that [this](https://github.com/espressif/arduino-esp32/issues/7405#issuecomment-1294482335) code does work in esp-idf so the hardware is good.

marchingband, 2023-09-27:
I also tried SDMMC_Test.ino in the examples folder with the same failure.

marchingband, 2023-09-27:
My hardware does not have pullups, is it possible that Arduino is refusing to let me use the internal pullups?

marchingband, 2023-10-05:
Is there a way to reopen this issue or should I open a new one?

P-R-O-C-H-Y, 2023-10-05:
@marchingband Can you please retest with 2.0.14 release?

marchingband, 2023-10-06:
@P-R-O-C-H-Y I tried 2.0.14 with the same result, please re-open, thanks

marchingband, 2023-10-14:
@P-R-O-C-H-Y @VojtechBartoska @me-no-dev @igrr 
Can someone reopen this please?

P-R-O-C-H-Y, 2023-10-16:
@marchingband Hi, thanks for reporting that. I will take a look and give it a try.

marchingband, 2023-10-27:
@P-R-O-C-H-Y Any progress? Is there anything I can help with? I have a board here with eMMC that works under esp-idf, and fails in Arduino. Happy to contribute any way I can. I would like to see this move along, I have a commercial project that has been on hold for over a year waiting for this patch.

marchingband, 2023-10-27:
[here](https://github.com/espressif/esp-idf/issues/8257#issuecomment-1117738699) is the patch

P-R-O-C-H-Y, 2023-11-27:
Hi @marchingband, sorry for no reply for some time. I will get the eMMC so I can test it.
If it is possible for you, may I ask you to give it a try on 3.0.0-alpha2 version? It will help a lot. Thanks

marchingband, 2023-11-27:
@P-R-O-C-H-Y sure! Are there instructions on how to instal it?

marchingband, 2023-11-27:
@P-R-O-C-H-Y my apologies, the error was in my code, I had the wrong pins selected in the Arduino version I was testing.
3.0.0-alpha2 -> works. 
2.0.13 -> also works

closed 2023-11-27 by marchingband

P-R-O-C-H-Y, 2023-11-30:
@marchingband No problem, thanks a lot for testing on both versions! I am happy to see that it works!