espressif/arduino-esp32 #7406: WiFi error when processing STA_DISCONNECTED event with reason 0

Kuenlun, 2022-10-27:
### Board

esp32-s3-devkitc-1

### Device Description

Bare esp32-s3-devkitc-1 board connected through micro-USB cable (on the UART port) to see serial monitor.

### Hardware Configuration

No other hardware is attached to the development board.

### Version

v2.0.4

### IDE Name

PlatformIO

### Operating System

Microsoft Windows 10 Home (10.0.19045 compilation 19045)

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

460800

### Description

I've been developing a WiFi keep alive FreeRTOS task implemented with events for maximum eficiency (as much as I can get with the Arduino framework).
Each time I get "ARDUINO_EVENT_WIFI_STA_DISCONNECTED" event, I try to reconnect in two different ways depending on the reason:
1. If the reason of the disconnection is "ASSOC_FAIL" probably my router is the problem and it won't let me connect for 60 seconds, so I wait.
[Extra info]
As my router won't let me connect for 60 seconds and the purpose of the task im developing is to keep the WiFi alive, I have disabled WiFi reconnection: "WiFi.setAutoReconnect(false);"
If I don't do this I get lots of warnings and errors each time WiFi tries to reconnect (each 100 ms). With WiFi reconnection disabled I only get it twice (because even with WiFi reconnection disabled, the ESP tries to reconnect once).
`[   181][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY`
`[   219][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START`
`E (383) wifi:Set status to INIT`
`[   270][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED`
`[   270][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 203 - ASSOC_FAIL`
`[   277][D][WiFiGeneric.cpp:966] _eventCallback(): WiFi Reconnect Running`
`[   293][D][TaskWiFi.cpp:101] WiFiKeepAliveTask(): Connecting to CETEM_nave`
`E (484) wifi:Set status to INIT`
`[   371][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED`
`[   371][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 203 - ASSOC_FAIL`
`[   378][D][WiFiGeneric.cpp:975] _eventCallback(): WiFi AutoReconnect Running`
`[   390][W][TaskWiFi.cpp:40] WiFiCallBack(): ARDUINO_EVENT_WIFI_STA_DISCONNECTED`
`[   394][D][TaskWiFi.cpp:101] WiFiKeepAliveTask(): Connecting to CETEM_nave`
`E (586) wifi:Set status to INIT`

2. Whatever other reason the WiFi got disconnected, I try to reconnect it immediately.

The problem arises when very rarely (sometimes I have to reboot the board 50 times) Arduino gets the "ARDUINO_EVENT_WIFI_STA_DISCONNECTED" event with reason "0" (it isn't even listed in "arduino-esp32/blob/master/tools/sdk/esp32s3/include/esp_wifi/include/esp_wifi_types.h"):
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;`typedef enum {`
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;`    WIFI_REASON_UNSPECIFIED              = 1,`
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;`    WIFI_REASON_AUTH_EXPIRE              = 2,`
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;...
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;`    WIFI_REASON_ASSOC_FAIL               = 203,`
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;`    WIFI_REASON_HANDSHAKE_TIMEOUT        = 204,`
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;`    WIFI_REASON_CONNECTION_FAIL          = 205,`
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;`    WIFI_REASON_AP_TSF_RESET             = 206,`
[... 168 lines cut ...]
{
    // This task has been deleted
    // Code here won't run
}
```

### Debug Message

```plain
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400556b4  PS      : 0x00060c30  A0      : 0x8205ea4c  A1      : 0x3fce6b10  
A2      : 0x00000001  A3      : 0xfffffffd  A4      : 0x000000ff  A5      : 0x0000ff00
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x3fce6d13  A12     : 0x00000000  A13     : 0x00000000
A14     : 0x00060e20  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000001  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffc

Backtrace:0x400556b1:0x3fce6b10 |<-CORRUPTED
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-10-27:
Just to clarify, you are getting this STA_DISCONNECTED at boot, before the device has ever connected to an AP?  I don't think this is coming from IDF, AFAICT that event code is an arduino thing.  Since the station has not actually disconnected, I think this is probably an erroneous pathway (hence the reason 0), and tightening the logic there should help out a lot of connection issues.

mrengineer7777, 2022-10-28:
@Kuenlun Excellent analysis.  Accessing array element system_event_reasons[-1] would indeed cause a crash.
Since the crash happens in reason2str(), I suggest a one line patch would be safer and easier than modifying 
> [ "system_event_reasons" array, also to the "esp_wifi_types.h" "wifi_err_reason_t" enum, and changing the "reason2str" macro to.

I would change
```
   uint8_t reason = event->event_info.wifi_sta_disconnected.reason;
   log_w("Reason: %u - %s", reason, reason2str(reason));
```
To
```
   uint8_t reason = event->event_info.wifi_sta_disconnected.reason;
   if(!reason) reason = 1;  //Reason 0 causes crash.  Use reason 1 "UNSPECIFIED".
   log_w("Reason: %u - %s", reason, reason2str(reason));
```

Why IDF returns 0 is an issue for someone to look into.  As you noted that is not a valid reason in "esp_wifi_types.h".
Event chain:
[WiFiGenericClass line 580]  esp_event_handler_instance_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &_arduino_event_cb, NULL, NULL))
> [WiFiGenericClass line 330]  _arduino_event_cb()
>> WiFiGenericClass::_eventCallback()

Be careful with xTimerCreate().  That is an ESP-IDF native call.  The Arduino core uses 1 of the system timers.  It's possible to break that timer and cause odd behaviors.  See the "Ticker" class for a safe way to fire timers in Arduino.

Possibly related: https://github.com/espressif/arduino-esp32/pull/7344

PR #7414 (2022-11-02): Fixed #7406 crash on WiFi STA_DISCONNECTED event with reason 0
## Description of Change
Fixed #7406 . The "reason2str" macro in WiFiGeneric.cpp tries to read memory from index "-1"  in "system_event_reasons" array when handling STA_DISCONNECTED event with reason 0. Dealing with reason 0 as a reason 1 (WIFI_REASON_UNSPECIFIED) will solve the problem (the reason for this event to arrive with reason 0 is unknown).
This proposed change has succesfuly solved the ESP32 Guru Meditation Error when hanling STA_DISCONNECTED event with reason 0.

## Tests scenarios
It has been tested on the ESP32 DevKitC [... cut]
commit f4d8a481 (2022-11-02): Fixed #7406 crash on WiFi STA_DISCONNECTED event with reason 0 (#7414)

Fixed #7406 . The "reason2str" macro in WiFiGeneric.cpp tries to read memory from index "-1"  in "system_event_reasons" array when handling STA_DISCONNECTED event with reason 0. Dealing with reason 0 as a reason 1 (WIFI_REASON_UNSPECIFIED) will solve the problem (the reason for this event to arrive with reason 0 is unknown). #7406
closed 2022-11-02 by me-no-dev