espressif/arduino-esp32 #7408: 能否支援UART DMA 傳送 & 接收

CYE-GitENG, 2022-10-28:
### Related area

支援UART DMA 發送與接收

### Hardware specification

ESP32-S3

### Is your feature request related to a problem?

是 非常相關

### Describe the solution you'd like

架構:
ESP32-S3 透過 UART 傳遞 資料給STM32FXX

使用STM32FXX UART(Rx Tx) + DMA ，資料收發完只產生一次中斷
那如果ESP32-S3 可以也像STM32一樣那就很完美了。

### Describe alternatives you've considered

有考慮 SPI +DMA方案  但是網路提供的lib編譯根本不會過....
這真的有驗證過麼....還是ESP32-S3 還沒支援..

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2022-10-28:
@CYE-GitENG - 
This issue doesn't look to be related to Arduino API.
Arduino HardwareSerial is implemented in the top of IDF UART driver. 

Looking quickly at [UART IDF Code](https://github.com/espressif/esp-idf/blob/master/components/driver/uart.c), it can see that it is Interrupt driven, but not DMA/GDMA driven.

You may want to ask about it in the IDF github or in the ESP Forum.

Based on the ESP32-S3 TRM, chapter 26.4.10, UART can use GDMA, but as said, we won't add it as feature because we rely on the IDF UART Driver.

CYE-GitENG, 2022-10-31:
@SuGlider
Sure,I got it.
Thank you~

closed 2022-10-31 by SuGlider