espressif/arduino-esp32 #7409: Problem with WiFi while BlueTooth ON

avryux, 2022-10-28:
### Board

NodeMCU ESP32-S V1.1

### Device Description

TX's ESP32 is connected to the RX PIN of Arduino

### Hardware Configuration

Partition Scheme: NO OTA (2MB APP /2MB SPIFFS)

### Version

v1.0.6

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

115200

### Description

I'm working into a project with an ESP32 and an Arduino. I need WiFi to connect the ESP32 to my Firebase RTDB but i have some problems when i try to connect it as long as the BL is ON. 
If i remove the SerialBT.begin() the WiFi works correctly and i can get the data in the DB, but if i turn ON the BL stop working

### Sketch

```cpp
#include "BluetoothSerial.h"
#include <Arduino.h>
#include <WiFi.h>
#include <Firebase_ESP_Client.h>
#include "time.h"

//BL
#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif

BluetoothSerial SerialBT;

void setup() {
  Serial.begin(115200);
  
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(300);
  }
  Serial.println(WiFi.localIP());
  config.api_key = API_KEY;
  config.database_url = DATABASE_URL;
  if (Firebase.signUp(&config, &auth, "", "")) {
    Serial.println("ok");
    signupOK = true;
  }
  else {
    Serial.printf("%s\n", config.signer.signupError.message.c_str());
  }
  config.token_status_callback = tokenStatusCallback; //see addons/TokenHelper.h
  Firebase.begin(&config, &auth);
  Firebase.reconnectWiFi(true);
  configTime(gmtOffset_sec, daylightOffset_sec, ntpServer);
  delay(4000);
  SerialBT.begin("VLOCK"); //Bluetooth device name
}
```

### Debug Message

```plain
[E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():216]: (-10368) X509 - Allocation of memory failed
[E][WiFiClientSecure.cpp:133] connect(): start_ssl_client: -10368
send request failed
[E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():216]: (-10368) X509 - Allocation of memory failed
[E][WiFiClientSecure.cpp:133] connect(): start_ssl_client: -10368
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2022-10-28:
I recommend using the latest version (2.0.5) of arduino-esp32.  1.0.6 is no longer supported.

That said, I've read there are issues running BT and WIFI at the same time.  Possibly because the two share the same antenna connection.  Not sure, haven't tried it yet.  I'd be interested to hear your experience!

avryux, 2022-10-31:
i'll try turn off BL when i need use WiFi and vice versa
Thanks!

closed 2022-10-31 by avryux