espressif/arduino-esp32 #7411: RainMaker : Can we send updates to client mobile App from the node

atanudasgupta, 2022-10-30:
### Related area

Node to Client App communication

### Hardware specification

ESP32

### Is your feature request related to a problem?

I am redeveloping my automation, which I had earlier done in Blynk.  At present I can switch one a device for say 10 minutes and as time progresses I can send time remaining back to my mobile app, so that the user know how much time is remaining.

For example : I switch one a device for 10 minutes and I expect it to switch off after 10 minutes, the node(esp32), send the app updates in terms of how much time is remaining , till it becomes zero.

Is it possible to do similar with RainMaker?

### Describe the solution you'd like

I could not find any specific documentation around Node->Client communication. Mostly it is Node->RainMaker Cloud.

The option to update a virtual device from the node is required

### Describe alternatives you've considered

No alternatives at this point, since I plan to move to rainMaker from Blynk. I like Rainmaker very much

Attached is screenshot from Blynk which shows time remaining

### Additional context

[image]

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2022-10-30:
@sanketwadekar - Do you want to comment?

sanketwadekar, 2022-10-31:
Hi @atanudasgupta,
Rainmaker has APIs for node-to-cloud communication. You'll need to create a parameter to track the time remaining and call `updateAndReport()` API to send its value from the device to the cloud/client. You can refer to the following snippet.
```
Param time_remaining = Param("Time Remaining", NULL, value(10), PROP_FLAG_READ | PROP_FLAG_WRITE);
time_remaining.addUIType(ESP_RMAKER_UI_TEXT);
my_switch.addParam(time_remaining);
..
..
time_remaining.updateAndReport(value(<new-value>)); // This API will update the parameter value and report the updated value to the cloud.
```

You can read the API Docs [here](https://github.com/espressif/arduino-esp32/blob/master/libraries/RainMaker/README.md)

atanudasgupta, 2022-10-31:
Thank you Sanket, I will try it. From what you described below, I guess it will work.

Is it fair to assume that rainmaker cloud and the client app is always in sync once updateAndReport() is called..?Regards,Atanu

Thank you,Atanu

atanudasgupta, 2022-11-19:
resolved thank you

closed 2022-11-19 by atanudasgupta