espressif/arduino-esp32 #7416: Built in configTime() has no effect when setting DST.

zekageri, 2022-10-31:
### Board

ESP32-Wrover-E

### Device Description

SPI,I2C

### Hardware Configuration

Eth_Lan8720, DS3231 etc...

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

80mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Built in `configTime(gmtOffset_sec, daylightOffset_sec, ntpServer);` does not count with DST offset.

I'm from hungary and there was a winter time set at last sunday which is +1 hour to GMT.
My sketch is configured with that in mind. I tought that the built in configTime will change according to my dst since the configTime `daylightOffset_sec` param is required. But it does not happen.

I have three user defined variables in my config file.

1. timeZone ( user can select in a select box. type is int )
2. timeShift ( user can write the dst time shift in seconds. 3600 for example )
3. summerTime ( user can toggle it either to false or true. represents dst )

I could check my `summerTime` variable and set my own time accordingly however.

### Sketch

```cpp
// config.timeZone is currently +1 since hungary is at +1
// config.timeShift is currently 1 since we have 1 hour dst
int gmtOffset_sec = config.timeZone * 60 * 60;
int daylightOffset_sec = config.timeShift * 60 * 60;
configTime(gmtOffset_sec, daylightOffset_sec, ntpServer);
```

### Debug Message

```plain
-1 hour of the current.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

zekageri, 2022-10-31:
It looks like i need to do it like this:

```cpp
int gmtOffset_sec = config.timeZone * 60 * 60;
int daylightOffset_sec = config.timeShift * 60 * 60;
if( !config.summerTime ){ daylightOffset_sec = 0; }  // Must set it to 0 if we do not have summer time.
configTime(gmtOffset_sec, daylightOffset_sec, ntpServer);
```

**With this approach i must reinitialize the ntp time if the user unchecks or checks the summerTime.**

mrengineer7777, 2022-11-03:
> With this approach i must reinitialize the ntp time if the user unchecks or checks the summerTime.

@zekageri Partly true.  configTime has no idea when DST starts/ends.  It only knows daylightOffset_sec.
You can change the timezone without restarting SNTP by calling:
`setTimeZone(-gmtOffset_sec, daylightOffset_sec);`
This is the same function that configTime() uses to set the offset parameters.
Note that configTime() is declared in Arduino.h (odd place), and I'm not sure setTimeZone() is exported at all.  You can import it yourself with the following declaration in one of your headers:
`extern void setTimeZone(long offset, int daylight);`

lbernstone, 2022-11-03:
Using timezones is a much easier way to deal with daylight savings.
https://github.com/lbernstone/setTZ

zekageri, 2022-11-03:
Thank you for the clarifications.
I will test these and will come back with results!

VojtechBartoska, 2022-12-15:
any updates @zekageri?

zekageri, 2022-12-15:
Sorry I had no time to try it. I must create the appropriate front end user interface to be able to try it exactly. But right now, i must reconfigure my time if the fat changes. I did not try this either. I wonder if the reconfig is working at all.

VojtechBartoska, 2023-02-16:
Hello, closing this issues as expired due to no answer.

closed 2023-02-16 by VojtechBartoska