espressif/arduino-esp32 #7417: Installation stops with error 13 - access denied

knherzog, 2022-10-31:
### Board

ESP32 DEV Module

### Device Description

not attched yet

### Hardware Configuration

no

### Version

v2.0.5

### IDE Name

Arduino IDE 2.0.1

### Operating System

Windows 10

### Flash frequency

none yet

### PSRAM enabled

yes

### Upload speed

none yet

### Description

When trying to install esp32 as an admin in the Arduino IDE I get an internal error

Error: 13 INTERNAL: Cannot install tool esp32:riscv32-esp-elf-gcc@gcc8_4_0-esp-2021r2-patch3: moving extracted archive to destination dir: rename C:\Users\Knut\AppData\Local\Arduino15\tmp\package-1577178718\riscv32-esp-elf c:\Users\Knut\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch3: Access is denied.

I run this acouple of times, making sure I am admin. i also checked the directories. they look fine. There should not be any access problem.

Any help is much appreciated.

### Sketch

```cpp
none yet
```

### Debug Message

```plain
Error: 13 INTERNAL: Cannot install tool esp32:riscv32-esp-elf-gcc@gcc8_4_0-esp-2021r2-patch3: moving extracted archive to destination dir: rename C:\Users\Knut\AppData\Local\Arduino15\tmp\package-1577178718\riscv32-esp-elf c:\Users\Knut\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2-patch3: Access is denied.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2022-10-31:
This sounds like a Windows file permission issue.  It fails on the file rename.  The installer is Arduinio IDE, so I would seek help there: https://forum.arduino.cc/

knherzog, 2022-10-31:
Yes, it was a general issue. The antivirus program killed the installation. When I paused antivirus the installation succeeded.

closed 2022-10-31 by knherzog