espressif/arduino-esp32 #7419: Request getBoardDetails failed with message: 14 UNAVAILABLE: No connection established

songdf666, 2022-11-01:
### Board

esp32 dev module

### Device Description

[image]

### Hardware Configuration

no

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino 2.0.1

### Operating System

macOS12.4

### Flash frequency

...

### PSRAM enabled

no

### Upload speed

...

### Description

Request getBoardDetails failed with message: 14 UNAVAILABLE: No connection established

### Sketch

```cpp
..
```

### Debug Message

```plain
..
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-11-01:
Please use latest actual Arduino Core v2.0.5

me-no-dev, 2022-11-01:
Issue report is not very well filed, but I think the issue is that the board menu item "Get Board Info" does not work?

9trebovirhome, 2022-11-02:
It is probably not relevant in this repo, but I have the same issue with ESP8266.

panciunio, 2022-11-02:
Issue approaching under Linux MInt 21 and ESP 2.0.5 library installed. Native esp idf.py working fine...

[image]

rucko24, 2022-11-05:
> Please use latest actual Arduino Core v2.0.5

Where is this version available ?

SuGlider, 2022-11-05:
> > Please use latest actual Arduino Core v2.0.5
> 
> Where is this version available ?

It is in the releases: https://github.com/espressif/arduino-esp32/releases/tag/2.0.5
It can be obtained by updating the ESP32 core version using IDE->menu->tools->board manager

SuGlider, 2022-11-05:
@songdf666 - My suggestion is to update the IDE to version 2.0.1, recently launched. It may solve this issue.

emrullahjack, 2022-11-12:
I'm having the same issue with an esp32 based board namely "deneyap kart". My OS is linux based and pyserial is already installed. Arduino Core v2.0.5 is also installed.

yoann-dufresne, 2022-11-14:
I have exactly the same error in popOS

DonFruendo, 2022-11-15:
Same here, PopOS 22.04, Arduino 2.0.2-nightly20221115, trying to run sketches on my Arduino Nano

KolleparaBharadwaj, 2022-11-16:
restart the IDE and the computer, it solved the issue for me.

yoann-dufresne, 2022-11-17:
Still not solved for me

DonFruendo, 2022-11-17:
> Still not solved for me

Did you try to revert to an older version and restart your system?

emrullahjack, 2022-11-17:
[This issue](https://github.com/arduino/arduino-cli/issues/1970) might help some of the folks but didn't solve the problem for me.

VojtechBartoska, 2022-12-15:
closing as solved.

closed 2022-12-15 by VojtechBartoska