espressif/arduino-esp32 #7420: Are Wifi crendentials stored in an unsecure way on ESP32?

codingABI, 2022-11-01:
### Board

ESP-WROOM-32 NodeMCU (ESP32 Dev Board)

### Device Description

Only a plain ESP-WROOM-32 

### Hardware Configuration

Only a plain ESP-WROOM-32

### Version

v2.0.5

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Everybody how can upload a sketch to an ESP32 can read the previous used Wifi credentials in plain text. This makes it very easy for hackers to get my Wifi credentials, if they get USB access to one of my ESP32s. 

### Sketch

```cpp
#include <WiFi.h>
#define WIFIMAXRETRIES 30

void setup() {
  int wifiRetry = 0;

  Serial.begin(115200);

  Serial.println("Connect without credentials");
  WiFi.begin();

  while ((WiFi.status() != WL_CONNECTED) && (wifiRetry <= WIFIMAXRETRIES)) { // wait while connecting
    wifiRetry++;
    delay(500);
    Serial.print(".");
  }
  Serial.println();
  
  if (WiFi.status() == WL_CONNECTED) {
    Serial.print("Successfully connected and ESP got IP ");
    Serial.println(WiFi.localIP());
  } else Serial.println("Connection failed");
  Serial.print("SSID ");
  Serial.println(WiFi.SSID());
  Serial.print("PSK ");
  Serial.println(WiFi.psk());
}

void loop() {
}
```

### Debug Message

```plain
Connect without credentials
.
Successfully connected and ESP got IP 192.168.170.26
SSID myssid
PSK mySecretPassword1
```

### Other Steps to Reproduce

I prepared a description to reproduce on https://github.com/codingABI/Arduino-ESP32-Wifi-easy-and-unsecure

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

codingABI, 2022-11-01:
I tried **WiFi.persistent(false);** before the **WiFi.begin(SSID,PASSWORD);** and the used Wifi credentials could not be read by another sketch uploaded afterwards => That seems to solve my problem. Thank you very much!

I'm just a little surprised that this security issue doesn't show up in the ESP32 tutorials I know of. **WiFi.persistent(false);** should be much more known. I'm afraid not many ESP32 newbies like me are aware of the problem.

me-no-dev, 2022-11-02:
@codingABI if you want that to be secure, you need to look past Arduino and use ESP-IDF with flash+nvs encryption (you could still have Arduino as component in the IDF project)

codingABI, 2022-11-02:
I'll give the ESP-IDF&encryption a try when I get a chance. At the moment **WiFi.persistent(false)** is an acceptable solution for me. Thank you all.

closed 2022-11-02 by codingABI