espressif/arduino-esp32 #7423: Can't override bool verifyRollbackLater()

zekageri, 2022-11-01:
### Board

ESP32-Wrover-E

### Device Description

custom

### Hardware Configuration

custom

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

80mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Can't overwrite bool verifyRollbackLater(); for OTA verify.
If i modify 

```cpp
#ifdef CONFIG_APP_ROLLBACK_ENABLE
bool verifyOta() __attribute__((weak));
bool verifyOta() { return true; }

bool verifyRollbackLater() __attribute__((weak));
bool verifyRollbackLater() { return false; }
#endif
```

in `esp32-hal-misc.c` to

```cpp
#ifdef CONFIG_APP_ROLLBACK_ENABLE
bool verifyOta() __attribute__((weak));
bool verifyOta() { return true; }

bool verifyRollbackLater() __attribute__((weak));
bool verifyRollbackLater() { return true; }
#endif
```

it is working.

Printing:

```cpp
[System] - Ota state: ESP_OTA_IMG_PENDING_VERIFY
[System] - App is valid, rollback cancelled successfully
```

### Sketch

```cpp
// Does not called.
bool verifyRollbackLater(){
    Serial.printf("HI IAM AN OVERRIDDEN FUNCTION!");
    return true;
}

void verifyFirmware(){
    Serial.printf("[SYSTEM] - Checking firmware...\n");
    const esp_partition_t *running = esp_ota_get_running_partition();
    esp_ota_img_states_t ota_state;
    if (esp_ota_get_state_partition(running, &ota_state) == ESP_OK) {
        const char* otaState = ota_state == ESP_OTA_IMG_NEW ? "ESP_OTA_IMG_NEW"
            : ota_state == ESP_OTA_IMG_PENDING_VERIFY ? "ESP_OTA_IMG_PENDING_VERIFY"
            : ota_state == ESP_OTA_IMG_VALID ? "ESP_OTA_IMG_VALID"
            : ota_state == ESP_OTA_IMG_INVALID ? "ESP_OTA_IMG_INVALID"
            : ota_state == ESP_OTA_IMG_ABORTED ? "ESP_OTA_IMG_ABORTED"
            : "ESP_OTA_IMG_UNDEFINED";
        Serial.printf( "[System] - Ota state: %s\n",otaState);

        if (ota_state == ESP_OTA_IMG_PENDING_VERIFY) {
            if (esp_ota_mark_app_valid_cancel_rollback() == ESP_OK) {
                Serial.printf( "[System] - App is valid, rollback cancelled successfully\n");
            } else {
                Serial.printf("[System] - Failed to cancel rollback\n");
            }
        }
    }else{
        Serial.printf("[System] - OTA partition has no record in OTA data\n");
    }
}

void setup(){
    Serial.begin(115200);
    vTaskDelay(1000);
    verifyFirmware();
}

void loop(){
}
```

### Debug Message

```plain
[SYSTEM] - Checking firmware...
[System] - Ota state: ESP_OTA_IMG_VALID
```

### Other Steps to Reproduce

-

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-11-02:
can you try:
```cpp
extern "C" bool verifyRollbackLater(){
    Serial.printf("HI IAM AN OVERRIDDEN FUNCTION!");
    return true;
}
```

me-no-dev, 2022-11-02:
though `Serial` would not be available at this time, so maybe use `ets_printf("HI IAM AN OVERRIDDEN FUNCTION!\n");` instead

zekageri, 2022-11-02:
Take a look at the gitter chat: [https://gitter.im/espressif/arduino-esp32?at=5d3a08336ec2e14223ec9d24](https://gitter.im/espressif/arduino-esp32?at=5d3a08336ec2e14223ec9d24)

We discussed several things but not this. I have tried a lot of variations suggested by @chegewara and @atanisoft.
I will give this a try too

zekageri, 2022-11-02:
Well it works.

```cpp
f� IAM AN OVERRIDDEN FUNCTION!
[SYSTEM] - Verifying firmware...
[System] - Ota state: ESP_OTA_IMG_PENDING_VERIFY
[System] - App is valid, rollback cancelled successfully
```

zekageri, 2022-11-02:
Why is the extern c required?

me-no-dev, 2022-11-03:
> Why is the extern c required?

Because the function is originally defined in `.c` file, it has a "C" linkage, so it requires `extern "C"` when being implemented/overridden inside a `.cpp` file

zekageri, 2022-11-03:
Thank you for the clarification.
Issue is solved. I really appreciate your help!

closed 2022-11-03 by zekageri