espressif/arduino-esp32 #7424: Cannot connect to WiFi on ESP32-S3-DevKitC-1

sayanee, 2022-11-02:
### Board

ESP32-S3-DevKitC-1

### Device Description

Just the dev board ESP32-S3-DevKitC-1 and nothing else.

### Hardware Configuration

No

### Version

v2.0.5

### IDE Name

Arduino IDE

### Operating System

macOS 12.6.1

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Cannot connect to WiFi SSID

### Sketch

```cpp
#include <WiFi.h>

char ssid[] = "WIFI_SSID";
char pass[] = "WIFI_PASSWORD";

void setup() {
  Serial.begin(115200);
  while (!Serial) { }

  Serial.print("Attempting to connect to SSID: ");
  Serial.println(ssid);

  WiFi.useStaticBuffers(true);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, pass);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("Connected to WiFi");
  printWifiStatus();
}

void loop() {}

void printWifiStatus() {
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());

  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);

  long rssi = WiFi.RSSI();
  Serial.print("signal strength (RSSI):");
  Serial.print(rssi);
  Serial.println(" dBm");
}
```

### Debug Message

```plain
No error message. The serial monitor output:

Attempting to connect to SSID: <WIFI_SSID_NAME_REDACTED>
............................................................................................................................................................................................
```
```

### Other Steps to Reproduce

I have successfully connected to the same WiFi network with Tasmota as suggested in #6430 (although it's for ESP32-C3). [WiFiScan](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiScan/WiFiScan.ino) example code is also successful and it shows all the WiFi networks nearby!

I have also added `WiFi.useStaticBuffers(true);` and `WiFi.mode(WIFI_STA);` before `WiFi.begin(ssid, pass);` as suggested in the issue #6430.

WiFi SSID details:
![wifi-network-details](https://user-images.githubusercontent.com/413879/199391039-08e7dfaf-dc63-4550-9400-85fc35012819.png)

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-11-02:
Can you please enable Debug to Info or Verbose and try again. Should print more info on what is the WiFi doing.

sayanee, 2022-11-03:
Apologies, I completely missed the Arduino IDE Tools menu on Core Debug Level. I have also updated the original post.

Here's the output on the serial monitor in `VERBOSE` mode:

```
Attempting to connect to SSID: Things.Chinnee
[  4216][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
[  4249][V][WiFiGeneric.cpp:338] _arduino_event_cb(): STA Started
[  4251][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
[  4250][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
....[  6674][V][WiFiGeneric.cpp:360] _arduino_event_cb(): STA Disconnected: SSID: Things.Chinnee, BSSID: 00:00:00:00:00:00, Reason: 201
[  6674][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[  6682][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
[  6689][D][WiFiGeneric.cpp:966] _eventCallback(): WiFi Reconnect Running
[  6697][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
.....[  9114][V][WiFiGeneric.cpp:360] _arduino_event_cb(): STA Disconnected: SSID: Things.Chinnee, BSSID: 00:00:00:00:00:00, Reason: 201
[  9114][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[  9122][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
[  9129][D][WiFiGeneric.cpp:975] _eventCallback(): WiFi AutoReconnect Running
[  9137][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
.....[ 11557][V][WiFiGeneric.cpp:360] _arduino_event_cb(): STA Disconnected: SSID: Things.Chinnee, BSSID: 00:00:00:00:00:00, Reason: 201
[ 11557][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 11565][W][WiFiGeneric.cpp:950] _eventCallback(): Reason: 201 - NO_AP_FOUND
```

me-no-dev, 2022-11-03:
Error says that it was not able to find the AP. Maybe that AP is only 5GHz? Or SSID is wrong? And yeah, maybe it could be the `.`, though if it's valid char, it should work.

me-no-dev, 2022-11-03:
Nevermind, I see up in the initial post that it is indeed 2.4GHz. Try with SSID without the dot

sayanee, 2022-11-05:
Thanks, @me-no-dev!

I tried with temporary hotspots with my iPhone and it seems the password is the issue, not the SSID name with the `.`.
Both the password are alphanumeric with less than 30 characters 🤷🏻‍♀️

| SSID has `.` | Same password | Connect?
| ------ | ------ | ------ |
| Yes | Yes | No
| Yes | No | Yes
| No | Yes | No
| No | No | Yes

```plain
Connected to WiFi
SSID: WIFI_SSID
IP Address: 172.20.10.3
signal strength (RSSI):-37 dBm
```

I have no idea how to reproduce this bug, but I got it working and connected to WiFi. 

So I will close this issue, I guess.

closed 2022-11-05 by sayanee