espressif/arduino-esp32 #7426: Dependency on Update.h in ArduinoOTA causes build failure

elcreator, 2022-11-02:
### Board

ESP32-WROOM-32E

### Device Description

ESP32-WROOM-32E

### Hardware Configuration

nothing, compilation time error prevents uploading something to the real board

### Version

other

### IDE Name

CLion + Platformio

### Operating System

Windows 11

### Flash frequency

240MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

ArduinoOTA is not required in my project but it's added during the build

### Sketch

```cpp
[env:Debug]
platform = espressif32
board = denky32
framework = arduino
lib_deps = 
	PubSubClient
	https://github.com/espressif/arduino-esp32
```

### Debug Message

```plain
|-- ArduinoOTA @ 2.0.0+sha.1163e8a
|   |-- WiFi @ 2.0.0
|-- ESP32WebServer @ 1.0
|   |-- WiFi @ 2.0.0
|   |-- FS @ 2.0.0
|-- EEPROM @ 2.0.0
|-- WiFi @ 2.0.0
|-- NTPClient @ 3.2.0
Building in debug mode
Compiling .pio\build\Debug\lib21a\ArduinoOTA\ArduinoOTA.cpp.o
In file included from .pio/libdeps/Debug/ArduinoOTA/src/ArduinoOTA.cpp:6:
.pio/libdeps/Debug/ArduinoOTA/src/ArduinoOTA.h:6:10: fatal error: Update.h: No such file or directory

****************************************************************
* Looking for Update.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Update.h"
* Web  > https://registry.platformio.org/search?q=header:Update.h
*
****************************************************************

 #include "Update.h"
          ^~~~~~~~~~
compilation terminated.
```

### Other Steps to Reproduce

-

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

elcreator, 2022-11-02:
lib_ignore =
	ArduinoOTA

solves the issue but this bug should be fixed with default configuration

Jason2866, 2022-11-04:
The config is not valid.
```
lib_deps = 
	https://github.com/espressif/arduino-esp32
```
Thats not possible there!

closed 2023-04-13 by mrengineer7777

m-mcgowan, 2024-04-04:
I'd like to repoen this. The headers in the IDF have changed in recent releases, meaning that the OTA library can't be compiled on the espressif32 platform.