espressif/arduino-esp32 #7427: about bootloader

hzhh110, 2022-11-04:
### Related area

ESP32-WROOM-32E

### Hardware specification

ESP32-WROOM-32E

### Is your feature request related to a problem?

What is the difference between Dio _ 40 m versions? What is the function of this file? Used the old Dio to burn, it seems to be able to run normally, and then my MD5 is different, and platformio will compile a bootloader without logo.

### Describe the solution you'd like

What is the difference between Dio _ 40 m versions? What is the function of this file? Used the old Dio to burn, it seems to be able to run normally, and then my MD5 is different, and platformio will compile a bootloader without logo.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2022-11-04:
esp32 uses a 2nd stage bootloader to provide the low level drivers needed to run your firmware.  the most important of these is the SPI flash driver, so that it can read your code off the flash. https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-guides/bootloader.html
The different versions of the bootloader correspond to different types of SPI flash.  Unless you have a reason not to do so, you should use the fastest possible configuration, which is QIO @ 80MHz for the esp32.

SuGlider, 2022-11-05:
Adding a bit to the good explanation @lbernstone has done, I can say that a practical difference from DIO to QIO is that the first uses 2 GPIOs to access the flash while the second uses 4.
QIO is faster, but it uses more pins. With DIO, the pins 9 and 10 are free to be used by the application.

Thus, if flash speed is not a big deal, and number of available GPIOs is important, DIO may be a good choice.
ESP32 has a sort of cache for flash, therefore, sometimes it doesn't make too much difference using DIO and QIO for certain applications.

If necessary, a more detailed information about flash cache and performance can be read here: 
https://esp32.com/viewtopic.php?t=8492

VojtechBartoska, 2022-12-15:
@hzhh110 Can I consider this as covered?

VojtechBartoska, 2023-01-18:
Hello, I'm closing this issue as expired. If needed, you can reopen it.

closed 2023-01-18 by VojtechBartoska