espressif/arduino-esp32 #7428: ***ERROR*** A stack overflow in task BTC_TASK has been detected.

0xPr0xy, 2022-11-04:
### Board

ESP32-S3-WROOM-1-N8R8

### Device Description

ESP32-S3-WROOM-1-N8R8

Connected to a Macbook usb port through uart

### Hardware Configuration

Nothing else is connected

### Version

v2.0.5

### IDE Name

Visual Studio Code

### Operating System

MacOS 12.6.1

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

Please provide information on under what conditions this stack overflow is triggered.

`***ERROR*** A stack overflow in task BTC_TASK has been detected.`

It seems to be related to doing work inside the onWrite callback of a characteristic. if we only set a boolean and then do the work in the loop() after checking the boolean this stack overflow doesn't happen.

I suspect it can be related to the ACK of the write not being done in time due to the work? however using `PROPERTY_WRITE_NR` on the characteristic results in the onWrite callback not being triggered at all.

Furthermore. this crash was not occurring on the default partition setup of Large Factory app no OTA. it started occurring on a custom partition setup without any code changes

the setup:

```
# ESP-IDF Partition Table
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x4000,
otadata,  data, ota,     0xd000,  0x2000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000,  2M,
ota_0,    app,  ota_0,   0x210000, 2M,
ota_1,    app,  ota_1,   0x410000, 2M,
```

### Sketch

```cpp
I can provide the code privately if needed. however for now an explanation of the crash would be sufficient as I can't find any information as to what it means and what are the conditions for it to occur.
```

### Debug Message

```plain
***ERROR*** A stack overflow in task BTC_TASK has been detected.

Backtrace:0x40375d8e:0x3fcec2b00x4037de55:0x3fcec2d0 0x4038139e:0x3fcec2f0 0x4037fe63:0x3fcec370 0x4037df0c:0x3fcec390 0x4037df02:0x00000000  |<-CORRUPTED
0x40375d8e: panic_abort at /Users/peter.ijlst/esp/esp-idf/components/esp_system/panic.c:402

0x4037de55: esp_system_abort at /Users/peter.ijlst/esp/esp-idf/components/esp_system/esp_system.c:128

0x4038139e: vApplicationStackOverflowHook at /Users/peter.ijlst/esp/esp-idf/components/freertos/port/xtensa/port.c:394

0x4037fe63: vTaskSwitchContext at /Users/peter.ijlst/esp/esp-idf/components/freertos/tasks.c:3505

0x4037df0c: _frxt_dispatch at /Users/peter.ijlst/esp/esp-idf/components/freertos/port/xtensa/portasm.S:436

0x4037df02: _frxt_int_exit at /Users/peter.ijlst/esp/esp-idf/components/freertos/port/xtensa/portasm.S:231
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2022-11-04:
Without code there's no way to reproduce your issue.  Create a separate project that demonstrates the issue and post that code.

If you are creating your own thread, increase the stack size parameter on that thread.

SuGlider, 2022-11-05:
From my experience, if the callback code from `onWrite()` uses too much stack space or has some sort of recursion, it may lead to stack overflow issues. 

My suggestion is to review the code of this function and just use it to send a message to another task that will actually do all the heavy work. This task can be created with a higher stack space.

SuGlider, 2022-11-05:
Bluetooth task that treats Characteristic events run on its own task and stack size. It can't be modified using Arduino API.
Therefore, the strategy of just setting some semaphore for another task that will do the job can be the best way to go.

0xPr0xy, 2022-11-07:
Thanks for the explanation, very helpful. I have one last question: could you explain how and if the stack size for the BT task somehow dependant on the partition size the app is running on (or the remaining non partitioned space)? Since I could reproduce or fix it by going between 2 different partition layouts.  Thanks!

SuGlider, 2022-11-09:
BTC Task Size is defined in the sdkconfig file.
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32s3/sdkconfig#L884

I see that for the ESP32 it is 8192, but for the C3 and S3 it has a value of just 3072...
That may be too low. 

Could you please test your sketch with the ESP32 instead and report back telling me if it also gets a stack overflow error?
Thanks!

SuGlider, 2022-11-09:
> Thanks for the explanation, very helpful. I have one last question: could you explain how and if the stack size for the BT task somehow dependant on the partition size the app is running on (or the [...]

I see no relationship of RAM stack with a flash partition size. 
I don't know why it may seem to fix the issue in your testing.

VojtechBartoska, 2022-12-15:
Closing this as expired, if needed, you can reopen it.

closed 2022-12-15 by VojtechBartoska