espressif/arduino-esp32 #7431: ESP32-S3 touch wake up from deep sleep problem

rubenfrolic, 2022-11-06:
### Board

ESP32-S3

### Device Description

Custom PCB, based on ESP32-S3FN8 

### Hardware Configuration

GPIO11 connected to touch pad. 

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

MacOS

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

I’m trying to use touch wake up from deep sleep using an ESP32-S3 in the Arduino framework. I have a custom pcb with a touch-pad connected to GPIO11 → T11

When I run the TouchRead example code I clearly get good results: when the touch pad is not touch I get readings of ±22920 and when I touch it goes to ±90000:

```tsx
12:13:25.552 -> 22921
12:13:26.527 -> 79766 // touched
12:13:27.535 -> 89048 
12:13:28.537 -> 90099
12:13:29.539 -> 90792
12:13:30.525 -> 22917 // released
12:13:31.552 -> 22920
12:13:32.554 -> 22924
12:13:33.541 -> 22926
```

When I use the touch wake up example from DeepSleep I don’t get good results. The ESP32 goes to sleep but wakes up instantly. I tried to set the threshold at different values: 1, 60000, 100000 etc. But regardless the ESP32 keep on having the same behaviour. Below you can see the behaviour. In the timestamp you can see it wakes up instantly, while it’s not being touched. 

```tsx
12:21:17.702 -> Wakeup was not caused by deep sleep: 0
12:21:17.739 -> Going to sleep now
12:21:17.923 -> ESP-ROM:esp32s3-20210327
12:21:17.923 -> Build:Mar 27 2021
12:21:17.958 -> rst:0x5 (DSLEEP),boot:0x8 (SPI_FAST_FLASH_BOOT)
12:21:17.958 -> SPIWP:0xee
12:21:17.958 -> mode:DIO, clock div:1
12:21:17.958 -> load:0x3fce3808,len:0x43c
12:21:17.958 -> load:0x403c9700,len:0xbec
12:21:17.958 -> load:0x403cc700,len:0x2a3c
12:21:17.958 -> entry 0x403c98d8
12:21:18.978 -> Boot number: 2
12:21:18.978 -> Wakeup caused by touchpad
12:21:19.014 -> Going to sleep now
12:21:19.121 -> ESP-ROM:esp32s3-20210327
12:21:19.121 -> Build:Mar 27 2021
12:21:19.121 -> rst:0x5 (DSLEEP),boot:0x8 (SPI_FAST_FLASH_BOOT)
12:21:19.155 -> SPIWP:0xee
12:21:19.155 -> mode:DIO, clock div:1
12:21:19.155 -> load:0x3fce3808,len:0x43c
12:21:19.155 -> load:0x403c9700,len:0xbec
12:21:19.155 -> load:0x403cc700,len:0x2a3c
12:21:19.155 -> entry 0x403c98d8
12:21:20.187 -> Boot number: 3
12:21:20.187 -> Wakeup caused by touchpad
12:21:20.187 -> Going to sleep now
12:21:20.261 -> ESP-ROM:esp32s3-20210327
12:21:20.261 -> Build:Mar 27 2021
12:21:20.261 -> rst:0x5 (DSLEEP),boot:0x8 (SPI_FAST_FLASH_BOOT)
12:21:20.261 -> SPIWP:0xee
12:21:20.261 -> mode:DIO, clock div:1
12:21:20.261 -> load:0x3fce3808,len:0x43c
12:21:20.261 -> load:0x403c9700,len:0xbec
12:21:20.261 -> load:0x403cc700,len:0x2a3c
12:21:20.261 -> entry 0x403c98d8
12:21:21.286 -> Boot number: 4
12:21:21.286 -> Wakeup caused by touchpad
12:21:21.320 -> Going to sleep now
```

I attached the example code. 

I notice that the touch wake-up behaviour of the ESP32-S3 is different than previous modules. In previous module the touch value was going down when being touched, while with the S3 it’s going up. 

Is there a way to get touch Wake-up to work within the Arduino framework with the ESP32-S3?

### Sketch

```cpp
/*
Deep Sleep with Touch Wake Up
=====================================
This code displays how to use deep sleep with
a touch as a wake up source and how to store data in
RTC memory to use it over reboots

This code is under Public Domain License.

Author:
Pranav Cherukupalli <cherukupallip@gmail.com>
*/

#define Threshold 60000 /* Greater the value, more the sensitivity */

RTC_DATA_ATTR int bootCount = 0;
touch_pad_t touchPin;
/*
Method to print the reason by which ESP32
has been awaken from sleep
*/
void print_wakeup_reason(){
  esp_sleep_wakeup_cause_t wakeup_reason;

  wakeup_reason = esp_sleep_get_wakeup_cause();

  switch(wakeup_reason)
  {
    case ESP_SLEEP_WAKEUP_EXT0 : Serial.println("Wakeup caused by external signal using RTC_IO"); break;
    case ESP_SLEEP_WAKEUP_EXT1 : Serial.println("Wakeup caused by external signal using RTC_CNTL"); break;
    case ESP_SLEEP_WAKEUP_TIMER : Serial.println("Wakeup caused by timer"); break;
    case ESP_SLEEP_WAKEUP_TOUCHPAD : Serial.println("Wakeup caused by touchpad"); break;
    case ESP_SLEEP_WAKEUP_ULP : Serial.println("Wakeup caused by ULP program"); break;
    default : Serial.printf("Wakeup was not caused by deep sleep: %d\n",wakeup_reason); break;
  }
}

/*
Method to print the touchpad by which ESP32
has been awaken from sleep
*/
void print_wakeup_touchpad(){
  touchPin = esp_sleep_get_touchpad_wakeup_status();

  switch(touchPin)
  {
    case 0  : Serial.println("Touch detected on GPIO 4"); break;
    case 1  : Serial.println("Touch detected on GPIO 0"); break;
    case 2  : Serial.println("Touch detected on GPIO 2"); break;
    case 3  : Serial.println("Touch detected on GPIO 15"); break;
    case 4  : Serial.println("Touch detected on GPIO 13"); break;
    case 5  : Serial.println("Touch detected on GPIO 12"); break;
    case 6  : Serial.println("Touch detected on GPIO 14"); break;
    case 7  : Serial.println("Touch detected on GPIO 27"); break;
    case 8  : Serial.println("Touch detected on GPIO 33"); break;
    case 9  : Serial.println("Touch detected on GPIO 32"); break;
    default : Serial.println("Wakeup not by touchpad"); break;
  }
}

void callback(){
  //placeholder callback function
}

void setup(){
  Serial.begin(115200);
  delay(1000); //Take some time to open up the Serial Monitor

  //Increment boot number and print it every reboot
  ++bootCount;
  Serial.println("Boot number: " + String(bootCount));

  //Print the wakeup reason for ESP32 and touchpad too
  print_wakeup_reason();
  //print_wakeup_touchpad();

  //Setup interrupt on Touch Pad 3 (GPIO15)
  touchAttachInterrupt(T11, callback, Threshold);

  //Configure Touchpad as wakeup source
  esp_sleep_enable_touchpad_wakeup();

  //Go to sleep now
  Serial.println("Going to sleep now");
  esp_deep_sleep_start();
  Serial.println("This will never be printed");
}

void loop(){
  //This will never be reached
}
```

### Debug Message

```plain
12:21:17.702 -> Boot number: 1
12:21:17.702 -> Wakeup was not caused by deep sleep: 0
12:21:17.739 -> Going to sleep now
12:21:17.923 -> ESP-ROM:esp32s3-20210327
12:21:17.923 -> Build:Mar 27 2021
12:21:17.958 -> rst:0x5 (DSLEEP),boot:0x8 (SPI_FAST_FLASH_BOOT)
12:21:17.958 -> SPIWP:0xee
12:21:17.958 -> mode:DIO, clock div:1
12:21:17.958 -> load:0x3fce3808,len:0x43c
12:21:17.958 -> load:0x403c9700,len:0xbec
12:21:17.958 -> load:0x403cc700,len:0x2a3c
12:21:17.958 -> entry 0x403c98d8
12:21:18.978 -> Boot number: 2
12:21:18.978 -> Wakeup caused by touchpad
12:21:19.014 -> Going to sleep now
12:21:19.121 -> ESP-ROM:esp32s3-20210327
12:21:19.121 -> Build:Mar 27 2021
12:21:19.121 -> rst:0x5 (DSLEEP),boot:0x8 (SPI_FAST_FLASH_BOOT)
12:21:19.155 -> SPIWP:0xee
12:21:19.155 -> mode:DIO, clock div:1
12:21:19.155 -> load:0x3fce3808,len:0x43c
12:21:19.155 -> load:0x403c9700,len:0xbec
12:21:19.155 -> load:0x403cc700,len:0x2a3c
12:21:19.155 -> entry 0x403c98d8
12:21:20.187 -> Boot number: 3
12:21:20.187 -> Wakeup caused by touchpad
12:21:20.187 -> Going to sleep now
12:21:20.261 -> ESP-ROM:esp32s3-20210327
12:21:20.261 -> Build:Mar 27 2021
12:21:20.261 -> rst:0x5 (DSLEEP),boot:0x8 (SPI_FAST_FLASH_BOOT)
12:21:20.261 -> SPIWP:0xee
12:21:20.261 -> mode:DIO, clock div:1
12:21:20.261 -> load:0x3fce3808,len:0x43c
12:21:20.261 -> load:0x403c9700,len:0xbec
12:21:20.261 -> load:0x403cc700,len:0x2a3c
12:21:20.261 -> entry 0x403c98d8
12:21:21.286 -> Boot number: 4
12:21:21.286 -> Wakeup caused by touchpad
12:21:21.320 -> Going to sleep now
```

### Other Steps to Reproduce

Issue also occurs using the ESP32-S3 dev board

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-11-07:
Issue confirmed. This example also has problems with ESP32-S2.

P-R-O-C-H-Y, 2022-11-07:
Already working on this. Will post a PR soon with new API and updated example :)

P-R-O-C-H-Y, 2022-11-07:
@rubenfrolic New version 2.0.6 will be released soon with these changes included :)

PR #7439 (2022-11-09): Add touch sleep wakeup API
## Description of Change
As reported in the issue #7431 , the example `DeepSleep/TouchWakeUp.ino` is not working for ESP32-S2 and ESP32-S3. For these chips setting touch pad as wake up source needs different calls than ESP32. 
So this new function works with all chips.
`void touchSleepWakeUpEnable(uint8_t pin, touch_value_t threshold);`

I have updated the example to work using this new function and made it work on all chips + updated docs.

## Tests scenarios

Tested on ESP32, ESP32-S2 and ESP32-S3 with the updated `TouchWakeUp.ino` example.

## Related links
Closes [... cut]
closed 2022-11-09 by SuGlider