espressif/arduino-esp32 #7432: Add a Arduino Matter library to the road map?

Humancell, 2022-11-06:
### Related area

Arduino Library

### Hardware specification

All Boards

### Is your feature request related to a problem?

The Matter specification has been released, and there has been IDF support for quite some time. At one level is would be good to see examples of using these APIs from the Arduino environment. At another level it would be great if there was a higher level Arduino library created that leveraged these APIs and provided examples for both implementing and controlling a Matter device.

### Describe the solution you'd like

It would be great if there was a higher level Arduino library created that leveraged the IDF Matter APIs and provided examples for both implementing and controlling a Matter device.

### Describe alternatives you've considered

There are some other people outside of this project starting to experiment with this, but they are not fully open source nor sponsored by Espressif.

### Additional context

For the Espressif platform to dominate in the market for Matter devices it could be engaging the Arduino developers better.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2022-11-06:
Why does this need to be part of the core?  It can be handled in a 3rd party library.

VojtechBartoska, 2022-11-07:
Hello @Humancell,

thanks for opening this feature Request.

I'm adding this to next major milestone 3.0.0 and we will evaluate it.

jakubdybczak, 2022-11-10:
You can try library I have created few days ago https://github.com/jakubdybczak/esp32-arduino-matter

Humancell, 2022-11-11:
Hello,

> You can try library I have created few days ago https://github.com/jakubdybczak/esp32-arduino-matter

I did see this library, but from my understanding you have not made this 100% open source, which is a problem for us.  We can not use any closed source code in our projects.

If I am incorrect, and this really is 100% open source then we could take a look.

jakubdybczak, 2022-11-11:
@Humancell 
Why do you think that this one is not open-source?
[arduino-esp32](https://github.com/espressif/arduino-esp32) also contains precompiled libraries and scripts that do this are in [esp32-arduino-lib-builder](https://github.com/espressif/esp32-arduino-lib-builder) repo.
Same with [esp32-arduino-matter](https://github.com/jakubdybczak/esp32-arduino-matter), you can find scripts used to build it there [esp32-arduino-matter-builder](https://github.com/jakubdybczak/esp32-arduino-matter-builder).

SuGlider, 2022-11-21:
We will consider this request for next releases in 2023... please be patient.

SuGlider, 2022-12-19:
I have done an initial effort to build Matter and Arduino as Component.
The steps are here:

https://github.com/espressif/esp-matter/issues/116#issuecomment-1356673244

HamzaHajeir, 2024-05-26:
Hi all

Is there any progress for supporting Matter to the Arduino core?

SuGlider, 2024-06-12:
@HamzaHajeir - We are currently working to release OpenThread as an Arduino Library. This shall be done later this month.
Next step is to work on releasing Matter as an Arduino Library.
I believe that it will be done by the end of this year.

SuGlider, 2024-06-12:
About Matter Library implementation for Arduino Core 3.0.x:

- We will initially implement it to be supported within WiFi. In the future also within Thread Protocol.
- WiFi Matter will be supported for ESP32, ESP32-C3, ESP32-S3, ESP32-S2 and ESP32-C6. No support for ESP32-H2 and ESP32-P4, which have no WiFi radio. ESP32-S2 has no BLE, it won't have Matter Device BLE advertising service.
- Thread Network Protocol will be supported for the ESP32-C6 and ESP32-H2. Both have also BLE, therefore, BLE Commissionable discovery is also available.
- Commissionable discovery will be done through BLE (whenever available in the SoC), WiFi SoftAP (future feature), DNS-SD for Thread Border Routers (future feature).
- Commissioning will follow Matter standard, including QR Code generated from a URL printed in the log output.
- ESP32 Matter implementation for BLE Commissionable discovery is based on NimBLE. Therefore, it won't be compatible with ESP32 BLE Arduino Library that uses Bluedroid instead. This means that there may be issues when using both together in the same Arduino Sketch. This will be a limitation. We are planning to move ESP32 Arduino BLE Library to use NimBLE as well in the future, which may solve this incompatibility problem.
- ESP32 Matter Arduino implementation will make available all possible Matter 1.0 device types as Arduino Classes: Lighting, Smart Plugs, Switches and Controls, Sensors, Closures and HVACs. Each Device Type Class will have necessary implementation of the device data model, endpoints, roles, clusters, commands and events.
- The current implementation is only for Matter Device. No plans for Matter Controllers or Matter Commissioner library.
- This implementation will use testing and development Matter Vendor ID (0xFFF1..4), therefore valid for CSA for testing purpose only. In case you need your own VID, please follow CSA processes and add certificates as necessary to your code, on your own.

HamzaHajeir, 2024-06-12:
> About Matter Library implementation for Arduino Core 3.0.x:
>
> * We will initially implement it to be supported within WiFi. In the future also within Thread Protocol.
> * WiFi Matter will be supported [...]

Glad to hear that's coming support of Matter in the Arduino core. I have just some questions and comments:
- ESP-IDF identifies several tools related to setup Matter device (as esp-matter-mfg-tool), are these included also?
- Per Commissioning, please consider On-Network commisioning method, in which some could have his own device on the network previously, by either Ethernet or custom implementation of WiFi-AP.
- Therefore, I think it's okay to have ESP32-S2 supported with respect to such user responsibility (or even with the feature of WiFiAP commissioning).
- Is it available for some to build a custom device using matter data and interaction model?

With thanks,
Hamza Hajeir

SuGlider, 2024-06-12:
@HamzaHajeir - Thanks for the comments.

> ESP-IDF identifies several tools related to setup Matter device (as esp-matter-mfg-tool), are these included also?

Not as part of the tools that ESP32 Arduino would provide.
These tools, as the CHIP tools, or any other Python tools, can be used independently of the ESP32 SDK used to build Matter Applications. These tools can be used while using IDF, Rust or Arduino Frameworks for Matter development.

>Per Commissioning, please consider On-Network commisioning method, in which some could have his own device on the network previously, by either Ethernet or custom implementation of WiFi-AP.

Commissioning will always be done on-network.
The device must be already on the network and WiFi/Thread credentials shall be already in place.
Ethernet must have the cable connected. 
In all cases, the device must have a valid IP address (DHCP or Static IP).

Therefore, On-Network Commissioning method will be included.

>Therefore, I think it's okay to have ESP32-S2 supported with respect to such user responsibility (or even with the feature of WiFiAP commissioning).

Yes, correct.

> Is it available for some to build a custom device using matter data and interaction model?

We may consider it for future feature. At first, we'll focus on the Matter 1.0 device types as described above.
But given that ESP32 Arduino is based on IDF and ESP-Matter, it is always possible to add code from those two to an Arduino Sketch and We may consider it for future feature. At first, we'll focus on the Matter 1.0 device types as described above.
But given that ESP32 Arduino is based on IDF and ESP-Matter, it is always possible to add code from those two to an Arduino Sketch and create custom devices, as well as custom interaction models.

HamzaHajeir, 2024-06-15:
@SuGlider Nice, that explains all, I was mixing between commissioning discovery and commissioning process, seems two different operations.

However, it seems that all [Factory Data Providers](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#factory-data-providers) would be 'custom' in Arduino, correct?

SuGlider, 2024-06-17:
> However, it seems that all [Factory Data Providers](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#factory-data-providers) would be 'custom' in Arduino, correct?

Arduino will set the Matter Device as a Testing unit using a VIDs allocated by the CSA for testing purposes.
https://developers.home.google.com/matter/get-started#vendor_id
No Real Provider information would be used within Arduino. 

Commissioning will happen as defined in the description from https://developers.home.google.com/matter/primer/commissioning

Discovery may be done using BLE and/or WiFi AP, as described here: https://developers.home.google.com/matter/primer/commissionable-and-operational-discovery

HamzaHajeir, 2024-06-18:
> Arduino will set the Matter Device as a Testing unit using a VIDs allocated by the CSA for testing purposes. https://developers.home.google.com/matter/get-started#vendor_id No Real Provider [...]

But this might contradict with this (regarding the providers), where it should be a custom one if someone would use the Arduino:
> * This implementation will use testing and development Matter Vendor ID (0xFFF1..4), therefore valid for CSA for testing purpose only. In case you need your own VID, please follow CSA processes and add certificates as necessary to your code, on your own.

What I could suggest is making it custom with overridable test providers: defaulted to a derived class of such, that points to test information.

> Commissioning will happen as defined in the description from https://developers.home.google.com/matter/primer/commissioning
> 
> Discovery may be done using BLE and/or WiFi AP, as described here: https://developers.home.google.com/matter/primer/commissionable-and-operational-discovery

I'll read about them soon and check if any suggestion or comment I can assist with.

dzungpv, 2024-08-17:
@SuGlider I try to build with 3.0 as component and latest esp matter but it not working, show error:
```
build/esp-idf/chip/gen/include/system/SystemBuildConfig.h:31:38:fatal error: platform/1/CHIPPlatformConfig.h: No such file or directory
   31 | #define CHIP_PLATFORM_CONFIG_INCLUDE <platform/ESP32/CHIPPlatformConfig.h>

```

HamzaHajeir, 2024-08-25:
> @SuGlider I try to build with 3.0 as component and latest esp matter but it not working, show error:
>
> ```
> build/esp-idf/chip/gen/include/system/SystemBuildConfig.h:31:38:fatal error: [...]

I think that goes after substituting ESP32 to its preprocessor definition 1. Therefore I think you'll need to define CHIP_PLATFORM_CONFIG_INCLUDE within CMakeLists definitions.

SuGlider, 2024-08-28:
> > > @SuGlider I try to build with 3.0 as component and latest esp matter but it not working, show error:
> > > ```
> > > build/esp-idf/chip/gen/include/system/SystemBuildConfig.h:31:38:fatal error: platform/1/CHIPPlatformConfig.h: No such file or directory
> > >    31 | #define CHIP_PLATFORM_CONFIG_INCLUDE <platform/ESP32/CHIPPlatformConfig.h>
> > > ```
> > 
> > 
> > I think that goes after substituting ESP32 to its preprocessor definition 1.
> > Therefore there should be the compile-time definition of ESP32. What steps you've followed to reach to this point?
> 
> I just using Matter sample light project and add Arduino as a components

### Matter + Arduino as IDF Component:

My suggestion is using Matter from the ESP Registry (Managed Components) 
https://components.espressif.com/components/espressif/esp_matter/versions/1.3.0

Start with this example:
https://github.com/espressif/esp-matter/tree/f4eaa699f0fd8433bc5ca134328934f68f1e1525/examples/managed_component_light

Build it as it is. It will work as a Matter Light with 5 different White tones plus a Dimmer.
I have commissioned and tested it with  the Alexa APP and it works fine (it requires an Alexa device), but it shall also work with Apple Home, or using Google HA - after setting the google applications to accept matter development accessories.

After making it work with your target ESP32 SoC, you can add `/components/arduino` or use the Arduino from Registry and modify the source code from the `\main` folder to add your Arduino Code.

SuGlider, 2024-08-28:
OK, using Registry, I also got the same error.
Let me check how to solve it.

dzungpv, 2024-08-28:
> OK, using Registry, I also got the same error. Let me check how to solve it.

Yes, hope you will find solution.

SuGlider, 2024-08-28:
@dzungpv 

I have found the origin of the problem...

Arduino Component has a `-DESP32`, which means `#define ESP32 1`
CHIP (ConnetedHome) has a `#define CHIP_PLATFORM_CONFIG_INCLUDE <platform/ESP32/CHIPPlatformConfig.h>`
which ends up expanded to `#define CHIP_PLATFORM_CONFIG_INCLUDE <platform/1/CHIPPlatformConfig.h>` because of Arduino Component defines.

This is the source of the issue.
To solve it, you can change Ardunino Component CMakeLists.txt content at line 323.

```
target_compile_options(${COMPONENT_TARGET} PUBLIC
    -DARDUINO=10812
    -DARDUINO_${idf_target_for_macro}_DEV
    -DARDUINO_ARCH_ESP32
    -DARDUINO_BOARD="${idf_target_caps}_DEV"
    -DARDUINO_VARIANT="${CONFIG_ARDUINO_VARIANT}"
    -DESP32)
 
```

SuGlider, 2024-08-28:
@dzungpv - You can change it to `-DESP32=ESP32` .... it shall solve the issue.

SuGlider, 2024-08-28:
@dzungpv - I have tested it. `-DESP32=ESP32` solves the issue. I'll add a PR to change it in Arduino repository.

TD-er, 2024-08-28:
Wow, nasty one.
Not easy to find the origin.

dzungpv, 2024-08-29:
> @dzungpv - I have tested it. `-DESP32=ESP32` solves the issue. I'll add a PR to change it in Arduino repository.

Yes, chang it make build sample light matter success, but other problem when I am using Arduino WiFi, the error log:
```
wind_FindEnclosingFunction  -Wl,--wrap=_Unwind_Resume  -Wl,--wrap=_Unwind_RaiseException  -Wl,--wrap=_Unwind_DeleteException  -Wl,--wrap=_Unwind_ForcedUnwind  -Wl,--wrap=_Unwind_Resume_or_Rethrow  -Wl,--wrap=_Unwind_Backtrace  -Wl,--wrap=__cxa_call_unexpected  -Wl,--wrap=__gxx_personality_v0  -u __cxa_guard_dummy  -lstdc++  esp-idf/pthread/libpthread.a  -lgcc  esp-idf/cxx/libcxx.a  -u __cxx_fatal_exception  -u include_esp_phy_override  -lphy  -lrtc  -lphy  -lrtc  esp-idf/esp_phy/libesp_phy.a  -u vfs_include_syscalls_impl  -lbtdm_app  -u ld_include_hli_vectors_bt && :
/Users/vietdzung/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/arduino-esp32/libarduino-esp32.a(WiFiGeneric.cpp.obj):(.literal._Z16wifiLowLevelInitb+0x40): undefined reference to `esp_netif_create_default_wifi_ap'
/Users/vietdzung/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/arduino-esp32/libarduino-esp32.a(WiFiGeneric.cpp.obj): in function `_Z16wifiLowLevelInitb':
/Users/vietdzung/dev/projects/esp32/sdks/5.1.4/arduino-esp32/libraries/WiFi/src/WiFiGeneric.cpp:178: undefined reference to `esp_netif_create_default_wifi_ap'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

```

SuGlider, 2024-08-29:
> > @dzungpv - I have tested it. `-DESP32=ESP32` solves the issue. I'll add a PR to change it in Arduino repository.
> 
> Yes, change it make build sample light matter success, but other problem when I am using Arduino WiFi, the error log:
> 
> ```
> wind_FindEnclosingFunction  -Wl,--wrap=_Unwind_Resume  -Wl,--wrap=_Unwind_RaiseException  -Wl,--wrap=_Unwind_DeleteException  -Wl,--wrap=_Unwind_ForcedUnwind  -Wl,--wrap=_Unwind_Resume_or_Rethrow  -Wl,--wrap=_Unwind_Backtrace  -Wl,--wrap=__cxa_call_unexpected  -Wl,--wrap=__gxx_personality_v0  -u __cxa_guard_dummy  -lstdc++  esp-idf/pthread/libpthread.a  -lgcc  esp-idf/cxx/libcxx.a  -u __cxx_fatal_exception  -u include_esp_phy_override  -lphy  -lrtc  -lphy  -lrtc  esp-idf/esp_phy/libesp_phy.a  -u vfs_include_syscalls_impl  -lbtdm_app  -u ld_include_hli_vectors_bt && :
> /Users/vietdzung/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/arduino-esp32/libarduino-esp32.a(WiFiGeneric.cpp.obj):(.literal._Z16wifiLowLevelInitb+0x40): undefined reference to `esp_netif_create_default_wifi_ap'
> /Users/vietdzung/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/arduino-esp32/libarduino-esp32.a(WiFiGeneric.cpp.obj): in function `_Z16wifiLowLevelInitb':
> /Users/vietdzung/dev/projects/esp32/sdks/5.1.4/arduino-esp32/libraries/WiFi/src/WiFiGeneric.cpp:178: undefined reference to `esp_netif_create_default_wifi_ap'
> collect2: error: ld returned 1 exit status
> ninja: build stopped: subcommand failed.
> ```

@dzungpv -  What is your target SoC? Please post a sketch that I can use to test it.
Please have in mind that matter will control WiFi communication as well and it may represent a conflict.

dzungpv, 2024-08-29:
> @dzungpv - What is your target SoC? Please post a sketch that I can use to test it. Please have in mind that matter will control WiFi communication as well and it may represent a conflict.

I user Arduino sample WifiClient for ESP32

```
/*
    Go to thingspeak.com and create an account if you don't have one already.
    After logging in, click on the "New Channel" button to create a new channel for your data. This is where your data will be stored and displayed.
    Fill in the Name, Description, and other fields for your channel as desired, then click the "Save Channel" button.
    Take note of the "Write API Key" located in the "API keys" tab, this is the key you will use to send data to your channel.
    Replace the channelID from tab "Channel Settings" and privateKey with "Read API Keys" from "API Keys" tab.
    Replace the host variable with the thingspeak server hostname "api.thingspeak.com"
    Upload the sketch to your ESP32 board and make sure that the board is connected to the internet. The ESP32 should now send data to your Thingspeak channel at the intervals specified by the loop function.
    Go to the channel view page on thingspeak and check the "Field1" for the new incoming data.
    You can use the data visualization and analysis tools provided by Thingspeak to display and process your data in various ways.
    Please note, that Thingspeak accepts only integer values.

    You can later check the values at https://thingspeak.com/channels/2005329
    Please note that this public channel can be accessed by anyone and it is possible that more people will write their values.
 */

#include <WiFi.h>

const char* ssid     = "your-ssid"; // Change this to your WiFi SSID
const char* password = "your-password"; // Change this to your WiFi password

const char* host = "api.thingspeak.com"; // This should not be changed
const int httpPort = 80; // This should not be changed
const String channelID   = "2005329"; // Change this to your channel ID
const String writeApiKey = "V6YOTILH9I7D51F9"; // Change this to your Write API key
const String readApiKey = "34W6LGLIFXD56MPM"; // Change this to your Read API key

// The default example accepts one data filed named "field1"
// For your own server you can ofcourse create more of them.
int field1 = 0;

int numberOfResults = 3; // Number of results to be read
int fieldNumber = 1; // Field number which will be read out

void setup()
{
    Serial.begin(115200);
    while(!Serial){delay(100);}

    // We start by connecting to a WiFi network

    Serial.println();
    Serial.println("******************************************************");
    Serial.print("Connecting to ");
    Serial.println(ssid);

    WiFi.begin(ssid, password);

    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
}

void readResponse(WiFiClient *client){
  unsigned long timeout = millis();
  while(client->available() == 0){
    if(millis() - timeout > 5000){
      Serial.println(">>> Client Timeout !");
      client->stop();
      return;
    }
  }

  // Read all the lines of the reply from server and print them to Serial
  while(client->available()) {
    String line = client->readStringUntil('\r');
    Serial.print(line);
  }

  Serial.printf("\nClosing connection\n\n");
}

void loop(){
  WiFiClient client;
  String footer = String(" HTTP/1.1\r\n") + "Host: " + String(host) + "\r\n" + "Connection: close\r\n\r\n";

  // WRITE --------------------------------------------------------------------------------------------
  if (!client.connect(host, httpPort)) {
    return;
  }

  client.print("GET /update?api_key=" + writeApiKey + "&field1=" + field1 + footer);
  readResponse(&client);

  // READ --------------------------------------------------------------------------------------------

  String readRequest = "GET /channels/" + channelID + "/fields/" + fieldNumber + ".json?results=" + numberOfResults + " HTTP/1.1\r\n" +
                       "Host: " + host + "\r\n" +
                       "Connection: close\r\n\r\n";

  if (!client.connect(host, httpPort)) {
    return;
  }

  client.print(readRequest);
  readResponse(&client);

  // -------------------------------------------------------------------------------------------------

  ++field1;
  delay(10000);
}
```

I know matter handle wifi but it show error event I not calling any Arduino wifi function in my project, I use WifiClient only for confirm the bug.

SuGlider, 2024-08-29:
I think that Arduino WiFi library won't work when Matter is used along with it.
Matter has many native IDF calls to WiFi, LwIP and other netowrk components.
I'd suggest that the project would need to use native IDF WiFi API in order to make it work.

SuGlider, 2024-08-29:
The same will happen with BLE. Actually it is worst because Arduino BLE Library uses Blueroid and IDF Matter uses NimBLE. Both won't work together.

dzungpv, 2024-08-29:
I will try comment out wifi client lib from Arduino. but it work with Arduino 2.x. At least not show error when build code

dzungpv, 2024-08-29:
And many lib using wifi lib, not for create connection but check IP address or something, so remove wifi lib make Arduino useless. I can write code in native C++ but loose advantage of Arduino. From my log it failed at final build state when merge all the libs together.

SuGlider, 2024-08-29:
Are you mixing WiFi call and ESP Matter calls in the same project?
Or just trying to build an Arduino WiFi Porject using Arduino as IDF component and adding to it ESP Matter library?

SuGlider, 2024-08-29:
ESP_Matter over WiFi will use the settings from NVS to establish a connection the AP.
If none, it will wait for Matter Commissioning to complete in order to establish a connection to the AP.

Arduino WiFi library may break it in case it starts a connection to a different AP or it stops the current connection started by ESP_Matter.

Once WiFi is set and the ESP32 STA connects to the AP, it may be possible to use some Arduino WiFi API calls, but it may depend on previously configuration and setting values for the internal WiFi Class members. Therefore, it may not work as well.

ESP_Matter can't work within WiFi AP mode.

If the Arduino WiFi API method calls LwIP directly, it may get correct results, but remember that ESP_Matter relies on IPv6.

SuGlider, 2024-08-29:
Adding on top of the above, I think that the WiFi Events would have a conflict when Arduino WiFi Library and, at the same time, the ESP Matter also consume them.

dzungpv, 2024-08-29:
I want to use matter with exist Arduino projects and libs. Esp idf or arduino wifi handle I think not difference. I have merge Rainmaker and Esp Homekit sdk too, RM handle wifi but I can use Arduino to check wifi state, ip and more  even Arduino only support Bluedroid. 
I try to delete DESP32 tag and see that problem may come with namespace Wifi using by Matter, the compiler confuse with ::Wifi and Wifi from Arduino lib. Sum up, it could not build, so we will not talk about compatible now, now I just want to be success build with Arduino lib.

SuGlider, 2024-08-30:
@dzungpv - Thanks for the explanation about your request.

Let me investigate it and try to come up with a way to get these information when using Arduino + Matter as IDF Component:
1) Get the WiFi connection Status
2) Get the local IP address

Blueroid is for BLE operation only. NimBLE is another BLE stack.
Both are not related to WiFi.

dzungpv, 2024-08-30:
For Nimbel and Bluedroid, it is only example. With Rainmaker project I use it to get Bssi, Rssi and Wifi scan too, it work fine with native IDF wifi

SuGlider, 2024-08-30:
RainMaker is an application protocol, based on Netowrk Client/Server. It doesn't care about the network transport layer.

Matter is also an application protocol, but it cares about the network transport layer.
Therefore Matter controls the WiFi, Ethernet and Thread stack. 
Matter has its own WiFi Event Loop.
Arduino WiFi Library also controls the WiFi Event Loop, which causes a conflict.

In order to get the WiFi Status, your application shall use the Matter Callbacks and Matter mechanism for getting events in the user application side.

It is already in the example code in the using the `void app_event_cb(const ChipDeviceEvent *event, intptr_t arg)`

https://github.com/espressif/esp-matter/blob/f4bc6b8dfaf06253499c38f227c8ab27b4a05d38/examples/managed_component_light/main/app_main.cpp#L45-L118

`ChipDeviceEvent` has all necessary information for knowing when it has connected, what is the IP address, etc.
`chip::DeviceLayer::DeviceEventType::` namespace is what you need.

SuGlider, 2024-08-30:
@dzungpv - It seems that CHIP/Matter provides a way to control WiFi using RPC.
https://gautesl.github.io/connectedhomeip/docs/guides/esp32/rpc_console.html

dzungpv, 2024-08-30:
@SuGlider  Yes, now how I can build project with Arduino core 3.x and Matter? Because it simply failed.
We must find a way to make build success, because many Arduino libs using Wifi lib.
Then later we can use alternative way to get Wifi information.

SuGlider, 2024-08-30:
This is a long term development process. 

The main goal of the current effort is to allow users to build Matter Accessories using an ESP32 Arduino Matter Library (to be created). In such case, it will use the native Matter WiFi control layer and the native Matter Thread control layer as the first approach.

Currently I'm working on:
- Making it work using Arduino Core 3.0.x + IDF 5.1 using Arduino + Matter as IDF Components.
- Making it work using Arduino Core 3.1.x + IDF 5.3 also using Arduino + Matter as IDF Components.
- Once it works fine, next step would be building the necessary Arduino pre-compiled libraries using Arduino Lib Builder tools.

I would say that making Arduino WiFi Library and Arduino Matter Library compatible with each other is not in the priority list of the current project. But it can be achieved in the long term.

 >Yes, now how I can build project with Arduino core 3.x and Matter? 
 
 I'll post an example using Arduino + Matter as IDF Component based on Arduino Core 3.0.5 and IDF 5.1.
 It shall be placed in this folder:
 https://github.com/espressif/arduino-esp32/tree/master/idf_component_examples/
 
 I think I may have it done by the end of the next week.
 This example will be based on the this code:
 https://github.com/espressif/esp-matter/tree/f4eaa699f0fd8433bc5ca134328934f68f1e1525/examples/managed_component_light

HamzaHajeir, 2024-08-30:
> Yes, change it make build sample light matter success, but other problem when I am using Arduino WiFi, the error log:
>
> ```
> wind_FindEnclosingFunction  -Wl,--wrap=_Unwind_Resume [...]

Undefined reference implies that some sources are not included in the build. I remember I had this one just recently, I just enabled something within WiFi component through esp-idf menuconfig, then it compiles. 

Per ESP-Matter and WiFi library coexistence, I believe it's possible through telling Matter that WiFi is controlled by Application. 

See: https://github.com/project-chip/connectedhomeip/blob/1912865681cd51790a6619eff633f467cee3b632/src/include/platform/ConnectivityManager.h#L89-L95

And: https://github.com/project-chip/connectedhomeip/blob/1912865681cd51790a6619eff633f467cee3b632/src/platform/ESP32/ConnectivityManagerImpl.h#L112

And, here the source shows not reacting to calls (and wifi events) if so: https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/ConnectivityManagerImpl_WiFi.cpp [Search for kWiFiStationMode_ApplicationControlled].

I believe that the designers of CHIP just considers this range of use cases, where clients implement their own wifi management code. Consider a product which was in market before adopting Matter and yet have its own wifi management, should the manufacturer deliberately use CHIP wifi management? Nope.

dzungpv, 2024-08-30:
> Undefined reference implies that some sources are not included in the build. I remember I had this one just recently, I just enabled something within WiFi component through esp-idf menuconfig, then [...]

Follow you hint I was find the config that make Arduino could not build, it is CONFIG_ESP_WIFI_SOFTAP_SUPPORT=n, I enable it then the code can build. Now let play with Wifi part.
@SuGlider It is https://github.com/espressif/esp-matter/blob/f4eaa699f0fd8433bc5ca134328934f68f1e1525/examples/managed_component_light/sdkconfig.defaults#L36 change **n** to **y** then it is OK to build

SuGlider, 2024-08-30:
> I believe that the designers of CHIP just considers this range of use cases, where clients implement their own wifi management code. Consider a product which was in market before adopting Matter and [...]

Correct. 
When using ESP32 Matter, WiFi is managed by Matter when `CONFIG_ENABLE_WIFI_STATION` or `CONFIG_ENABLE_WIFI_AP` are set in the`sdkconfig` file.
When both are unset, Matter won't manage WiFi. In such case, WiFi provisioning shall be managed by the application.

HamzaHajeir, 2024-08-30:
> Correct. When using ESP32 Matter, WiFi is managed by Matter when `CONFIG_ENABLE_WIFI_STATION` or `CONFIG_ENABLE_WIFI_AP` are set in the`sdkconfig` file. When both are unset, Matter won't manage [...]

Some correction by considering that `CONFIG_ENABLE_WIFI_STATION` needs to be defined, so Matter can react to change its state onConnect and onDisconnect. Rather the developer needs to set `ConnectivityMgrImpl()._SetWiFiStationMode(kWiFiStationMode_ApplicationControlled)` if he wants to manage WiFi, if he have solid and reliable implementation of.

Jason2866, 2024-08-31:
@SuGlider The PR has a ugly side effect. It causes mass of redefine warnings with pioarduino / Platformio.
```

Building in release mode
Compiling .pio/build/tasmota32-nspanel/src/tasmota.ino.cpp.o
Building .pio/build/tasmota32-nspanel/bootloader.bin
Generating partitions .pio/build/tasmota32-nspanel/partitions.bin
<command-line>: warning: "ESP32" redefined
<command-line>: note: this is the location of the previous definition
esptool.py v4.7.6
Creating esp32 image...
Merged 2 ELF sections
Successfully created esp32 image.
Compiling .pio/build/tasmota32-nspanel/lib3cf/Wire/Wire.cpp.o
<command-line>: warning: "ESP32" redefined
<command-line>: note: this is the location of the previous definition
Compiling .pio/build/tasmota32-nspanel/lib53b/Adafruit_CCS811-1.0.0.14/Adafruit_CCS811.cpp.o
<command-line>: warning: "ESP32" redefined
<command-line>: note: this is the location of the previous definition
Compiling .pio/build/tasmota32-nspanel/lib18d/SPI/SPI.cpp.o
<command-line>: warning: "ESP32" redefined
<command-line>: note: this is the location of the previous definition
Archiving .pio/build/tasmota32-nspanel/lib3cf/libWire.a
Indexing .pio/build/tasmota32-nspanel/lib3cf/libWire.a
Compiling .pio/build/tasmota32-nspanel/libc70/Adafruit_MAX31865-1.1.0-custom/Adafruit_MAX31865.cpp.o
Archiving .pio/build/tasmota32-nspanel/lib18d/libSPI.a
Archiving .pio/build/tasmota32-nspanel/lib53b/libAdafruit_CCS811-1.0.0.14.a
Indexing .pio/build/tasmota32-nspanel/lib53b/libAdafruit_CCS811-1.0.0.14.a
<command-line>: warning: "ESP32" redefined
Indexing .pio/build/tasmota32-nspanel/lib18d/libSPI.a
<command-line>: note: this is the location of the previous definition
Compiling .pio/build/tasmota32-nspanel/libf9d/Adafruit_MCP9808_Tasmota/Adafruit_MCP9808.cpp.o
<command-line>: warning: "ESP32" redefined
<command-line>: note: this is the location of the previous definition
Compiling .pio/build/tasmota32-nspanel/lib2f4/Adafruit_BusIO/Adafruit_BusIO_Register.cpp.o
<command-line>: warning: "ESP32" redefined
<command-line>: note: this is the location of the previous definition
Archiving .pio/build/tasmota32-nspanel/libc70/libAdafruit_MAX31865-1.1.0-custom
Indexing .pio/build/tasmota32-nspanel/libc70/libAdafruit_MAX31865-1.1.0-custom
Compiling .pio/build/tasmota32-nspanel/lib2f4/Adafruit_BusIO/Adafruit_I2CDevice.cpp.o
<command-line>: warning: "ESP32" redefined
<command-line>: note: this is the location of the previous definition
Archiving .pio/build/tasmota32-nspanel/libf9d/libAdafruit_MCP9808_Tasmota.a
Indexing .pio/build/tasmota32-nspanel/libf9d/libAdafruit_MCP9808_Tasmota.a
Compiling .pio/build/tasmota32-nspanel/lib2f4/Adafruit_BusIO/Adafruit_SPIDevice.cpp.o
<command-line>: warning: "ESP32" redefined
<command-line>: note: this is the location of the previous definition
Compiling .pio/build/tasmota32-nspanel/libc3a/Adafruit_PM25AQI-1.0.6/Adafruit_PM25AQI.cpp.o
<command-line>: warning: "ESP32" redefined
<command-line>: note: this is the location of the previous definition
Compiling .pio/build/tasmota32-nspanel/lib0ea/Adafruit_SGP30-1.2.0/Adafruit_SGP30.cpp.o
<command-line>: warning: "ESP32" redefined
<command-line>: note: this is the location of the previous definition
Archiving .pio/build/tasmota32-nspanel/lib2f4/libAdafruit_BusIO.a
Indexing .pio/build/tasmota32-nspanel/lib2f4/libAdafruit_BusIO.a
Compiling .pio/build/tasmota32-nspanel/libdee/Adafruit_SGP40-1.1.0/Adafruit_SGP40.cpp.o
<command-line>: warning: "ESP32" redefined
<command-line>: note: this is the location of the previous definition
Archiving .pio/build/tasmota32-nspanel/libc3a/libAdafruit_PM25AQI-1.0.6.a
Indexing .pio/build/tasmota32-nspanel/libc3a/libAdafruit_PM25AQI-1.0.6.a
Compiling .pio/build/tasmota32-nspanel/libdee/Adafruit_SGP40-1.1.0/sensirion_voc_algorithm.c.o
<command-line>: warning: "ESP32" redefined
<command-line>: note: this is the location of the previous definition
...
```

SuGlider, 2024-08-31:
Yes, https://github.com/espressif/esp32-arduino-lib-builder/pull/217 will fix PIO adding `-DESP32=ESP32` to its settings.

SuGlider, 2024-09-01:
> Some correction by considering that `CONFIG_ENABLE_WIFI_STATION` needs to be defined, so Matter can react to change its state onConnect and onDisconnect. Rather the developer needs to set [...]

Thank @HamzaHajeir for the information!

I see that, when both `CONFIG_ENABLE_WIFI_STATION` and `CONFIG_ENABLE_WIFI_AP` are disabled in the `sdkconfig`, the header file `ConnectivityManagerImpl.h` will use the `platform/internal/GenericConnectivityManagerImpl_NoWiFi.h` as the Matter Network Manager Implementation. 

Therefore, maybe, we can override `GenericConnectivityManagerImpl_NoWiFi` class to use Arduino WiFi Library calls and achieve its integration with Matter.

Something to dig out.

HamzaHajeir, 2024-09-01:
> Thank @HamzaHajeir for the information!
>
> I see that, when both `CONFIG_ENABLE_WIFI_STATION` and `CONFIG_ENABLE_WIFI_AP` are disabled in the `sdkconfig`, the header file `ConnectivityManagerImpl.h` [...]

To my understanding, we don't need to override the No_WiFi implementation. we just keep CONFIG_ENABLE_WIFI_STATION defined, and CONFIG_ENABLE_WIFI_AP if needed to activate AP. But telling Matter to not touch WiFi with 'ApplicationControlled' concept. 

That way the application should be responsible to manage WiFi (+WiFiAP) machine state by itself. Considering some issues as WiFi AP SSID should comply to Matter's standard, for an example.

Therefore we have something like WiFiManager library. But however I don't know what Arduino WiFi library intersect with that matter, is it something as just managing automatic reconnection? We might consider relying on CHIP's implementation of WiFi management, at least as an available option...

SuGlider, 2024-09-02:
> Therefore we have something like WiFiManager library. But however I don't know what Arduino WiFi library intersect with that matter, is it something as just managing automatic reconnection? We might consider relying on CHIP's implementation of WiFi management, at least as an available option...

This integration with Arduino WiFi Library will be left for the long run.

As a first approach, the Arduino Matter Library will use CHIP and ESP_Matter as it is now, meaning that ESP_Matter will manage WiFi/Ethernet/Thread directly through ESP-IDF calls.

I also need to understand the implication of using Arduino WiFi Lib instead of CHIP WiFi Manager regarding Network Provisioning and Matter Fabric Commissioning processes.

SuGlider, 2024-10-16:
Initial PR for Arduino Matter Library is posted in PR #10467

SuGlider, 2024-10-21:
https://github.com/espressif/arduino-esp32/pull/10467/ is merger and Matter Library will be released into Arduino Core 3.1.0-RC2 or into initial release of Core 3.1.0.

Next step is the implementation of more Endpoint profiles as in the list:
- Color RGB Light
- Dimmable Light
- Smart Buttons
- Thermostat
- Fan
- Temperature Sensor
- Humidity Sensor
- Pressure Sensor
- Contact Sensor
- Occupancy Sensor

Fideus, 2025-01-22:
Hi, I created my own Matter RGBWW LED Spot. It works fine, but I'm not able to find any easy solution to give a device a unique code to setup multiple devices. Did I missed something?

me-no-dev, 2025-01-22:
@Fideus the library is using development license, which can provide only one code. To get your own code, you need to have a device and apply for and buy certificatation from the appropriate authorities: https://matter-smarthome.de/en/development/how-the-matter-certification-works/

me-no-dev, 2025-01-22:
You can setup the devices one by one using the same code and it will be OK

SuGlider, 2025-01-22:
> Hi, I created my own Matter RGBWW LED Spot. It works fine, but I'm not able to find any easy solution to give a device a unique code to setup multiple devices. Did I missed something?

As @me-no-dev said, the setup code for commissioning (34970112332) can be used for all the devices you may need in the same Matter network. Just commission one by one and each will have its own ID in the Matter environment.

Each device will be added to the Matter Controller with an Unique IPv6 address and each Matter RGBWW will be controlled individually by the APP (smartphone) or by an Automation.

Scenes and Groups can also be created for each Matter RGBWW light.

Jason2866, 2025-01-22:
In general this approach is possible. In project Tasmota we use the development key for all devices which are setup as Matter device. It is no problem to set up more devices. So it is not an issue in general. Cant speak for this Matter implementation since the one in Tasmota is completely written from scratch but Tasmota is written with actual Arduino Core.

Fideus, 2025-01-23:
I can't find a way to connect more than 1 device to my google home nest hub.
Isn't it possible to generate different test devices (with same test vendor)?
Or is there any other idea to solve this?

SuGlider, 2025-01-25:
> Unfortunately this does not work with my google nest hub. As soon I try to add another device compiled with the same code, It will remove the old one, and the new one also isn't connected.

It sounds like Google has updated their APP and their device Matter firmware in 2025.

I'd suggest trying to update the firmware of your Google Nest Hub.

Fideus, 2025-01-26:
The google Nest Hub is always up to date. II created a support ticket for that to google. Maybe they can solve that.
For the Arduino Matter library there isn't any chance to generate multiple different device ID's?

[19 more replies]