espressif/arduino-esp32 #7433: Rainmaker provisioning using BLE : QRcode, often does not , unreliable

atanudasgupta, 2022-11-06:
### Board

ESP32

### Device Description

Very often the device provisioning does not work even for simple examples like switch. Either I get 
1) It will ask me to provision manually
2) Or it would go through BLE pair and then cannot find the available WIFI

But it keep printing that device claiming is successful on logs.
Have tried many times , and may be once in 20 times works with the same code.

Something is terrible work with device claiming and provisioning, today i tried both with ESP-IDF and Ardruno, none is working now.  Although morning I was able to provision it. 

Issue is it does not provide any log for error in provisioning or claiming that can help me guess

Although I was successful in provision one device today, which was my own custom code after few attempts.

### Hardware Configuration

simple switch example 

### Version

v2.0.5

### IDE Name

Ardruino IDE, IDF

### Operating System

macOS

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

921600

### Description

quite unpredictable behavior from the android ESP Rainmaker provisioning of devices.

Finally you may the screen on the mobile app, Saying Rescan and no WIFI shows up, if try many times may be it will work once. But that makes it quite unreliable for any serious work.

### Sketch

```cpp
RMakerSwtich example
```

### Debug Message

```plain
8:52:10.064 -> ⸮YD⸮⸮#\⸮Y⸮ 	⸮[     4][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
18:52:10.509 -> [    23][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
18:52:10.545 -> [    49][I][RMaker.cpp:14] event_handler(): RainMaker Initialised.
18:52:10.545 -> [    54][V][WiFiGeneric.cpp:513] _arduino_event_cb(): Provisioning Initialized!
18:52:10.545 -> [    55][I][WiFiProv.cpp:133] beginProvision(): Starting AP using BLE. service_name : PROV_1235, pop : abcd1234
18:52:10.545 -> [    54][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 33 - PROV_INIT
18:52:10.653 -> [   148][V][WiFiGeneric.cpp:338] _arduino_event_cb(): STA Started
18:52:10.653 -> [   148][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
18:52:11.022 -> [   551][V][WiFiGeneric.cpp:519] _arduino_event_cb(): Provisioning Start!
18:52:11.057 -> [   552][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 35 - PROV_START
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

sanketwadekar, 2022-11-07:
Hi @atanudasgupta,
1. If Bluetooth pairing does not work, please check if your phone was previously paired with the node/device and unpair it from your Bluetooth settings.
2. It is possible that the device does not show the list of WiFi Access Points. In such a scenario, you can manually enter your network credentials through the phone app.
3. Claiming and Provisioning are different processes.
4. Erase your flash before provisioning.
5. In Arduino IDE, set the Output Log level to "Info" to avoid crashes due to low memory.

You can also refer to this [blog](https://blog.espressif.com/esp-rainmaker-now-in-arduino-cf1474526172).

Your logs show that the provisioning process was started on the device.
After trying the above steps, if you still get errors, even on the "Switch" example, then please share the detailed logs here.

VojtechBartoska, 2022-12-15:
@atanudasgupta Can I consider this covered?

atanudasgupta, 2022-12-15:
Yes this is covered

closed 2023-01-18 by VojtechBartoska