espressif/arduino-esp32 #7435: Versions above 2.0.2 won't work with MAX30102 sensor

nikkopt, 2022-11-06:
### Board

NodeMCU 32

### Device Description

Sensor is connected directly to board with dupont wires.

### Hardware Configuration

Connected by I2C. Nothing else connected.

### Version

v2.0.5

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Using the library provided by sparkfun, named "SparkFun_MAX3010x_Sensor_Library". Using the example nr 8 named "Example8 SPO2".

If i use arduino ESP32 version 2.0.2 or anything below, the sketch works. 
If i use version 2.0.3 or anything above, the board enters a boot loop and shows the following on the terminal: 
```
rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13160
load:0x40080400,len:3036
entry 0x400805e4
```

### Sketch

```cpp
https://github.com/sparkfun/SparkFun_MAX3010x_Sensor_Library/blob/master/examples/Example8_SPO2/Example8_SPO2.ino
```

### Debug Message

```plain
rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13160
load:0x40080400,len:3036
entry 0x400805e4
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

TD-er, 2022-11-07:
Which pins do you use for I2C?
Any change you're using GPIO 16 or 17?

nikkopt, 2022-11-07:
@TD-er no, "default" ones 21 and 22

Jason2866, 2022-11-07:
Probably the way how to the i2c bus is"talked too" is not conform to the way how it should be done. Since 2.0.3. it is less tolerant for such faulty implementations.
Raise an issue in the github of the lib.

mrengineer7777, 2022-11-07:
The I2C library in this repository has changed significantly between 1.X and 2.X.  It's likely your sensor library is not compatible with 2.X.

nikkopt, 2022-11-07:
@Jason2866 won't help much.. the source code hasn't been updated in more than 4 years. I'll try to find an alternative or i'll keep using version 2.0.2 just for this sensor. Thanks.

closed 2022-11-07 by nikkopt